espressif/arduino-esp32 #5908: Using Hardware Serial 0 ESP32

hicham-vv, 2021-11-19:
### Hardware:
Board:							esp dev kit v4
IDE name:						Platform.io
Flash Frequency:					40Mhz
Upload Speed:					115200

### Description:
Hello i am trying to use Hardware Serial 0 to communicate with another device, but it doesn't  work, as default hardware Serial 0 is used to communicate with the esp32 for programming and during reset/boot, i don't know if its that the problem or there is something else,
because when i change the communication  to Serial 1 or Serial 2, everything go fine.

my problem here that is i really need to communicate via hardware Serial 0, 
any ideas or help , Thank you

me-no-dev, 2021-11-19:
Could you please explain a bit better what you are doing? Connecting what to Serial0 how? You can use other pins for Serial0 if the USB UART chips is giving you headaches :)

hicham-vv, 2021-11-19:
i am connecting a sensor to uart 0 via two wires TX 0 and RX 0

hicham-vv, 2021-11-19:
i removed the usb uart chips to see if the problem from there but still got the same issues " no communication "

SuGlider, 2021-11-19:
@hicham-vv 
In order to better assist you, could you please run this code in order to get more information about your board and development environment.

``` dart
void setup () {
    Serial.printf("Internal Total heap %d, internal Free Heap %d\n", ESP.getHeapSize(), ESP.getFreeHeap());
    Serial.printf("SPIRam Total heap %d, SPIRam Free Heap %d\n", ESP.getPsramSize(), ESP.getFreePsram());
    Serial.printf("ChipRevision %d, Cpu Freq %d, SDK Version %s\n", ESP.getChipRevision(), ESP.getCpuFreqMHz(), ESP.getSdkVersion());
    Serial.printf("Flash Size %d, Flash Speed %d\n", ESP.getFlashChipSize(), ESP.getFlashChipSpeed());
}
```

hicham-vv, 2021-11-19:
Internal Total heap 402248, internal Free Heap 376776
SPIRam Total heap 0, SPIRam Free Heap 0
ChipRevision 1, Cpu Freq 240, SDK Version v3.3.4-432-g7a85334d8
Flash Size 16777216, Flash Speed 40000000

hicham-vv, 2021-11-19:
> Could you please explain a bit better what you are doing? Connecting what to Serial0 how? You can use other pins for Serial0 if the USB UART chips is giving you headaches :)

I can't use other pins, all the pins are used, i have only the uart 0 available, i need it only fo flashing a new program after that i need it to communicate with the sensor

SuGlider, 2021-11-19:
@hicham-vv 
Thanks for the fast answer.
From the information you provided, I can tell you are using PlatformIO with Arduino Core 1.0.6 and IDF 3.3.

hicham-vv, 2021-11-19:
i am using PlatformIO

SuGlider, 2021-11-19:
> I can't use other pins, all the pins are used, i have only the uart 0 available, i need it only fo flashing a new program after that i need it to communicate with the sensor

Your question is about changing pins for UART0?

SuGlider, 2021-11-19:
OR you can use pins 1 and 3 but you just need to change UART mode (baudrate, parity, stopbits etc)?

hicham-vv, 2021-11-19:
no i can't change pins for uart 0 ,  can the wires from the sensor are connected to uart 0 " GPIO 1 and GPIO 3 "

SuGlider, 2021-11-19:
So What is exactly your need?

hicham-vv, 2021-11-19:
exactly i need to communicate via UART 0,  " Serial 0"

hicham-vv, 2021-11-19:
i just wanna know if its possible or not

SuGlider, 2021-11-19:
Yes, it is 100% possible.
Just use ```Serial.xxxx()``` following Arduino API.
Connect RX to TX for each device (ESP32 and your sensor).

SuGlider, 2021-11-19:
Can you provide the code used to communicate youe sensor with Serial2, for instance?

hicham-vv, 2021-11-19:
> Yes, it is 100% possible. Just use `Serial.xxxx()` following Arduino API. Connect RX to TX for each device (ESP32 and your sensor).

yes that what  i wrote but my bad, it's not worked :(

SuGlider, 2021-11-19:
Can you provide the code used to successfully communicate your sensor with Serial2, for instance?

hicham-vv, 2021-11-19:
` 

    bool setupNano(){
    nano.msg[0]=1;
    nano.begin(Serial2);
    Serial2.begin(115200);
    while(!Serial2);
    delay(500);
    while (Serial2.available()) Serial2.read();
    delay(500);
    if (nanoSetTagProtocol()){
          return true;
     }else {return false;}
    }

`

SuGlider, 2021-11-19:
The main difference from Serial to Serial2 and Serial1 is that Serial is used to send boot information and error/debug messages when it 's enabled to do so.

Try this code:

``` dart

bool setupNano() {
  nano.msg[0] = 1;
  nano.begin(Serial);
  //Serial2.begin(115200);
  while (!Serial);
  delay(500);
  while (Serial.available()) Serial.read();
  delay(500);
  if (nanoSetTagProtocol()) {
    return true;
  } else {
    return false;
  }
}
```

hicham-vv, 2021-11-19:
> 

its not worked, :( i think  we need to set the baude rate speed

SuGlider, 2021-11-19:
OK. It makes a lot of sense to set it up correctly with the right parameters.
Those are the functions you can use with Arduino Core 1.0.6 (PlatformIO):

``` dart
Serial.updateBaudRate(unsigned long baud);
Serial.end();
Serial.begin(unsigned long baud, uint32_t config, int8_t rxPin, int8_t txPin, bool invert, unsigned long timeout_ms);

```

```uint32_t config``` can be one of these (for number of bits, parity and stop bits - default is ```SERIAL_8N1``` - 8 bits, No parity, 1 stop bit):
``` cpp
#define SERIAL_5N1 0x8000010
#define SERIAL_6N1 0x8000014
#define SERIAL_7N1 0x8000018
#define SERIAL_8N1 0x800001c
#define SERIAL_5N2 0x8000030
#define SERIAL_6N2 0x8000034
#define SERIAL_7N2 0x8000038
#define SERIAL_8N2 0x800003c
#define SERIAL_5E1 0x8000012
#define SERIAL_6E1 0x8000016
#define SERIAL_7E1 0x800001a
#define SERIAL_8E1 0x800001e
#define SERIAL_5E2 0x8000032
#define SERIAL_6E2 0x8000036
#define SERIAL_7E2 0x800003a
#define SERIAL_8E2 0x800003e
#define SERIAL_5O1 0x8000013
#define SERIAL_6O1 0x8000017
#define SERIAL_7O1 0x800001b
#define SERIAL_8O1 0x800001f
#define SERIAL_5O2 0x8000033
#define SERIAL_6O2 0x8000037
#define SERIAL_7O2 0x800003b
#define SERIAL_8O2 0x800003f
```

hicham-vv, 2021-11-19:
@SuGlider i already did that :(   ....
can you tell me how to force state of TX0 pin " GPIO 3 "  from HIGH to LOW  before Serial.begin() ,??

i tryed pin Mode output and digitalWrite LOW but nothing happend

SuGlider, 2021-11-19:
Maybe runing ```Serial.end()``` before trying to use ```pinMode()``` and ```digitalWrite()```.

Please remember that UART0 is used by boot software to send information when ESP32 starts up.
Not sure if this "boot serial data" may set your sensor in a wrong state right in the bootup, when using UART0.

As alternative, you may want to use Serial2 with other pins than 16 and 17.

hicham-vv, 2021-11-19:
Ok thank you anyways ...

SuGlider, 2021-11-19:
You may also want to try using Arduino Core 2.0.1 to verify is this issue also occurs.

SuGlider, 2021-11-19:
> @SuGlider i already did that :( .... can you tell me how to force state of TX0 pin " GPIO 1 " from HIGH to LOW before Serial.begin() ,??

Not sure about what you want to do, but there is an inverted mode in UART.
Inverting would set the idle UART state to 0v instead of 3.3v (UART default behavior).

```Serial.begin(begin(unsigned long baud, uint32_t config, int8_t rxPin, int8_t txPin, *bool invert*, unsigned long timeout_ms);```

hicham-vv, 2021-11-19:
> > @SuGlider i already did that :( .... can you tell me how to force state of TX0 pin " GPIO 1 " from HIGH to LOW before Serial.begin() ,??
> 
> Not sure about what you want to do, but there is an inverted mode in UART. Inverting would set the idle UART state to 0v instead of 3.3v (UART default behavior).
> 
> `Serial.begin(begin(unsigned long baud, uint32_t config, int8_t rxPin, int8_t txPin, bool invert, unsigned long timeout_ms);`
> 
> set _**bool invert**_ to true.
> 
> You can also only invert RX pin using `Serial.setRxInvert(true);`
i will try that  👍  because the sensor datasheet said that  the UART RX Serial input, 3.3V logic. Must be
low when module is powered on or off.
i will tell you if it's worked or not 
Thank you

hicham-vv, 2021-11-19:
it convert Tx too ? or it work only for Rx pin

SuGlider, 2021-11-19:
> it convert Tx too ? or it work only for Rx pin

```Serial.begin(....)``` with ```invert``` as ```true``` will invert the pins that are not equal to ```-1``` (which means to use default pin number).

### Examples (that work for Arduino Core 1.0.6 with PlatformIO):

```Serial.begin(9600, SERIAL_8N1, -1, 1, true);``` -- RX will **NOT** be inverted and will use default pin 3, **BUT TX** (set as pin 1) will be inverted because it is not ```-1```.
```Serial.begin(9600, SERIAL_8N1, -1, -1, true);``` -- **NO pin** will be inverted as both are ```-1``` and it will use default pins 1 and 3
```Serial.begin(9600, SERIAL_8N1, 3, 1, true);``` -- **both pins** RX (pin 3) and TX (pin 1) will be inverted.

hicham-vv, 2021-11-19:
> `Serial.begin(....)` with `invert` as `true` will invert the pins that are not equal to `-1` (which means to use default pin number).
>
> ### Examples (that work for Arduino Core 1.0.6 with [...]

allright 👍  i will try that and i will let you know the result , i hope it work

SuGlider, 2021-11-19:
> i will try that  👍  because the sensor datasheet said that  the UART RX Serial input, 3.3V logic. Must be
> low when module is powered on or off.
> i will tell you if it's worked or not
> Thank you

Humm... 
Not sure that UART_0 will work in that case beacuse, as I said, pin 1 (default TX) will send a lot of debug messages in boot time. 
Your sketch will start executing only after those messages are already sent out. 
This may diturb the initial setup of your sensor. 

These boot messages may be the reason for having trouble with using UART_0 for this sensor...

hicham-vv, 2021-11-19:
> > i will try that  👍  because the sensor datasheet said that  the UART RX Serial input, 3.3V logic. Must be
> > low when module is powered on or off.
> > i will tell you if it's worked or not
> > Thank you
> 
> Humm... Not sure that UART_0 will work in that case because, as I said, pin 1 (default TX) will send a lot of debug messages in boot time. Your sketch will start executing only after those messages are already sent out. This may diturb the initial setup of your sensor.
> 
> These boot messages may be the reason for having trouble with using UART_0 for this sensor...

my bad it not worked :( :( :(

SuGlider, 2021-11-22:
@hicham-vv 
Can we close this issue or you have some other related question?

Mesihas, 2021-11-29:
> ### Hardware:
> Board: esp dev kit v4 IDE name: Platform.io Flash Frequency: 40Mhz Upload Speed: 115200
>
> ### Description:
> Hello i am trying to use Hardware Serial 0 to communicate with another device, [...]

If you use the Arduino IDE to program the ESP32, you might be interested in [Andreas Spiess’] latest video . In it, he shows an example of using all three ESP32 UARTs from an Arduino program. He calls the third port “secret” although that’s really a misnomer. However, it does require a quick patch to the Arduino library to make it work.

Just gaining access to the additional UARTs isn’t hard. You simply use one of the additional serial port objects available. However, enabling UART 1 causes the ESP32 to crash! The reason is that by default, UART 1 uses the same pins as the ESP32 flash memory.

Luckily, the chip has a matrix switch that can put nearly any logical I/O pin on any physical I/O pin. [Andreas] shows how to modify the code, so that UART 1 maps to unused pins, which makes everything work. it is a simple change, replacing two parameters to a call that — among other things — maps the I/O pins. You could use the technique to relocate the UARTs to other places if you choose.

If you want to learn more about the ESP32, we covered a good set of tutorials for you to check out. Or if you just want a quick overview, you can start here.

SuGlider, 2021-11-29:
> If you use the Arduino IDE to program the ESP32, you might be interested in [[Andreas Spiess’] latest video](https://www.youtube.com/watch?v=GwShqW39jlE&ab_channel=AndreasSpiess) . In it, he shows [...]

This a 2017 video. It is way too old. 
Do not follow this "secret" because the current ESP32 Arduino version 2.0.1 has refactored UART.
In order to use UART0, UART1 and UART2, just use Arduino ```Serial.begin()```, ```Serial1.begin()``` and ```Serial2.begin()```

ESP32 has those 3 UART ports available. It is possible to change RX and TX pins in the ```begin()``` call.

One important note is that ESP32 UART1 default pins are GPIO 9 and 10. Those pins are also used for Flash access when QIO mode is activated. In the Arduino IDE, it is possible to change to the DIO mode in order to let GPIO 9 and 10 to be used in UART1.Or just change UART1 pins to something else.

hicham-vv, 2021-11-29:
yes i changed the uart 1 to 27 and 26 pin

SuGlider, 2021-11-29:
@hicham-vv 
We recommend using Arduino Core 2.0.1 because UART has been refactored in version 2.0.0+.
You are using Core 1.0.6 as per your messages. This version has some bugs.

hicham-vv, 2021-11-30:
@SuGlider  Yes u can close it *

closed 2021-11-30 by SuGlider

sabado, 2022-03-13:
Problem is real, i have the same situation. I want to communicate two ESP32 using UART 0, and that config don't work at all. If i send a command from A to B or vice versa , nothing happens. Also FTDI chip dont works when both devices are connected ( trying to sniffe ). I guess GPIOS are disabled because the FTDI chip is present. I have to check everything and will update later.

me-no-dev, 2022-03-14:
FTDI will pull the lines. Try with different pins or without the FTDI

cpapir, 2022-06-28:
The capability to access UART 0 through the Rx0 and Tx0 pins is real and obviously a problem to a number of users.  I have the same problem and I have NO solution. The technical solution is POSSIBLE no one has explained exactly how.  The solution needs to be available for Arduino developers (on Arduino IDE) because a great number of developers this is what they use.

HardwareSerial.h appears to have the infrastructure to accomplish this task. BUT how to use from an Arduino scketch is NOT well documented. Additionally when one tries to use it it crashes (In  the utube https://github.com/espressif/arduino-esp32/issues/5908a number of the problems with HardwareSerial are explained).

Corrections to HardwareSerial.h and or the core may be the solution,
\

If any one in espressif reads this comment please address this issue.  It is significant

If some one has a way around this issue please help

SuGlider, 2022-06-28:
@cpapir - What exactly are you trying to achive with UART 0?

Arduino Serial (UART 0) works exactly the same way that the other UART ports (1 and 2).
The main difference is that it is used by default to output BOOT information and also debug information when this is active.

Please feed us with more details and information about what you need to use with UART 0.

SuGlider, 2023-04-02:
@empewoow 
What is the SoC (esp32, S2, C3, S3) and what Arduino Core are you are using?

bassamkassem, 2023-05-30:
> ### Hardware:
> Board: esp dev kit v4 IDE name: Platform.io Flash Frequency: 40Mhz Upload Speed: 115200
>
> ### Description:
> Hello i am trying to use Hardware Serial 0 to communicate with another device, [...]

Here is the solution at least it worked for me 
Board ESP32 DEV KIT , communicating with Arduino Mega Pro 2560 using Adafruit 4 channels LLC in between, noting that Mega is using all serial ports in other components , such as Adafruit rotary encoder , RTC , OLED , ... etc ..

Now I can receive data from Mega to ESP and vise versa 

Serial.begin(115200); //, SERIAL_8N1, 16, 17);
Serial2.begin(9600, SERIAL_8N1, 3, -1,false); //
Wire.begin(SDA,SCL); // #define WIRE_SLAVE_ADDRESS 9 Add this definition at the beginning for slave declaration 
while (!Serial2) delay(1000);

empewoow, 2023-05-31:
> @empewoow What is the SoC (esp32, S2, C3, S3) and what Arduino Core are you are using?

I have the ESP32 DevKitC V4. Not really sure what is the SoC. I used the latest Arduino at the time, so 2.0.X probably.

mathemaphysics, 2024-07-15:
I'm trying to write to an RS485 breakout board via RX0/TX0, i.e. via Serial0. If I'm reading from it then it works, but I'm running an event loop in which I'm also connected to Serial and taking input from it. After reading from Serial0 begins, the next time Serial.available() runs it freezes and reboots.

It would be great if someone could explain to me why Serial0 would care what the USB serial connection is doing.

SuGlider, 2024-08-02:
ESP32-S3/C3/C6/H2/S2 Arduino 3.0.x works ike this:

1- Is "ARDUINO_USB_CDC_ON_BOOT" enabled (1)?
- Serial is USB CDC
- Serial0, Serial1 and Serial2 (S3/C6 only) are UART 0, 1, 2.

2- "ARDUINO_USB_CDC_ON_BOOT" is disabled (0)
- Serial is UART0
- Serial1 is UART1 and Serial2 is UART2 (S3/C6 only)