espressif/arduino-esp32 #5912: ESP-C3-12F crashes when serving static files with AsyncWebServer and LittleFS

AcuarioCat, 2021-11-20:
### Hardware:
Board:							ESP-32-C3
Core Installation version:			2.0.0
IDE name:						Visual Studio/Visual Micro
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
I don't know if this is the ESP code or AsyncWebServer code that is causing the problem but...

When using the ESP-C3 with the AsyncWebServer and trying to serve a static file with `server.serveStatic("/", LittleFS, "/");` the device crashes and resets.

Compiling and running the same code on an ESP32 works correctly with no crash.

Serving a file using server.on... works ok, no crash.
i.e. opening http://192.168.0.116/deb works ok
opening http://192.168.0.116/debug1.html crashes (debug1.html does not depend on any other LittleFS download, js, png etc..)

### Sketch:  
You need to create and upload some web pages using makelittlefs.exe to test this out. The code is from the simple_server.ino sample modified to serve files from LittleFS.

I'm running mklittlefs  version: 0.2.3-27-gc41e51a

```cpp

#include <Arduino.h>
#ifdef ESP32
#include <WiFi.h>
#include <AsyncTCP.h>
#elif defined(ESP8266)
#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>
#endif
#include <ESPAsyncWebServer.h>
#include "LITTLEFS.h"

AsyncWebServer server(80);

const char* ssid = "YOUR_SSID";
const char* password = "YOUR_PASSWORD";

const char* PARAM_MESSAGE = "message";

void listDir(fs::FS& fs, const char* dirname, uint8_t levels) {
    Serial.printf("Listing directory: %s\r\n", dirname);

    File root = fs.open(dirname);
    if (!root) {
        Serial.println("- failed to open directory");
        return;
    }
    if (!root.isDirectory()) {
        Serial.println(" - not a directory");
        return;
    }

    File file = root.openNextFile();
    while (file) {
        if (file.isDirectory()) {
            Serial.print("  DIR : ");
            Serial.println(file.name());
            if (levels) {
                listDir(fs, file.path(), levels - 1);
            }
        }
        else {
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("\tSIZE: ");
            Serial.println(file.size());
        }
        file = root.openNextFile();
    }
}

void notFound(AsyncWebServerRequest* request) {
    request->send(404, "text/plain", "Not found");
}

void setup() {

    Serial.begin(115200);
    WiFi.mode(WIFI_STA);
    WiFi.begin(ssid, password);
    if (WiFi.waitForConnectResult() != WL_CONNECTED) {
        Serial.printf("WiFi Failed!\n");
        return;
    }

    Serial.print("IP Address: ");
    Serial.println(WiFi.localIP());

    // Format FileFS if not yet
    if (!LittleFS.begin())
    {
        Serial.println("LITTLEFS failed! AutoFormatting.");
        LittleFS.format();
    }

    Serial.println("Level 1:");
    listDir(LittleFS, "/", 1);

    server.on("/deb", HTTP_GET, [](AsyncWebServerRequest* request) {
        request->send(LittleFS, "/debug1.html");
        });

    server.on("/", HTTP_GET, [](AsyncWebServerRequest* request) {
        request->send(200, "text/plain", "Hello, world");
        });

    // Send a GET request to <IP>/get?message=<message>
    server.on("/get", HTTP_GET, [](AsyncWebServerRequest* request) {
        String message;
        if (request->hasParam(PARAM_MESSAGE)) {
            message = request->getParam(PARAM_MESSAGE)->value();
        }
        else {
            message = "No message sent";
        }
        request->send(200, "text/plain", "Hello, GET: " + message);
        });

    // Send a POST request to <IP>/post with a form field message set to <message>
    server.on("/post", HTTP_POST, [](AsyncWebServerRequest* request) {
        String message;
        if (request->hasParam(PARAM_MESSAGE, true)) {
            message = request->getParam(PARAM_MESSAGE, true)->value();
        }
        else {
            message = "No message sent";
        }
        request->send(200, "text/plain", "Hello, POST: " + message);
        });

    server.serveStatic("/", LittleFS, "/");
    server.onNotFound(notFound);

    server.begin();
}

void loop() {
}
```

### Debug Messages:
```
Trying to open http://192.168.0.126/debug1.html throws error:
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Listing directory: /js
  FILE: jquery.min.js.gz	SIZE: 29240

Trying to open http://192.168.0.126/js/jquery.min.js throws the following error:
[ 41735][E][vfs_api.cpp:64] open(): /littlefs/js/jquery.min.js does not exist
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x4200862a  RA      : 0x4200868c  SP      : 0x3fcad5f0  GP      : 0x3fc8e800  
TP      : 0x3fc98bcc  T0      : 0x4005890e  T1      : 0x4038c298  T2      : 0x00300010  
S0/FP   : 0x3fca7764  S1      : 0x3fc961a4  A0      : 0x00000001  A1      : 0x98940ca0  
A2      : 0x3fca7764  A3      : 0x00000001  A4      : 0x3fc96000  A5      : 0x3fc961a4  
A6      : 0x3fe00000  A7      : 0x00000008  S2      : 0x3fca81dc  S3      : 0x3fca81ec  
S4      : 0x3fc961a4  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x0070015a  T4      : 0x7759990c  T5      : 0x00601700  T6      : 0x00723800  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x98940caf  
MHARTID : 0x00000000  

Stack memory:
3fcad5f0: 0x0a002c00 0x06ff0000 0x3fc961a4 0x00000000 0x00000000 0x00000000 0x00000000 0x0300b29e
```

DmytroKorniienko, 2021-11-23:
@AcuarioCat , Hi, can you check please behavior with this fork/branch of [ESPAsyncWebServer](https://github.com/DmytroKorniienko/ESPAsyncWebServer/tree/ESPAsyncWebServerMod)?

Or as another point for testing you can try comment out this line in ESPAsyncWebServer\src\StringArray.h :

```
    bool remove(const T& t){
      auto it = _root;
      auto pit = _root;
      while(it){
        if(it->value() == t){
          if(it == _root){
            _root = _root->next;
          } else {
            pit->next = it->next;
          }
          
          if (_onRemove) {
            _onRemove(it->value());
          }
          
          //delete it; // WARNING: for testing purposes only, memory leak if leave commented line
          return true;
        }
        pit = it;
        it = it->next;
      }
      return false;
    }
```

* This issue related with http-headers handling.

AcuarioCat, 2021-11-23:
Hi, great, I tried both suggestions and both work. I guess I need to stay with the forked version for now.
I have been using Aardvark MQTT but this 'demands' it's own modified (and broken :-( ) AsyncWebServer. 
I guess it's time to investigate MQTT options again.

DmytroKorniienko, 2021-11-23:
@me-no-dev Can you please take a look at this issue? Maybe you can provide better solution for ESP32-C3 support and clarify whats going wrong there...

--------------------

@AcuarioCat 
We using AsyncMqttClient for our EmbUI project (currently supported ESP8266, ESP32, ESP32-S2, ESP32-C3; ESP32-S3 - not tested yet), you can check my github page - this is open sourse software, maybe something will be useful.

VojtechBartoska, 2022-04-08:
Hello, can you please help with testing this on 2.0.3-rc1? Thanks!

VojtechBartoska, 2022-05-05:
2.0.3 stable is out, it's now easier to validate this.

DmytroKorniienko, 2022-05-08:
@VojtechBartoska 2.0.3 not working with LittleFS for me at all, more info there: https://github.com/espressif/arduino-esp32/issues/6579#issuecomment-1120362411 , can't re-check.

DmytroKorniienko, 2022-05-08:
```
Compiling .pio\build\esp32c3\lib7d0\ESP Async WebServer\WebAuthentication.cpp.o
.pio/libdeps/esp32c3/ESP Async WebServer/src/AsyncWebSocket.cpp: In member function 'IPAddress AsyncWebSocketClient::remoteIP()':
.pio/libdeps/esp32c3/ESP Async WebServer/src/AsyncWebSocket.cpp:832:28: error: call of overloaded 'IPAddress(unsigned int)' is ambiguous
         return IPAddress(0U);
```

for https://github.com/me-no-dev/ESPAsyncWebServer after patching to return IPAddress((uint32_t)0U); build is successful but not working with a https://github.com/tasmota/platform-espressif32/releases/download/v.2.0.3/platform-espressif32-v.2.0.3.zip and crashing by littlefs error on framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.3

My patched version from https://github.com/espressif/arduino-esp32/issues/5912#issuecomment-976459778 works with a [tasmota release](https://github.com/tasmota/platform-espressif32/releases/download/v.2.0.3/platform-espressif32-v.2.0.3.zip) and stil fails with a  framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.3.

```
17:26:55.575 > ./components/esp_littlefs/src/littlefs/lfs.c:1790:debug: Bad block at 0x0
17:26:55.581 > ./components/esp_littlefs/src/littlefs/lfs.c:1796:warn: Superblock 0x0 has become unwritable
17:26:55.589 > E (12125) esp_littlefs: Failed to format filesystem
17:26:55.592 > [ 12092][E][LittleFS.cpp:119] format(): Formatting LittleFS failed! Error: -1
17:26:55.600 > [ 12099][E][LittleFS.cpp:94] begin(): Mounting LittleFS failed! Error: -1
17:26:55.695 > E (12242) esp_littlefs: Partition already used
17:26:55.699 > E (12242) esp_littlefs: Failed to initialize LittleFS
17:26:55.704 > [ 12205][E][LittleFS.cpp:94] begin(): Mounting LittleFS failed! Error: 259
17:26:55.800 > ./components/esp_littlefs/src/littlefs/lfs.c:1071:error: Corrupted dir pair at {0x0, 0x1}
17:26:55.806 > E (12346) esp_littlefs: mount failed,  (-84)
17:26:55.811 > E (12347) esp_littlefs: Failed to initialize LittleFS
```

closed 2025-01-16 by Parsaabasi