espressif/arduino-esp32 #5913: Is core v2.0.1 compatible with W5500 ethernet module ?

kbssa, 2021-11-22:
I can't manage to get the W5500 working over the v2.0.1 core with Arduino Ethernet library.

The ETH.h library is compatible with W5500 ? If so, is there any example or any guide where I can see how to make it work ?

Any help would be appreciated.

Now it's working over de core v1.0.6 with Arduino Ethernet library, but I want to upgrade to v2.0.1.

Thanks in advance.

felmue, 2021-11-22:
Hello @kbssa 

I don't think ETH.h library is compatible with W5500 given this list in ETH.h:
```
typedef enum { ETH_PHY_LAN8720, ETH_PHY_TLK110, ETH_PHY_RTL8201, ETH_PHY_DP83848, ETH_PHY_DM9051, ETH_PHY_KSZ8081, ETH_PHY_MAX } eth_phy_type_t;
```
What is the exact issue you are getting with the Arduino Ethernet library? Compilation? Error at runtime?

Thanks
Felix

SuGlider, 2021-11-23:
@kbssa 

W5500 is supported by IDF 4.4 but the current Arduino v2.0.0+ Lib ETH.h does not implement it, yet.

### About W5500 IDF support in Arduino Core v2.0.0+:

It can be found in sdkconfig file for ESP32.
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/sdkconfig#L618

W5500 initialization function from IDF 4.4 is included here:
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/esp32/include/esp_eth/include/esp_eth_mac.h#L456-L487

**ESP-IDF example** with W5500 is here, but this piece of code is not included within core 2.0.1 version of ETH.cpp:
https://github.com/espressif/esp-idf/blob/master/examples/ethernet/basic/main/ethernet_example_main.c#L225-L250

SuGlider, 2021-11-23:
@kbssa 

I could not find W5500 in Arduino Core v1.0.6

From ETH.h I found this list:
```typedef enum { ETH_PHY_LAN8720, ETH_PHY_TLK110, ETH_PHY_IP101, ETH_PHY_MAX } eth_phy_type_t;```

Which one of those options did you use to make W5500 work?

kbssa, 2021-11-23:
> @kbssa
>
> W5500 is supported by IDF 4.4 but the current Arduino v2.0.0+ Lib ETH.h does not implement it, yet.
>
> ### About W5500 IDF support in Arduino Core v2.0.0+:
> It can be found in sdkconfig file [...]

Is there any working example ? 

I have no idea on how to include it in my code as I am completely noob with IDF implemention.

kbssa, 2021-11-23:
> @kbssa
> 
> I could not find W5500 in Arduino Core v1.0.6
> 
> From ETH.h I found this list: `typedef enum { ETH_PHY_LAN8720, ETH_PHY_TLK110, ETH_PHY_IP101, ETH_PHY_MAX } eth_phy_type_t;`
> 
> Which one of those options did you use to make W5500 work?

The W5500 works with the Arduino Ethernet library in Arduino Core v1.0.6.

SuGlider, 2022-03-15:
@kbssa
W5500 is compatible with any core since it uses SPI for communication with ESP32.
You may need to use the right Arduino Library.

Arduino has a standard library that works fine with W5500/W5100/W5200:
https://www.arduino.cc/en/Reference/Ethernet
https://github.com/arduino-libraries/Ethernet

closed 2022-03-15 by SuGlider