espressif/arduino-esp32 #5914: S3 support

wxzed, 2021-11-22:
When will s3 support

chegewara, 2021-11-22:
When its ready.

wxzed, 2021-11-22:
Is there an approximate time?

igrr, 2021-11-22:
Hi @wxzed, please follow the link to https://github.com/espressif/arduino-esp32/issues/5594 and click "subscribe" there to get notified about S3 support progress.

@chegewara I appreciate you helping respond to Github issues, your comments are generally very helpful! I kindly ask you to keep making constructive comments in the future, or ignore the issue if you don't see a way to respond constructively.

closed 2021-11-22 by igrr

chegewara, 2021-11-22:
Constructive response:
- please use search:
- https://github.com/espressif/arduino-esp32/issues?q=is%3Aissue+s3+support+is%3Aclosed