espressif/arduino-esp32 #5921: After some time esp32 module reseting for same work

ebineser, 2021-11-23:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do?
  i can do some work with this module, like when the water is filling led is off.
Describe your system( Hardware, computer, O/S, core version, environment).
  esp32 dev module,arduino ide software using
Describe what is failing.
 In starting this woriking properly when the l]water is filling the signal is coming the led is off but after few minutes it entering into the same loop take reseting i dont know why. this problem i identify by using debugging when the system entering into the loop one of the text is printing. So, this reseting when it entering into the same loop because same text printing after that it get resetting. But it happens only after few minutes exactlly 15-20 minutes but starting it working properly

All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? ?node32? ?ttgo_lora?
Core Installation version:			?1.0.0? ?1.0.1-rc4? ?1.0.1? ?1.0.1-git? ?1.0.2? ?1.0.3?
IDE name:						?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no? ?yes?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10? ?Mac OSX? ?Ubuntu?

### Description:
Describe your problem here

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

SuGlider, 2021-11-24:
@ebineser 
It lacks information in order to help you with this issue. 
Can you provide a minimum sketch that causes the reset you mentioned?

It also will help a lot if you can send the debug output message associated with the execution and reset.

VojtechBartoska, 2021-12-14:
@ebineser Can you please provide more info?

closed 2022-01-27 by VojtechBartoska