espressif/arduino-esp32 #5923: Problem with using HSPI and WiFi simultenously. WiFi can't connect to AP with AUTH_EXPIRE error.

dexion, 2021-11-23:
### Hardware:
Board:							?ESP-WROOM-32?
Core Installation version:			?1.0.4? ?2.0.1? ?latest build from github?
IDE name:						?Arduino IDE? ?Platform.io? doesn't matter
Flash Frequency:					?40Mhz?
PSRAM enabled:                                    ?no?
Upload Speed:					?115200?
Computer OS:                                        ?Windows 10?

### Description:
We have PCB with esp-wroom-32. PINs used as:

Hardware PIN | GPIO | Used as
-- | -- | --
0 | GND | GND
1 | GND | GND
2 | 3,3V | VDD
3 | ChipPU | 
4 | 36 |  
5 | 39 |  
6 | 34 | 
7 | 35 |  
8 | 32 | BUTTON
9 | 33 |  
10 | 25 | 
11 | 26 | 
12 | 27 |  
13 | 14 | SPI_CLK
14 | 12 |  
15 | GND |  
16 | 13 | SPI_MOSI
17 | 9 |  
18 | 10 |  
19 | 11 |  
20 | 6 |  
21 | 7 |  
22 | 8 |  
23 | 15 | 
24 | 2 |  
25 | 0 | JUMPER
26 | 4 | 
27 | 16 | 
28 | 17 | 
29 | 5 |  
30 | 18 | SPI_CS_1
31 | 19 | SPI_CS_2
32 | NC |  
33 | 21 | BAR_I2C2_SDA
34 | 3 | PROGRAM_RX
35 | 1 | PROGRAM_TX
36 | 22 | BAR_I2C2_SLC
37 | 23 | SPI_CS_3
38 | GND | GND

We use HSPI at PINS GPIO 13 (MOSI), 14 (CLK) (aka ADC2), 18 (CS_1), 19 (CS_2). On another side of SPI we have HT1632 controller. When we use a simple sketch just for connecting to WiFi we have the same problem as with full program sketch (which uses HT1632) - esp can not connect to the AP with error AUTH_EXPIRED. We tried some workarounds from the Internet - no one make it works. Another strange behavior - esp will connect to the AP only at the exact nearest place to the AP (~10-15cm), after that, we can move esp to different distance and they still work (problem only at connection time to the AP). What we also tried:
* different routers and APs (Mikrotik, Asus, Xiaomi, Zyxel)
* B\G\N mode, combine it
* change power of AP
* change channel (1-13), the width of the channel (20/40)
* change power supply for esp
When we deleted HT1632 (esp pins "left unplugged") - esp successfully connects to any "problem" AP. We decide that GPIO14 used in WiFi working and it works unexpectedly.

### Sketch:
```cpp
#include <WiFi.h>

int PINS[28] = {25, 26, 14, 13, 15, 4, 36, 39, 34, 35, 32, 33, 27, 12, 9, 10, 11, 7, 8, 2, 21, 22, 16, 17, 18, 19, 23, 6};

// PIN_6 corrupts serial output

void setup()
{
  for (int i = 0; i < 27; i++)
  {
    pinMode(PINS[i], INPUT_PULLUP); // try to set pinMode explicitly - didn't help
  }

  delay(1000);

  const char *ssid = "SSID";
  const char *password = "password";

  Serial.begin(115200);
  delay(10);

  // We start by connecting to a WiFi network

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  delay(100);
  Serial.println("persisten false");
  WiFi.persistent(false);
  delay(100);
  Serial.println("disconnect");
  WiFi.disconnect(false, true);
  delay(100);
  Serial.println("mode off");
  WiFi.mode(WIFI_OFF);
  delay(100);
  Serial.println("enable sta");
  WiFi.enableSTA(true);
  delay(100);
  Serial.println("mode STA");
  WiFi.mode(WIFI_STA);
  delay(100);
  Serial.println("hostname");
  // WiFi.setHostname("NeboMy");
  delay(100);
  Serial.println("connection");
  WiFi.begin(ssid, password);

  int notConnectedCounter = 0;

  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
    notConnectedCounter++;
    if (notConnectedCounter > 150)
    { // Reset board if not connected after 5s
      Serial.println("Resetting due to Wifi not connecting...");
      ESP.restart();
    }
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {}

```

### Debug Messages:
log outputs is very long, so I think that link to the pastebin should be better
[HSPI connected but not used (sketch attached), connection failed](https://pastebin.com/0PinNsxt)
[same PCB without connected HSPI - connection is successfully](https://pastebin.com/X6Z6tVWs)

How we can use HSPI and WiFi simultaneously? We already use VSPI for different periferal and can not use another PINs (we don't have another free PINs, in the table above I wrote not all our PINs, only related to the WiFi and HSPI)

atanisoft, 2021-11-23:
@dexion Usually this is a power related issue, are you supplying at least 750mA (1A if shared with other devices) on the 3v3 inputs?

dexion, 2021-11-23:
Power supply is ok

atanisoft, 2021-11-23:
Stating "power supply is ok" doesn't answer the Q of are you providing sufficient current for what you are trying to run on your custom PCB. What LDO are you using on the PCB, what sort of power supply is feeding the LDO, etc.

I also have custom PCBs that also use the 12-15 range of pins for HSPI and don't see issues. On my PCB I'm using a 2A 5v buck converter supplied by an external 15-28v supply and a 1A 3v3 LDO to supply the esp32 module. Average current draw from the external supply is around 250mA 15V DC with everything active.

dexion, 2021-11-24:
We are using laboratory power supply for testing, and see 120 mA at 5V DC. So I don't think that problem is in power supply.

TD-er, 2021-11-24:
Not saying the power is an issue, nor that your lab psu is of bad quality.
But just to let you know about what happens between the lab PSU and the ESP.
You have of course wires, but let's assume those are thick enough to not cause a significant voltage drop.
Then there is a voltage regulator turning 5V into 3V3.
That one is underpowered on many boards sold with ESP's.
When connecting to WiFi the ESP may surge upto 500 mA, but if your onboard voltage regulator is only capable of handling 150 mA, then the voltage will drop on the 3V3 line.
ESP processors really need a stable 3V3 line when making the WiFi connection.
If this voltage isn't stable during WiFi initialization, then you may end up with all kinds of strange issues.
It may even operate at the wrong frequency and thus not receive any reply from the access point. (hence Auth Expire)
But if you manage to make a connection, the network responsiveness may not be as snappy as it can be. (e.g. ping times are fluctuating, or give timeouts).

It could still very well be a totally different issue here, but make sure to always check the 3V3 line to see if it remains stable during WiFi connect. Preferrably using a scope.
Very thin copper traces on the board for the 3V3 line or GND may also cause issues which are harder to measure as you need to measure the voltage at the ESP directly to see them.

VojtechBartoska, 2022-04-08:
Is this still valid? If yes, can you please retest this with 2.0.3-rc1?

dexion, 2022-04-09:
I was valid. Sorry, I can't retest it because I've already changed PCB (now I just don't use some pins). As I said earlier, problem isn't in power supply or something. Problem was in simulteonsly HSPI and WIFI. We change pins for our HT1632, and we don't use old HSPI pins. Now problem resolved by using different pins

VojtechBartoska, 2022-04-11:
ok, thanks for answer. Closing.

closed 2022-04-11 by VojtechBartoska

anthonywww, 2023-12-13:
Having a similar issue. Need to use HSPI for Adafruit_SSD1351