espressif/arduino-esp32 #5924: xQueueGenericReceive)- assert failed

wenzhicode, 2021-11-24:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			not sure,  installed by Platform IO with vscode
IDE name:						Platform.io
Flash Frequency:					?40Mhz?
PSRAM enabled:                                     ?no
Upload Speed:					460800
Computer OS:                                         Windows 10

### Description:
Hi .
I use ESP32 for  [ ELRS project,](https://github.com/ExpressLRS/ExpressLRS) ,and  a SSD1306_128x64 OLED for menu operation.
when I use  [u8g2 lib](https://github.com/olikraus/u8g2) send  1024byte to OLED for updating the menu, it shows a  **queue.c:1446 (xQueueGenericReceive)- assert failed**,and reboot.

form the backtrace, I saw that I2CWrite() will call i2cProcQueue,then into xEventGroupWaitBits(), but this time a hwtimer occured, and it went to process hwtimer callback ,which used SPI bus to transferBytes to another chip, when this callback finished, it should go back to i2cProcQueue. BUT something wrong this time , **queue.c:1446 (xQueueGenericReceive)- assert failed**.

I do not how this happened, any advice will be appreciated.

### Sketch:  
//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:1446 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x4008c0df on core 1

0x4008b0ec: **invoke_abort** at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
0x4008b369: **abort** at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
0x4008c0df: **xQueueGenericReceive** at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c:2038
0x4014e6e5: **spiTransferBytes** at C:\Users\wenzh\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-spi.c:283
... ...
0x4008239d: **hwTimer::callback()** at E:\GitHub\ExpressLRS\src/lib\HWTIMER/ESP32_hwTimer.cpp:20
0x40082dd9: **__timerISR** at C:\Users\wenzh\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-timer.c:174
0x4008d4fd: **vTaskExitCritical** at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
0x4008c9a0: **xEventGroupWaitBits** at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/event_groups.c:338
0x415618f: **i2cProcQueue** at C:\Users\wenzh\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-i2c.c:1287
0x4015657f: **i2cWrite** at C:\Users\wenzh\.platformio\packages\framework-arduinoespressif32\cores\esp32/esp32-hal-i2c.c:1629

atanisoft, 2021-11-24:
Do you have a smaller reproducible test case other than the very complex ELRS project? If I was to guess you have multiple ISRs bound to the same core (SPI, I2C and HWTimer) all competing for exclusive access to the same cpu core. 

What specific build are you using from PlatformIO?

Can you also post the full backtrace as there may be more clues in the output which you have omitted.

VojtechBartoska, 2022-04-07:
Hello, I'm closing this as expired as there is missing info and issue is quite old. If needed, please open new issue. Thanks

closed 2022-04-07 by VojtechBartoska