espressif/arduino-esp32 #5925: undefined reference to `MD5Update' in arduino as component

qt1, 2021-11-24:
The compilation of a project referencing MD5Builder fails link with undefined reference to `MD5Update' 

Creating a minimal project with this error:

1. Latest ESP-IDF, install.sh and export.sh
2. Create new project using 

        idf.py create-project proj1
        cd proj1

At this point the project compiles.

3. Add arduino-esp32 as in the [docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html)

        mkdir -p components && \
        cd components && \
        git clone https://github.com/espressif/arduino-esp32.git arduino && \
        cd arduino && \
        git submodule update --init --recursive && \
        cd ../.. && \
        idf.py menuconfig

4. Rename 'proj1.c' to 'proj1.cpp' and change in 'CMakeList.txt', content from docs. At this point the project still compiles

5. Add reference to 'MD5Builder.h' like so: (based on the example in the docs)
        
        //file: proj1.cpp
        #include "Arduino.h"
        
        #include "MD5Builder.h"
        
        void setup(){
            Serial.begin(115200);
        
            MD5Builder hasher;
            hasher.add("hello");
            Serial.println( hasher.toString());
        }
        
        void loop(){
            Serial.println("loop");
            delay(1000);
        }
        
        extern "C" {
        void app_main(void)
        {
            setup();
            while(1)
                loop();
        }
        }

Now the compilation fails with linker error:

        ../components/arduino/cores/esp32/MD5Builder.cpp:37: undefined reference to `MD5Update'
        collect2: error: ld returned 1 exit status
        ninja: build stopped: subcommand failed.
        ninja failed with exit code 1

Please advise.

atanisoft, 2021-11-24:
arduino-esp32 master (assuming that is what you are using since you didn't specify) only supports ESP-IDF v4.4 and not earlier. You state "Latest ESP-IDF" but that doesn't indicate what version you are using.

qt1, 2021-11-24:
This happens with multiple matching versions of esp-idf and arduino-esp32.

Specifically the latest versions are:
components/arduino   399f4ecbb3a4cef21e2bffa37adb6190356dfb76 
esp-idf                         ddc44956bf718540d5451e17e1becf6c7dffe5b8

MD5Update is a C function that may also be available in the ROM, but I don't know.

I suspect that there is a linker setting or some flag that is needed in the the CMakeList.txt', or maybe some additional flag in some configuration file.

I would appreciate if someone that knows the system could look and help with this.

Thanks

atanisoft, 2021-11-24:
which specific ESP32 target are you using via idf.py?

qt1, 2021-11-25:
The target is esp32 (default). Same result when the target explicitly using `idf.py set-target esp32` .

atanisoft, 2021-11-28:
@qt1 Try disabling `CONFIG_WPA_MBEDTLS_CRYPTO` in your sdkconfig. It looks like when this is enabled there is no actual implementation compiled of the various MD5 functions that are used currently by `MD5Builder`.

qt1, 2021-11-28:
According to some other issues referencing  MD5Update

Adding the following to `esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld ` seems to solve the problem.

        PROVIDE ( MD5Final = 0x4005db1c );
        PROVIDE ( MD5Init = 0x4005da7c );
        PROVIDE ( MD5Update = 0x4005da9c );

I is possible that theses where left out for a reason, so I m not sure if I should PR esp-idf or not..

Setting `CONFIG_WPA_MBEDTLS_CRYPTO=n` also solves the linker error.  (I guess the first option uses ROM implementation and the other uses compiled version)

Thanks!

atanisoft, 2021-11-28:
Those entries are intentionally omitted from the ld scripts as they used to be provided by the WPA code, as part of migrating to ESP-IDF v4.4 though this became optionally compiled based on `CONFIG_WPA_MBEDTLS_CRYPTO`. The PR I've created migrates to the underlying ROM APIs which are available on all targets.

closed 2021-12-14 by me-no-dev