espressif/arduino-esp32 #5926: esp_eth_set_mac does nothing. Can't change ESP's MAC Address

GeorgeFlorian, 2021-11-24:
### Hardware:
Board:							ESP32 EVB
IDE name:						Platform.io
PSRAM enabled:                                    no
Computer OS:                                        Windows 10

### Description:
Hello.

I'm trying to set the MAC of the ETH interface the same as the WiFi interface using the following code:
```
    ETH.begin();

    uint8_t mac[] = {};
    esp_wifi_get_mac(WIFI_IF_STA, &mac[0]);
    esp_eth_set_mac(&mac[0]);
    Serial.print("ETH MAC: ");
    Serial.println(ETH.macAddress());
```
This prints the right MAC, which is the same as the WiFi's MAC (7c-9e-bd-30-2d-a8), but using arp -a in cmd or looking in the router's device list gives me the original ETH MAC: 7c-9e-bd-30-2d-ab

How do I properly change the ETH MAC Address ?

I have an ESP32-EVB with the following MACs:
WiFi Mac: 7c-9e-bd-30-2d-a8
ETH Mac: 7c-9e-bd-30-2d-ab

GeorgeFlorian, 2021-12-05:
Sooo ? Should it work or not ?

smuellener, 2022-03-21:
@GeorgeFlorian Have you found a solution for this? Thank you

smuellener, 2022-03-21:
Possible fix: https://github.com/espressif/arduino-esp32/issues/6458

VojtechBartoska, 2022-03-23:
Thanks for linking this @smuellener. Closing this one as duplicate and @GeorgeFlorian please follow mentioned issue above as there is more extensive description and also PR. If it will be needed, you can reopen this issue.