espressif/arduino-esp32 #5927: arduino libraries as component of esp-idf build error: (IDFGH-5509)

Gokula-Krishnan-Sekar, 2021-11-25:
i just trying add Arduino libraries as component of esp-idf  but its shows error only. i tried several methods but nothing make sense. i tried with V4.0 (release) & V4.3 (master) 

Generated /root/Desktop/esp_aruindo/build/bootloader/bootloader.bin
[93/93] cd /root/Desktop/esp_aruindo/build/bootloader/esp-idf/esptool_py && /root/.espressif/python_env/idf4.4_py3.9_env/bin/python /root/esp2/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /root/Desktop/esp_aruindo/build/bootloader/bootloader.bin
Bootloader binary size 0x60f0 bytes. 0xf10 bytes (16%) free.
[980/1092] Building C object esp-idf/...o.dir/cores/esp32/esp32-hal-adc.c.obj
../components/arduino/cores/esp32/esp32-hal-adc.c: In function '__analogReadMilliVolts':
../components/arduino/cores/esp32/esp32-hal-adc.c:182:17: warning: 'adc2_vref_to_gpio' is deprecated [-Wdeprecated-declarations]
                 if(adc2_vref_to_gpio(__analogVRefPin) == ESP_OK){
                 ^~
In file included from /root/esp2/esp-idf/components/driver/esp32/include/driver/adc.h:9,
                 from ../components/arduino/cores/esp32/esp32-hal-adc.c:23:
/root/esp2/esp-idf/components/driver/include/driver/adc_common.h:446:11: note: declared here
 esp_err_t adc2_vref_to_gpio(gpio_num_t gpio) __attribute__((deprecated));
           ^~~~~~~~~~~~~~~~~
[983/1092] Building C object esp-idf/...o.dir/cores/esp32/esp32-hal-cpu.c.obj
FAILED: esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-cpu.c.obj 
/root/.espressif/tools/xtensa-esp32-elf/esp-2021r1-8.4.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -Iconfig -I../components/arduino/variants/esp32 -I../components/arduino/cores/esp32 -I../components/arduino/libraries/ArduinoOTA/src -I../components/arduino/libraries/AsyncUDP/src -I../components/arduino/libraries/BLE/src -I../components/arduino/libraries/BluetoothSerial/src -I../components/arduino/libraries/DNSServer/src -I../components/arduino/libraries/EEPROM/src -I../components/arduino/libraries/ESP32/src -I../components/arduino/libraries/ESPmDNS/src -I../components/arduino/libraries/FFat/src -I../components/arduino/libraries/FS/src -I../components/arduino/libraries/HTTPClient/src -I../components/arduino/libraries/HTTPUpdate/src -I../components/arduino/libraries/NetBIOS/src -I../components/arduino/libraries/Preferences/src -I../components/arduino/libraries/SD_MMC/src -I../components/arduino/libraries/SD/src -I../components/arduino/libraries/SimpleBLE/src -I../components/arduino/libraries/SPIFFS/src -I../components/arduino/libraries/SPI/src -I../components/arduino/libraries/Ticker/src -I../components/arduino/libraries/Update/src -I../components/arduino/libraries/WebServer/src -I../components/arduino/libraries/WiFiClientSecure/src -I../components/arduino/libraries/WiFi/src -I../components/arduino/libraries/Wire/src -I../components/arduino/cores/esp32/libb64 -I/root/esp2/esp-idf/components/newlib/platform_include -I/root/esp2/esp-idf/components/freertos/include -I/root/esp2/esp-idf/components/freertos/port/xtensa/include -I/root/esp2/esp-idf/components/esp_hw_support/include -I/root/esp2/esp-idf/components/esp_hw_support/include/soc -I/root/esp2/esp-idf/components/esp_hw_support/port/esp32/. -I/root/esp2/esp-idf/components/heap/include -I/root/esp2/esp-idf/components/log/include -I/root/esp2/esp-idf/components/lwip/include/apps -I/root/esp2/esp-idf/components/lwip/include/apps/sntp -I/root/esp2/esp-idf/components/lwip/lwip/src/include -I/root/esp2/esp-idf/components/lwip/port/esp32/include -I/root/esp2/esp-idf/components/lwip/port/esp32/include/arch -I/root/esp2/esp-idf/components/soc/include -I/root/esp2/esp-idf/components/soc/esp32/. -I/root/esp2/esp-idf/components/soc/esp32/include -I/root/esp2/esp-idf/components/hal/esp32/include -I/root/esp2/esp-idf/components/hal/include -I/root/esp2/esp-idf/components/hal/platform_port/include -I/root/esp2/esp-idf/components/esp_rom/include -I/root/esp2/esp-idf/components/esp_rom/esp32 -I/root/esp2/esp-idf/components/esp_rom/include/esp32 -I/root/esp2/esp-idf/components/esp_common/include -I/root/esp2/esp-idf/components/esp_system/include -I/root/esp2/esp-idf/components/esp_system/port/soc -I/root/esp2/esp-idf/components/esp_system/port/public_compat -I/root/esp2/esp-idf/components/esp32/include -I/root/esp2/esp-idf/components/driver/include -I/root/esp2/esp-idf/components/driver/esp32/include -I/root/esp2/esp-idf/components/esp_pm/include -I/root/esp2/esp-idf/components/esp_ringbuf/include -I/root/esp2/esp-idf/components/efuse/include -I/root/esp2/esp-idf/components/efuse/esp32/include -I/root/esp2/esp-idf/components/xtensa/include -I/root/esp2/esp-idf/components/xtensa/esp32/include -I/root/esp2/esp-idf/components/vfs/include -I/root/esp2/esp-idf/components/esp_wifi/include -I/root/esp2/esp-idf/components/esp_event/include -I/root/esp2/esp-idf/components/esp_netif/include -I/root/esp2/esp-idf/components/esp_eth/include -I/root/esp2/esp-idf/components/tcpip_adapter/include -I/root/esp2/esp-idf/components/esp_phy/include -I/root/esp2/esp-idf/components/esp_phy/esp32/include -I/root/esp2/esp-idf/components/app_trace/include -I/root/esp2/esp-idf/components/esp_timer/include -I/root/esp2/esp-idf/components/spi_flash/include -I/root/esp2/esp-idf/components/mbedtls/port/include -I/root/esp2/esp-idf/components/mbedtls/mbedtls/include -I/root/esp2/esp-idf/components/mbedtls/esp_crt_bundle/include -I/root/esp2/esp-idf/components/mdns/include -I/root/esp2/esp-idf/components/console -I/root/esp2/esp-idf/components/esp_adc_cal/include -I/root/esp2/esp-idf/components/fatfs/diskio -I/root/esp2/esp-idf/components/fatfs/vfs -I/root/esp2/esp-idf/components/fatfs/src -I/root/esp2/esp-idf/components/wear_levelling/include -I/root/esp2/esp-idf/components/sdmmc/include -I/root/esp2/esp-idf/components/nvs_flash/include -I/root/esp2/esp-idf/components/app_update/include -I/root/esp2/esp-idf/components/bootloader_support/include -I/root/esp2/esp-idf/components/spiffs/include -I/root/esp2/esp-idf/components/openssl/include -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fmacro-prefix-map=/root/Desktop/esp_aruindo=. -fmacro-prefix-map=/root/esp2/esp-idf=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.4-dev-1849-g8e3e65a47\" -DESP_PLATFORM -MD -MT esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-cpu.c.obj -MF esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-cpu.c.obj.d -o esp-idf/arduino/CMakeFiles/__idf_arduino.dir/cores/esp32/esp32-hal-cpu.c.obj -c ../components/arduino/cores/esp32/esp32-hal-cpu.c
In file included from ../components/arduino/cores/esp32/esp32-hal-cpu.c:22:
/root/esp2/esp-idf/components/soc/esp32/include/soc/rtc.h:22:14: error: expected identifier or '(' before numeric constant
 #define MHZ (1000000)
              ^~~~~~~
../components/arduino/cores/esp32/esp32-hal-cpu.c:47:16: note: in expansion of macro 'MHZ'
 const uint32_t MHZ = 1000000;
                ^~~
[988/1092] Building C object esp-idf/...lake2b/ref/blake2b-compress-ref.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

in main.cpp file

#include "Arduino.h"

// To enable CPP support
extern "C" void app_main()
{
	initArduino();
	Serial.begin(115200);
	while(true){
		// Similar arduino loop
		Serial.println("Test");
		delay(1000);
	}
}

VojtechBartoska, 2021-11-25:
Please take a look on this updated [documentation](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html). Current Arduino core version is compatible with [ESP-IDFv4.4 branch](https://github.com/espressif/esp-idf/tree/release/v4.4).

VojtechBartoska, 2022-04-07:
Closing as expired.

closed 2022-04-07 by VojtechBartoska