espressif/arduino-esp32 #5928: No timeout on start_ssl_client(): Performing the SSL/TLS handshake... ESP occasionnaly hangs indefinitely

Ben79543, 2021-11-25:
I have a large program, that at some points connects to AWS using mqtt PubSubClient.
It works 99,9% of the time, but once in a while the ESP32 just hangs while performing the SSL/TLS handshake. 
My issue is not to understand why it fails as it is so rare, but to understand why there is no timeout that I could then handle properly. 
As it is, my device becomes a brick until manual reset. 

### Hardware:
Board:							ESP32_WROOM_32
Core Installation version:			2.0.0
IDE name:						Arduino IDE (with vscode)
Flash Frequency:					80Mhz
PSRAM enabled:                                     no
Upload Speed:					512000
Computer OS:                                        Windows 10

This is the end of the log. After that nothing happens, while it should continue the connection and data transfer.
### Debug Messages:
```
11:44:07.862 -> [  8025][D][OB1-121-AR.ino:831] connectToNetwork_admin(): ********** CONNECTED & GOTIP
11:44:07.862 -> [  8025][D][OB1-121-AR.ino:759] ConnectToMQTT(): **********mqtt status: -1
11:44:07.862 -> [  8028][D][OB1-121-AR.ino:762] ConnectToMQTT(): **********reconnecting mqtt client 840D8EE1F494 to : xxxxxxxxxxx.amazonaws.com
11:44:07.862 -> [  8041][V][ssl_client.cpp:59] start_ssl_client(): Free internal heap before TLS 131403
11:44:07.909 -> [  8049][V][ssl_client.cpp:65] start_ssl_client(): Starting socket
11:44:07.955 -> [  8121][V][ssl_client.cpp:141] start_ssl_client(): Seeding the random number generator
11:44:07.955 -> [  8124][V][ssl_client.cpp:150] start_ssl_client(): Setting up the SSL/TLS structure...
11:44:07.955 -> [  8127][V][ssl_client.cpp:166] start_ssl_client(): Loading CA cert
11:44:08.097 -> [  8251][V][ssl_client.cpp:215] start_ssl_client(): Loading CRT cert
11:44:08.190 -> [  8369][V][ssl_client.cpp:224] start_ssl_client(): Loading private key
11:44:08.381 -> [  8532][V][ssl_client.cpp:234] start_ssl_client(): Setting hostname for TLS session...
11:44:08.381 -> [  8533][V][ssl_client.cpp:249] start_ssl_client(): Performing the SSL/TLS handshake...
```

This occur on a simple connection attempt:
(only code extract as I have thousands of lines)
### Sketch:  
```cpp

if (connectToNetwork_admin())
        {
          if (ConnectToMQTT())
          {
            transmitBuffer();
            SynchAndWait();
          }
        }

boolean ConnectToMQTT()
{
  unsigned long mqttTimestamp = millis();
  if (WiFi.status() != WL_CONNECTED)
  {
    ESP_LOGE(TAG, "********** NOT CONNECTED TO WIFI, CAN'T CONNECT TO MQTT");
    return false;
  }
  ESP_LOGD(TAG, "**********mqtt status: %d", mqttClient.state());
  while (!!!mqttClient.connected())
  {
    ESP_LOGD(TAG, "**********reconnecting mqtt client %s to : %s", device_id, mqttServer);
    if (mqttClient.connect(device_id))
    {
      ESP_LOGD(TAG, "********** MQTT CONNECTED - STATUS %d", mqttClient.state());
      if (!mqttClient.subscribe(SHADOW_UPDATE_RES, 0))
      {
        ESP_LOGE(TAG, "********** SUBSCRIBE TO SYNCH FAILED");
        return false;
      }
      else
      {
        ESP_LOGD(TAG, "**********subscribe to synch succeeded");
        return true;
      }
    }
    else
    {
      delay(1000); //Delay between 2 attempts
      if ((millis() - mqttTimestamp) > MQTT_TIMEOUT)
      {
        ESP_LOGE(TAG, "********** CONNECTION TO MQTT TIMED OUT - STATUS %d - clientID %s", mqttClient.state(), device_id);
        return false;
      }
    }
  }
  return true;
}
```

VojtechBartoska, 2022-04-08:
Hello @Ben79543, sorry for getting to you late. Is this still valid? I see you run your sketch on v2.0.0 which is now a bit outdated, will you be able to retest this on v2.0.3-rc1?

VojtechBartoska, 2022-04-22:
Hello, due to no answer I'm closing this issue as expired.

If you still need help, please reopen it.

Thanks for understanding!

closed 2022-04-22 by VojtechBartoska

shaneSAT, 2022-05-23:
I'm also having a problem without a timeout on start_ssl_client() and my ESP occasionally hanging. I've been using HTTPClient and posting to a https address without a problem the past 2 years - until recently when our local internet connection started having intermittent delays. The only way I can replicate this is by disconnecting the internet during "start_ssl_client(): Performing the SSL/TLS handshake...". Disconnecting before or after "start_ssl_client()" is not a problem, the  connection aborts as expected.

shaneSAT, 2022-05-23:
Further info: Am using a Esp32 DevKit v1, have tried various Arduino IDE versions with multiple upload configurations, and am using standard HTTPClient POST sketch found in most examples. I have also tried various setConnectTimeout and setTimout features to no avail. The question - is there a Timeout feature for the SSL/TLS handshake, and if so, how can I enable it? If not, is there any code I can modify to solve the problem, or an alternate method of https POST that will work on a flaky internet connection?

shaneSAT, 2022-05-23:
Some more info: If using GET instead of POST I can disrupt the connection at any time during, before, and after the ""start_ssl_client()" process and the connection aborts with Error code: -1. So somewhere there is a Timeout feature on the GET using the same sketch parameters, just not on the POST

XiloXL, 2022-06-19:
I am experiencing the same issues.
Posting data to an API every 30 seconds works most of the time, but once in 1-2 day's it will hang on the handshake, and only way to recover is to reboot the ESP. 
Anybody got a clue what is going on?

shaneSAT, 2022-06-19:
Seems the handshake timeout was fixed a few weeks ago. I am now performing httpPost without ssl timeout issues using the BasicHTTPSClient.ino from the Arduino  IDE examples.
I think the default timeout is set at 2 minutes, if you need a shorter time then include “client -> setHandshakeTimeout(20);” for a 20 second timeout.

XiloXL, 2022-06-19:
Can you tell me in which version of the esp32 library the fix is implemented? I am running 2.0.3 now.

shaneSAT, 2022-06-19:
I'm using 2.0.3 in Arduino 1.8.19