espressif/arduino-esp32 #5929: IDF (Arduino) OTA problem

marcelwinter04, 2021-11-25:
I'm trying to get the OTA SD Card example working. This thing worked with the Arduino IDE unfortunately there is no option to change the menuconfig with the arduino ide. So know im trying the same example with the esp idf but with the arduino component. 

```#include "Arduino.h"
#include <FS.h>
#include <SD_MMC.h>
#include <Update.h>

void rebootEspWithReason(String reason){
    Serial.println(reason);
    delay(1000);
    ESP.restart();
}

void performUpdate(Stream &updateSource, size_t updateSize) {
   if (Update.begin(updateSize)) {      
      size_t written = Update.writeStream(updateSource);
      if (written == updateSize) {
         Serial.println("Written : " + String(written) + " successfully");
      }
      else {
         Serial.println("Written only : " + String(written) + "/" + String(updateSize) + ". Retry?");
      }
      if (Update.end()) {
         Serial.println("OTA done!");
         if (Update.isFinished()) {
            Serial.println(Update.hasError());
            Serial.println(Update.isRunning());
            Serial.println(Update.rollBack());
            rebootEspWithReason("Update successfully completed. Rebooting.");
         }  
         else {
            Serial.println("Update not finished? Something went wrong!");
         }
      }
      else {
         Serial.println("Error Occurred. Error #: " + String(Update.getError()));
      }

   }
   else
   {
      Serial.println("Not enough space to begin OTA");
   }
}

void updateFromFS(fs::FS &fs) {
   File updateBin = fs.open("/update.bin");
   if (updateBin) {
      if(updateBin.isDirectory()){
         Serial.println("Error, update.bin is not a file");
         updateBin.close();
         return;
      }

      size_t updateSize = updateBin.size();

      if (updateSize > 0) {
         Serial.println("Try to start update");
         performUpdate(updateBin, updateSize);
      }
      else {
         Serial.println("Error, file is empty");
      }

      updateBin.close();
    
      // whe finished remove the binary from sd card to indicate end of the process
     // fs.remove("/update.bin");      
   }
   else {
      Serial.println("Could not load update.bin from sd root");
   }
}

extern "C" void app_main()
{
    initArduino();
    pinMode(4, OUTPUT);
    digitalWrite(4, HIGH);
    
	Serial.begin(115200);
	
    if(!SD_MMC.begin()){
        Serial.println("Card Mount Failed");
        return;
    }
    uint8_t cardType = SD_MMC.cardType();

    if(cardType == CARD_NONE){
        Serial.println("No SD_MMC card attached");
        return;
    }

    Serial.print("SD_MMC Card Type: ");
    if(cardType == CARD_MMC){
        Serial.println("MMC");
    } else if(cardType == CARD_SD){
        Serial.println("SDSC");
    } else if(cardType == CARD_SDHC){
        Serial.println("SDHC");
    } else {
        Serial.println("UNKNOWN");
    }

    uint64_t cardSize = SD_MMC.cardSize() / (1024 * 1024);
    Serial.printf("SD_MMC Card Size: %lluMB\n", cardSize);

    updateFromFS(SD_MMC);

    while (1) {
        vTaskDelay(1);
    }
}
```
But when i build this project i get following error message 

```
FAILED: blink.elf
cmd.exe /C "cd . && C:\Users\MW\Documents\.espressif\tools\xtensa-esp32-elf\esp-2021r2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address  @CMakeFiles\blink.elf.rsp -o blink.elf  && cd ."
c:/users/mw/documents/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/arduino/libarduino.a(MD5Builder.cpp.obj):(.literal._ZN10MD5Builder5beginEv+0x0): undefined reference to `MD5Init'
c:/users/mw/documents/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/arduino/libarduino.a(MD5Builder.cpp.obj):(.literal._ZN10MD5Builder3addEPht+0x0): undefined reference to `MD5Update'
c:/users/mw/documents/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/arduino/libarduino.a(MD5Builder.cpp.obj):(.literal._ZN10MD5Builder9calculateEv+0x0): undefined reference to `MD5Final'
c:/users/mw/documents/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/arduino/libarduino.a(MD5Builder.cpp.obj): in function `MD5Builder::begin()':
c:\users\mw\documents\projects\blink\build/../components/arduino/cores/esp32/MD5Builder.cpp:32: undefined reference to `MD5Init'
c:/users/mw/documents/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/arduino/libarduino.a(MD5Builder.cpp.obj): in function `MD5Builder::add(unsigned char*, unsigned short)':
c:\users\mw\documents\projects\blink\build/../components/arduino/cores/esp32/MD5Builder.cpp:37: undefined reference to `MD5Update'
c:/users/mw/documents/.espressif/tools/xtensa-esp32-elf/esp-2021r2-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/arduino/libarduino.a(MD5Builder.cpp.obj): in function `MD5Builder::calculate()':
c:\users\mw\documents\projects\blink\build/../components/arduino/cores/esp32/MD5Builder.cpp:97: undefined reference to `MD5Final'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```
Does anybody knows what the problem may be? 

ESP-IDF: 4.4

marcelwinter04, 2021-12-01:
#5941

closed 2021-12-01 by marcelwinter04

KARTIK-eng, 2022-02-21:
> 

@marcelwinter04 How did u get it working on the arduino IDE? I'm getting a core panicked error.
Using https://github.com/espressif/arduino-esp32/blob/master/libraries/Update/examples/SD_Update/SD_Update.ino example.
ERROR :
Welcome to the SD-Update example!
12:02:11.604 -> Try to start update
12:02:11.652 -> abort() was called at PC 0x400efcd5 on core 1
12:02:11.652 -> 
12:02:11.652 -> ELF file SHA256: 0000000000000000
12:02:11.652 -> 
12:02:11.652 -> Backtrace: 0x40085184:0x3ffb1df0 0x400853f9:0x3ffb1e10 0x400efcd5:0x3ffb1e30 0x400843ae:0x3ffb1e50 0x400efb89:0x3ffb1e70 0x400d4571:0x3ffb1e90 0x400d1799:0x3ffb1eb0 0x400d19c7:0x3ffb1ed0 0x400d11b3:0x3ffb1ef0 0x400d1407:0x3ffb1f40 0x400d14b2:0x3ffb1f80 0x400d50c6:0x3ffb1fb0 0x40086409:0x3ffb1fd0

Could you please help me figure what's going wrong?

marcelwinter04, 2022-02-22:
@KARTIK-eng on the arduino ide it worked without doing anything.

KARTIK-eng, 2022-02-23:
> @KARTIK-eng on the arduino ide it worked without doing anything.

Which board are you using?