espressif/arduino-esp32 #5930: Dedicated MQTT Arduino library from ESP-MQTT

hutje, 2021-11-25:
### Description:
There is no dedicated MQTT Library for Arduino that is implemented from ESP-MQTT.
This library would be extremely useful to improve the stability of the connection. Now the MQTT function is dependent on third-party libraries and cause connection errors.

It would be a great option to implement the ESP-IDF ESP-MQTT library in Arduino.

bertmelis, 2021-11-26:
The Arduino core for the two platforms are built on different toolchains.
ESP-MQTT is out-of-the-box usable on ESP32.

hutje, 2021-11-26:
> The Arduino core for the two platforms are built on different toolchains. ESP-MQTT is out-of-the-box usable on ESP32.

That I understand, but that would make an Arduino library of ESP-MQTT from the ESP-IDF so much more stable.

me-no-dev, 2021-11-26:
There is no reason to say that Arduino MQTT libraries are less stable than IDF's MQTT lib. MQTT is not rocket science and you do have access to IDF's MQTT through Arduino. It is just not wrapped into a class.

closed 2021-11-26 by me-no-dev