espressif/arduino-esp32 #5933: Boards.txt is missing ttgo-lora32-v2 profile although listed in variants folder

tobozo, 2021-11-27:
The entry `ttgo-lora32-v21new` in `boards.txt` has many duplicate properties.

Those duplicate properties are probably a result of a script trying to produce variants choice in the boards menu, and apparently failing.

Current symptom: only `ttgo-lora32-v21new` is available from the boards menu, no trace of `ttgo-lora32-v2`.

The expected result is having either two separate board choices, or two variant choices in the boards menu for those board variants:

- `ttgo-lora32-v21new`
- `ttgo-lora32-v2`

```ini
ttgo-lora32-v21new.name=TTGO LoRa32-OLED v2.1.6

ttgo-lora32-v21new.upload.tool=esptool_py
ttgo-lora32-v21new.upload.maximum_size=1310720
ttgo-lora32-v21new.upload.maximum_data_size=294912
ttgo-lora32-v21new.upload.wait_for_upload_port=true
ttgo-lora32-v21new.upload.flags=
ttgo-lora32-v21new.upload.extra_flags=

ttgo-lora32-v21new.serial.disableDTR=true
ttgo-lora32-v21new.serial.disableRTS=true

ttgo-lora32-v21new.build.tarch=xtensa
ttgo-lora32-v21new.build.bootloader_addr=0x1000
ttgo-lora32-v21new.build.target=esp32
ttgo-lora32-v21new.build.mcu=esp32
ttgo-lora32-v21new.build.core=esp32
ttgo-lora32-v21new.build.variant=ttgo-lora32-v21new
ttgo-lora32-v21new.build.board=TTGO_LoRa32_v21new

ttgo-lora32-v21new.build.f_cpu=240000000L
ttgo-lora32-v21new.build.flash_mode=dio
ttgo-lora32-v21new.build.flash_size=4MB
ttgo-lora32-v21new.build.boot=dio
ttgo-lora32-v21new.build.partitions=default

ttgo-lora32-v21new.menu.FlashFreq.80=80MHz
ttgo-lora32-v21new.menu.FlashFreq.80.build.flash_freq=80m
ttgo-lora32-v21new.menu.FlashFreq.40=40MHz
ttgo-lora32-v21new.menu.FlashFreq.40.build.flash_freq=40m

ttgo-lora32-v21new.menu.UploadSpeed.921600=921600
ttgo-lora32-v21new.menu.UploadSpeed.921600.upload.speed=921600
ttgo-lora32-v21new.menu.UploadSpeed.115200=115200
ttgo-lora32-v21new.name=TTGO LoRa32-OLED v2.1.6

ttgo-lora32-v21new.upload.tool=esptool_py
ttgo-lora32-v21new.upload.maximum_size=1310720
ttgo-lora32-v21new.upload.maximum_data_size=294912
ttgo-lora32-v21new.upload.wait_for_upload_port=true

ttgo-lora32-v21new.serial.disableDTR=true
ttgo-lora32-v21new.serial.disableRTS=true

ttgo-lora32-v21new.build.tarch=xtensa
ttgo-lora32-v21new.build.bootloader_addr=0x1000
ttgo-lora32-v21new.build.target=esp32
ttgo-lora32-v21new.build.mcu=esp32
ttgo-lora32-v21new.build.core=esp32
ttgo-lora32-v21new.build.variant=ttgo-lora32-v21new
ttgo-lora32-v21new.build.board=TTGO_LoRa32_v21new

ttgo-lora32-v21new.build.f_cpu=240000000L
ttgo-lora32-v21new.build.flash_mode=dio
ttgo-lora32-v21new.build.flash_size=4MB
ttgo-lora32-v21new.build.boot=dio
ttgo-lora32-v21new.build.partitions=default

ttgo-lora32-v21new.menu.FlashFreq.80=80MHz
ttgo-lora32-v21new.menu.FlashFreq.80.build.flash_freq=80m
ttgo-lora32-v21new.menu.FlashFreq.40=40MHz
ttgo-lora32-v21new.menu.FlashFreq.40.build.flash_freq=40m

ttgo-lora32-v21new.menu.UploadSpeed.921600=921600
ttgo-lora32-v21new.menu.UploadSpeed.921600.upload.speed=921600
ttgo-lora32-v21new.menu.UploadSpeed.115200=115200
ttgo-lora32-v21new.menu.UploadSpeed.115200.upload.speed=115200
ttgo-lora32-v21new.menu.UploadSpeed.256000.windows=256000
ttgo-lora32-v21new.menu.UploadSpeed.256000.upload.speed=256000
ttgo-lora32-v21new.menu.UploadSpeed.230400.windows.upload.speed=256000
ttgo-lora32-v21new.menu.UploadSpeed.230400=230400
ttgo-lora32-v21new.menu.UploadSpeed.230400.upload.speed=230400
ttgo-lora32-v21new.menu.UploadSpeed.460800.linux=460800
ttgo-lora32-v21new.menu.UploadSpeed.460800.macosx=460800
ttgo-lora32-v21new.menu.UploadSpeed.460800.upload.speed=460800
ttgo-lora32-v21new.menu.UploadSpeed.512000.windows=512000
ttgo-lora32-v21new.menu.UploadSpeed.512000.upload.speed=512000

ttgo-lora32-v21new.menu.DebugLevel.none=None
ttgo-lora32-v21new.menu.DebugLevel.none.build.code_debug=0
ttgo-lora32-v21new.menu.DebugLevel.error=Error
ttgo-lora32-v21new.menu.DebugLevel.error.build.code_debug=1
ttgo-lora32-v21new.menu.DebugLevel.warn=Warn
ttgo-lora32-v21new.menu.DebugLevel.warn.build.code_debug=2
ttgo-lora32-v21new.menu.DebugLevel.info=Info
ttgo-lora32-v21new.menu.DebugLevel.info.build.code_debug=3
ttgo-lora32-v21new.menu.DebugLevel.debug=Debug
ttgo-lora32-v21new.menu.DebugLevel.debug.build.code_debug=4
ttgo-lora32-v21new.menu.DebugLevel.verbose=Verbose
ttgo-lora32-v21new.menu.DebugLevel.verbose.build.code_debug=5w.menu.UploadSpeed.115200.upload.speed=115200
ttgo-lora32-v21new.menu.UploadSpeed.256000.windows=256000
ttgo-lora32-v21new.menu.UploadSpeed.256000.upload.speed=256000
ttgo-lora32-v21new.menu.UploadSpeed.230400.windows.upload.speed=256000
ttgo-lora32-v21new.menu.UploadSpeed.230400=230400
ttgo-lora32-v21new.menu.UploadSpeed.230400.upload.speed=230400
ttgo-lora32-v21new.menu.UploadSpeed.460800.linux=460800
ttgo-lora32-v21new.menu.UploadSpeed.460800.macosx=460800
ttgo-lora32-v21new.menu.UploadSpeed.460800.upload.speed=460800
ttgo-lora32-v21new.menu.UploadSpeed.512000.windows=512000
ttgo-lora32-v21new.menu.UploadSpeed.512000.upload.speed=512000

ttgo-lora32-v21new.menu.DebugLevel.none=None
ttgo-lora32-v21new.menu.DebugLevel.none.build.code_debug=0
ttgo-lora32-v21new.menu.DebugLevel.error=Error
ttgo-lora32-v21new.menu.DebugLevel.error.build.code_debug=1
ttgo-lora32-v21new.menu.DebugLevel.warn=Warn
ttgo-lora32-v21new.menu.DebugLevel.warn.build.code_debug=2
ttgo-lora32-v21new.menu.DebugLevel.info=Info
ttgo-lora32-v21new.menu.DebugLevel.info.build.code_debug=3
ttgo-lora32-v21new.menu.DebugLevel.debug=Debug
ttgo-lora32-v21new.menu.DebugLevel.debug.build.code_debug=4
ttgo-lora32-v21new.menu.DebugLevel.verbose=Verbose
ttgo-lora32-v21new.menu.DebugLevel.verbose.build.code_debug=5
```

atanisoft, 2021-11-28:
https://github.com/espressif/arduino-esp32/commit/8869d39d7997eb57177f1cd95f32fd3c26300d13 added the variant but never created an entry in boards.txt.

https://github.com/espressif/arduino-esp32/commit/ccab428e4dc67e47a2db22ef66d983164a21e7fc created the ttgo-lora32-v21new variant along with the duplicated entries, which might have been a copy/paste error.

Feel free to submit a PR to fix this.

tobozo, 2021-11-28:
@atanisoft thanks for your feedback.

I've ordered a unit and will do a few tests as soon as it's delivered (hopefully next wednesday).

Does it make sense to rename ttgo-lora32-v21new to ttgo-lora32-v2 and create "v1", "v2", "v2.1" menu variants as it is done in TTGO-TWatch menu ? 

[image]

ping @LilyGO

atanisoft, 2021-11-28:
@tobozo I think it does make sense for visual purposes but likely the entry keys should remain as-is.

tobozo, 2021-12-01:
I've received my TTGO-Lora32-v2 board and was able to make some tests.

[image]

As an extra I've copied the LoRa Region and Debug level selectors from the heltec32 board options, it can't hurt to have such option anyway :-)

In the proposed changes, `TTGO_LoRa32_V1` and `TTGO_LoRa32_v21new` distinct board entries have been removed and merged with the initially missing `TTGO_LoRa32_V2` board entry into a new shared board entry named `TTGO LoRa32-OLED`.

It was quite easy to decide since they all share common menu options, and all necessary pins_arduino.h versions already exist.

Please confirm I can make a PR with that.

```
ttgo-lora32.name=TTGO LoRa32-OLED

ttgo-lora32.upload.tool=esptool_py
ttgo-lora32.upload.maximum_size=1310720
ttgo-lora32.upload.maximum_data_size=294912
ttgo-lora32.upload.flags=
ttgo-lora32.upload.extra_flags=

ttgo-lora32.serial.disableDTR=true
ttgo-lora32.serial.disableRTS=true

ttgo-lora32.build.tarch=xtensa
ttgo-lora32.build.bootloader_addr=0x1000
ttgo-lora32.build.target=esp32
ttgo-lora32.build.mcu=esp32
ttgo-lora32.build.core=esp32
ttgo-lora32.build.board=TTGO_LoRa32

ttgo-lora32.menu.Revision.TTGO_LoRa32_V1=TTGO LoRa32 V1 (No TFCard)
ttgo-lora32.menu.Revision.TTGO_LoRa32_V1.build.board=TTGO_LoRa32_V1
ttgo-lora32.menu.Revision.TTGO_LoRa32_V1.build.variant=ttgo-lora32-v1

ttgo-lora32.menu.Revision.TTGO_LoRa32_V2=TTGO LoRa32 V2
ttgo-lora32.menu.Revision.TTGO_LoRa32_V2.build.board=TTGO_LoRa32_V2
ttgo-lora32.menu.Revision.TTGO_LoRa32_V2.build.variant=ttgo-lora32-v2

ttgo-lora32.menu.Revision.TTGO_LoRa32_v21new=TTGO LoRa32 V2.1 (1.6.1)
ttgo-lora32.menu.Revision.TTGO_LoRa32_v21new.build.board=TTGO_LoRa32_v21new
ttgo-lora32.menu.Revision.TTGO_LoRa32_v21new.build.variant=ttgo-lora32-v21new

ttgo-lora32.menu.LORAWAN_REGION.0=REGION_EU868
ttgo-lora32.menu.LORAWAN_REGION.0.build.band=REGION_EU868
ttgo-lora32.menu.LORAWAN_REGION.1=REGION_EU433
ttgo-lora32.menu.LORAWAN_REGION.1.build.band=REGION_EU433
ttgo-lora32.menu.LORAWAN_REGION.2=REGION_CN470
ttgo-lora32.menu.LORAWAN_REGION.2.build.band=REGION_CN470
ttgo-lora32.menu.LORAWAN_REGION.3=REGION_US915
ttgo-lora32.menu.LORAWAN_REGION.3.build.band=REGION_US915
ttgo-lora32.menu.LORAWAN_REGION.4=REGION_AU915
ttgo-lora32.menu.LORAWAN_REGION.4.build.band=REGION_AU915
ttgo-lora32.menu.LORAWAN_REGION.5=REGION_CN779
ttgo-lora32.menu.LORAWAN_REGION.5.build.band=REGION_CN779
ttgo-lora32.menu.LORAWAN_REGION.6=REGION_AS923
ttgo-lora32.menu.LORAWAN_REGION.6.build.band=REGION_AS923
ttgo-lora32.menu.LORAWAN_REGION.7=REGION_KR920
ttgo-lora32.menu.LORAWAN_REGION.7.build.band=REGION_KR920
ttgo-lora32.menu.LORAWAN_REGION.8=REGION_IN865
ttgo-lora32.menu.LORAWAN_REGION.8.build.band=REGION_IN865
ttgo-lora32.menu.LORAWAN_REGION.9=REGION_US915_HYBRID
ttgo-lora32.menu.LORAWAN_REGION.9.build.band=REGION_US915_HYBRID

ttgo-lora32.menu.LoRaWanDebugLevel.0=None
ttgo-lora32.menu.LoRaWanDebugLevel.0.build.LoRaWanDebugLevel=0
ttgo-lora32.menu.LoRaWanDebugLevel.1=Freq
ttgo-lora32.menu.LoRaWanDebugLevel.1.build.LoRaWanDebugLevel=1
ttgo-lora32.menu.LoRaWanDebugLevel.2=Freq && DIO
ttgo-lora32.menu.LoRaWanDebugLevel.2.build.LoRaWanDebugLevel=2
ttgo-lora32.menu.LoRaWanDebugLevel.3=Freq && DIO && PW
ttgo-lora32.menu.LoRaWanDebugLevel.3.build.LoRaWanDebugLevel=3

ttgo-lora32.build.f_cpu=240000000L
ttgo-lora32.build.flash_mode=dio
ttgo-lora32.build.flash_size=4MB
ttgo-lora32.build.boot=dio
ttgo-lora32.build.partitions=default

ttgo-lora32.menu.FlashFreq.80=80MHz
ttgo-lora32.menu.FlashFreq.80.build.flash_freq=80m
ttgo-lora32.menu.FlashFreq.40=40MHz
ttgo-lora32.menu.FlashFreq.40.build.flash_freq=40m

ttgo-lora32.menu.UploadSpeed.921600=921600
ttgo-lora32.menu.UploadSpeed.921600.upload.speed=921600
ttgo-lora32.menu.UploadSpeed.115200=115200
ttgo-lora32.menu.UploadSpeed.115200.upload.speed=115200
ttgo-lora32.menu.UploadSpeed.256000.windows=256000
ttgo-lora32.menu.UploadSpeed.256000.upload.speed=256000
ttgo-lora32.menu.UploadSpeed.230400.windows.upload.speed=256000
ttgo-lora32.menu.UploadSpeed.230400=230400
ttgo-lora32.menu.UploadSpeed.230400.upload.speed=230400
ttgo-lora32.menu.UploadSpeed.460800.linux=460800
ttgo-lora32.menu.UploadSpeed.460800.macosx=460800
ttgo-lora32.menu.UploadSpeed.460800.upload.speed=460800
ttgo-lora32.menu.UploadSpeed.512000.windows=512000
ttgo-lora32.menu.UploadSpeed.512000.upload.speed=512000

ttgo-lora32.menu.DebugLevel.none=None
ttgo-lora32.menu.DebugLevel.none.build.code_debug=0
ttgo-lora32.menu.DebugLevel.error=Error
ttgo-lora32.menu.DebugLevel.error.build.code_debug=1
ttgo-lora32.menu.DebugLevel.warn=Warn
ttgo-lora32.menu.DebugLevel.warn.build.code_debug=2
ttgo-lora32.menu.DebugLevel.info=Info
ttgo-lora32.menu.DebugLevel.info.build.code_debug=3
ttgo-lora32.menu.DebugLevel.debug=Debug
ttgo-lora32.menu.DebugLevel.debug.build.code_debug=4
ttgo-lora32.menu.DebugLevel.verbose=Verbose
ttgo-lora32.menu.DebugLevel.verbose.build.code_debug=5

```

atanisoft, 2021-12-01:
I think your proposed update looks fine but I'm not sure how `build.band` or `build.LoRaWanDebugLevel` are used. They likely need special build flags to be added for the code to consume those.

tobozo, 2021-12-01:
oh now I understand where those are [coming from](https://github.com/HelTecAutomation/CubeCell-Arduino/blob/master/platform.txt#L39).

It doesn't make sense to add so many LORA_* options to the compiler flags just for the sakes of a few devices.

I'll remove that from my snippet, should those unusable options also be removed from all heltec profiles ?

atanisoft, 2021-12-01:
Use `build.defines` to add them and they should be usable. But removal is also an option. I don't know about the Heltec profiles but likely they will need some sort of update to pick those up, but I'd suggest track that via another issue

tobozo, 2021-12-01:
okay I'll keep it simple, focus on the ttgo board and just remove the extra flags for the meantime.

tobozo, 2021-12-10:
see PR [#5961](https://github.com/espressif/arduino-esp32/pull/5961)

VojtechBartoska, 2022-04-07:
I'm closing this one as PR was already merged. @tobozo Thanks for your contribution.

closed 2022-04-07 by VojtechBartoska