espressif/arduino-esp32 #5937: Weird serial behavior on Pico kit

Huckies, 2021-11-28:
I'm using a PICO-D4 dev board with code from an aarch64 mac running VScode and the latest extension from Espressif.

The IDF version is 4.4 and arduino component is 2.0.1, if they matter too.
Also I don't think the macOS operation system has any interference, it occurred in Big Sur 11.6 and remain the same after upgraded to Monterey 12.0.1.

It might be a long story so I'll list my requirements in part firstly.

### **UART(Serial):**
Initializing to serial ports, one to receive data from a piece of STM32 and the other to print the debug message.
The STM32 sends message in 64bytes package, one package per second.

The arduino-based ESP32 receive logic is designed to act like this: Check if the local Rx buffer is empty every 10 milliseconds, if not then set a 30-millisecond count down and take the char out of the buffer, a new coming char resets the buffer. Once the count down finishes, a string is forged and handled down.

### **WiFi:**
Connection to the Internet.

### **MQTT:**
Using a PubSubClient library, and subscribing a single topic.

Then it comes the strange thing: If I initialize these parts in order Serial->WiFi->MQTT, debug serial prints as normal, but the receive serial will somehow fall into convulsion, the data received is handled in piece of uncertain length like 24bytes, 33bytes and last quite a few seconds. The duration is uncertain too, it can vary from 2 seconds to 1 minute (maybe even longer). Every time the convulsion ends with a huge string received (~290 bytes). 

So the serial become unusable randomly after the MQTT subscribing procedure is finished.

But I also found a way out: that's re-initializing the receiving serial port after MQTT started. In this scenario, the first pack received will be normal.

All the initialization procedures are performed in the `setup()` function, then a `xTaskCreatePinnedToCore()`is called to initialize the serial receive function, once a full string is forged, print it out through the debug serial so the RTOS task never quits.

Thanks for reading thoroughly, hope someone can help.

Huckies, 2021-11-28:
I forgot to mention before that the receiving serial is HardwareSerial No.1 and the debug serial is the default one.

SuGlider, 2021-11-28:
@Huckies 
Could you please post a minimal sketch that causes the issue you have reported?

It will help a lot in figuring out the issue.

Thanks.

Huckies, 2021-11-29:
@SuGlider 
Please check this repo [https://github.com/Huckies/serialcheck](url).
You might want to add a public MQTT server IP to HuckAL.h line 13.
Also a modification to HuckAL_WiFi.cpp line 5&6 might be needed.

I've mistaken something, the glitch I described above was on a platformio framework which uses an arduino-component v1.0.6 [https://github.com/espressif/arduino-esp32/releases/tag/1.0.6](url)
The current situation is: The first pack received is always a huge one (~209 bytes), if the receiving serial port is not initialized again it freezes and the re-initializing operation still works.

SuGlider, 2021-11-29:
@Huckies 
We have refactored UART in Arduino Core v2.0.0+
A lot of UART bugs have been solved in this newer version.

Could you please test your sketch using Arduino Core 2.0.1 (current latest release) and report back if those issues are still happening?
Thanks!

Huckies, 2021-11-29:
@SuGlider 
Like I said, currently running a 2.0.1 version and still occurs the glitch.

It just become reticent instead of receiving broken sentences before re-initialized, but it should work fine even without the second initialization, shouldn't it?

SuGlider, 2021-11-29:
@Huckies 
I'll ckeck it out and see if I can reproduce the issue. I'll report back here along this week.

Huckies, 2021-12-07:
@SuGlider 
No urging, is everything going well? This glitch is pretty "stable" here so...I'm at a loss now.

SuGlider, 2021-12-09:
@Huckies - Just some few guesses here... 
- Have you tried to increase the stack size of task_handle_UART?
- What about increasing the buffer size for Serial1, by using ```PassthroughSerial.setRxBufferSize(size_t newRXBufferSize);``` 

Example:
``` dart
void HuckAL::UART_Init()
{
    Serial.begin(115200);
    Serial.print("Fundamental serial port initialized with Core ");
    Serial.println(xPortGetCoreID());
    PassthroughSerial.setRxBufferSize(2048);   // Must be executed before Serial.begin()
    PassthroughSerial.begin(115200, SERIAL_8N1, RXD1_PIN, TXD1_PIN);
    Serial.println("Prepared UART No.0 and No.1");
    delay(1000);
}
```

Huckies, 2021-12-13:
@SuGlider 
The increasing of buffer size for Serial1 works, it's normal now
Would you please explain why?

SuGlider, 2021-12-13:
@Huckies 
Currently ESP32 implementation for SerialHardware in version 2.0.0+ is based on IDF 4.4.
When the IDF ISR fails to push data to its RX RingBuffer, it will have to stash the data, and send it to the RingBuffer at a next time. Stashing the data means not freeing internal FIFO and this may lead to loosing some UART received data. 
This RingBuffer is where UART data are kept until Arduino API is used to read it. 
The default RingBuffer size, in the current SerialHardware implementation, is 256 bytes. 

I guess that this size may be too small for the execution of your application.
Thus, in this specific case for your application, increasing the buffer size (which is a FreeRTOS RingBuffer) solves the issue.

closed 2021-12-13 by SuGlider