espressif/arduino-esp32 #5938: The esp32 i2s internal DAC output is not working with the release 2.0.1

cicciocb, 2021-11-28:
The audio output using I2S works both with an external DAC or in PDM mode but not using the internal DAC.
This seems to be related to the ESP-IDF [https://github.com/espressif/esp-idf/issues/7684](url)

VojtechBartoska, 2021-11-30:
PTAL, @PilnyTomas.

relates to: https://github.com/espressif/esp-idf/issues/7684

FedericoBusero, 2021-12-08:
It did work until V2.0.0-rc2

Jason2866, 2022-01-08:
Do you have news for this @PilnyTomas ?

PilnyTomas, 2022-01-12:
Hi, sorry for the late response - I was working on other tasks.
I can confirm that in version 2.0.1 and newer it doesn't work and that version 2.0.0. does work.
I will investigate to pinpoint the breaking commit and find the reason for this issue.

PilnyTomas, 2022-01-26:
I found the braking commit, from Oct 1, 2021: [00214d5c2a1c2b1904f2caf6f0d5ddfe952331ff](https://github.com/espressif/arduino-esp32/commit/00214d5c2a1c2b1904f2caf6f0d5ddfe952331ff)

VojtechBartoska, 2022-01-26:
We are now discussing the issue with ESP-IDF team, we will let you know about the update.

Jason2866, 2022-01-27:
Maybe this https://www.esp32.com/viewtopic.php?t=6025

s-hadinger, 2022-01-27:
> Maybe this https://www.esp32.com/viewtopic.php?t=6025

I have tested using `I2S_MODE_PDM ` instead of `I2S_MODE_DAC_BUILT_IN` and I get sound again with a DAC output. It seems that PDM and DAC have been swapped.

s-hadinger, 2022-01-28:
I did some more tests, on ESP32 Rev1 and Pico D4. Actually the DAC output works, but the speed is 11x more than real time. I.e. I sent PCM at 16K samples/s, and it was played at ~176K samples/s.

Original waveform is 660ms long
[image]

The signal recorded on GPIO26 is playing in 60ms
[image]

cicciocb, 2022-01-31:
Maybe this could be related to ?
https://github.com/espressif/esp-idf/commit/1d02e638bd05b80d6fd648d51c14a8ecd6c2dbdc

s-hadinger, 2022-01-31:
I doubt. I did some detour on `i2s_hal_tx_clock_config()` to check what clock values were calculated, and force-injected some different clock values until the sound was good again. I will work backwards to see why these values are wrong in the first place.

cicciocb, 2022-01-31:
@s-hadinger What is the version of the arduino core that you are using?

s-hadinger, 2022-01-31:
I use 2.0.2 with IDF 4.4. I believe the problem is in idf, not Arduino Core.

One highly possible cause is that the calculated `mclk_div` is 1250 when using 16000Hz audio on DAC. I'm not sure yet what are the values of `a` and `b` divider, but I realize now that `mclk_div` is 8-bit long. 1250 is clearly an overflow.

Values calculated by IDF 4.4 for 16000Hz audio, DAC, 2 channels
`sclk=160000000 mclk=128000 bclk=64000 mclk_div=1250 bclk_div=2`

Hardware registers:
```
    union {
        struct {
            uint32_t clkm_div_num: 8;
            uint32_t clkm_div_b:   6;
            uint32_t clkm_div_a:   6;
            uint32_t clk_en:       1;
            uint32_t clka_en:      1;
            uint32_t reserved22:  10;
        };
        uint32_t val;
    } clkm_conf;
```

The overflow may explain why I got random results, and why I only could get something to work by increasing `bclk_div`

s-hadinger, 2022-01-31:
I have now confirmation. For 16000Hz, 2 channels, internal DAC is 8 bits.

Knowing that the `sclk` is 160MHz
`esp-idf` set the following `mclk_div=1250` and `bclk_div=2` which makes an overall divider around at 1250*2 = 2500, i.e. a bit frequency of 160000000/2500 = 64000 bps. But the problem is that `mclk` is overflowing the 8 bits.

My working version sets: `mclk_div=156`, `bclk_div=32`, so the overall divider is 156*32 = 4992. So the bit frequency is 160000000/4992 = 32051 bps, which is in line with 16000 * 2 (for 2 channels).

I'm still unclear whether a and b factors have an influence.

s-hadinger, 2022-01-31:
I opened a new issue with these information https://github.com/espressif/esp-idf/issues/8326

s-hadinger, 2022-02-01:
Recompiling esp-idf with the fix above(thanks @Jason2866) makes Audio DAC working again.

Here are the new values computed:
`sclk=160000000 mclk=1024000 bclk=32000 mclk_div=156 bclk_div=32`

Jason2866, 2022-02-01:
The framework with the idf i2s fix is [here](https://github.com/Jason2866/esp32-arduino-lib-builder/releases/download/618/framework-arduinoespressif32-IDF44.tar.gz)
There are some different sdkconfig settings choosen to optimize for our needs.
The build can directly used with platformio using `platform_packages`

cicciocb, 2022-02-01:
@Jason2866  Thanks for your framework but what is exactly the fix done inside the idf?
Thanks

cicciocb, 2022-02-01:
@s-hadinger  Thanks for your answer, I found the fix that is here
https://github.com/espressif/esp-idf/pull/8327/files

cicciocb, 2022-02-03:
@Jason2866  I tried but I still have the same issue, works well with external DAC or PDM but not with the internal DAC
@s-hadinger Could you publish the source code that you are using for the test?

Jason2866, 2022-02-03:
@cicciocb Used with Tasmota ;-)

cicciocb, 2022-02-03:
@Jason2866 
Yes, unfortunately I'm sure, it works rolling back to a previous version.
Could you just publish your test case?

Jason2866, 2022-02-03:
@cicciocb  You can use this setup https://github.com/arendst/Tasmota/pull/12795

cicciocb, 2022-02-03:
@Jason2866  Still not working, at least for me. Maybe someone else could confirm if working or not

Jason2866, 2022-02-03:
@cicciocb Can you publish your test case?

VojtechBartoska, 2022-04-06:
@cicciocb Wil you be able to retest this with v2.0.3-RC1? It should be already fixed :) Thank you.

pjmi1, 2022-05-12:
This is still broken (I2S using I2S_MODE_DAC_BUILT_IN) - it still does not work for Arduino ESP32 core v. 2.0.3

PilnyTomas, 2022-05-13:
> This is still broken (I2S using I2S_MODE_DAC_BUILT_IN) - it still does not work for Arduino ESP32 core v. 2.0.3

@pjmi1 Please open a new issue with specifications for your problem. Please include a complete sketch that demonstrates the issue.

ifrew, 2022-06-11:
I can confirm this is still broken in Arduino 2.0.4, IDF 4.4.1 

Not sure why we need to open a new bug for this @PilnyTomas? 

Here is the sketch I used. If I change from internal DAC  to PDM it works although really noisy.

// Arduino Zero / Feather M0 I2S audio tone generation example.
// Author: Tony DiCola
//
// Connect an I2S DAC or amp (like the UDA1334A) to the Arduino Zero
// and play back simple sine, sawtooth, triangle, and square waves.
// Makes your Zero sound like a NES!
//
// NOTE: The I2S signal generated by the Zero does NOT have a MCLK /
// master clock signal.  You must use an I2S receiver that can operate
// without a MCLK signal (like the UDA1334A).
//
// For an Arduino Zero / Feather M0 connect it to you I2S hardware as follows:
// - Digital 0 -> I2S LRCLK / FS (left/right / frame select clock)
// - Digital 1 -> I2S BCLK / SCLK (bit / serial clock)
// - Digital 9 -> I2S DIN / SD (data output)
// - Ground
//
// Released under a MIT license: https://opensource.org/licenses/MIT

//modified for M5Stack Gray Core

#include <M5Stack.h>
#include <driver/i2s.h>

#define SPEAKER_I2S_NUMBER I2S_NUM_0

#define SAMPLERATE_HZ 44100  // The sample rate of the audio.  Higher sample rates have better fidelity,
                             // but these tones are so simple it won't make a difference.  44.1khz is
                             // standard CD quality sound.

#define AMPLITUDE     ((1<<31)-1)   // Set the amplitude of generated waveforms.  This controls how loud
                             // the signals are, and can be any value from 0 to 2**31 - 1.  Start with
                             // a low value to prevent damaging speakers!

#define WAV_SIZE      256    // The size of each generated waveform.  The larger the size the higher
                             // quality the signal.  A size of 256 is more than enough for these simple
                             // waveforms.

// Define the frequency of music notes (from http://www.phy.mtu.edu/~suits/notefreqs.html):
#define C4_HZ      261.63
#define D4_HZ      293.66
#define E4_HZ      329.63
#define F4_HZ      349.23
#define G4_HZ      392.00
#define A4_HZ      440.00
#define B4_HZ      493.88

// Define a C-major scale to play all the notes up and down.
float scale[] = { C4_HZ, D4_HZ, E4_HZ, F4_HZ, G4_HZ, A4_HZ, B4_HZ, A4_HZ, G4_HZ, F4_HZ, E4_HZ, D4_HZ, C4_HZ };

// Store basic waveforms in memory.
int32_t sine[WAV_SIZE]     = {0};
int32_t sawtooth[WAV_SIZE] = {0};
int32_t triangle[WAV_SIZE] = {0};
int32_t square[WAV_SIZE]   = {0};
size_t writeSize;

// Create I2S audio transmitter object.
//Adafruit_ZeroI2S i2s;

#define Serial Serial

void generateSine(int32_t amplitude, int32_t* buffer, uint16_t length) {
  // Generate a sine wave signal with the provided amplitude and store it in
[... 116 lines cut ...]
    delay(100);
  }
  Serial.println("Sawtooth wave");
  for (int i=0; i<sizeof(scale)/sizeof(float); ++i) {
    // Play the note for a quarter of a second.
    playWave(sawtooth, WAV_SIZE, scale[i], 0.25);
    // Pause for a tenth of a second between notes.
    delay(100);
  }
  Serial.println("Triangle wave");
  for (int i=0; i<sizeof(scale)/sizeof(float); ++i) {
    // Play the note for a quarter of a second.
    playWave(triangle, WAV_SIZE, scale[i], 0.25);
    // Pause for a tenth of a second between notes.
    delay(100);
  }
  Serial.println("Square wave");
  for (int i=0; i<sizeof(scale)/sizeof(float); ++i) {
    // Play the note for a quarter of a second.
    playWave(square, WAV_SIZE, scale[i], 1.0);
    // Pause for a tenth of a second between notes.
    //delay(1);
    delay(100);
  }
}

PilnyTomas, 2022-06-21:
Hi @ifrew I tried your code and it takes 110 seconds to generate that 0.25sec tone. Please try improving tone generation.

PilnyTomas, 2022-07-20:
I have tried IDf example and the output is very laggy, so I opened an issue for the IDF team
https://github.com/espressif/esp-idf/issues/9389

JoseEnriqueFA, 2022-09-27:
The esp32    I2s internal DAC output is    NOT working with the release 2.0.5

I tested yesterday the last Board Manager release 2.0.5  and  a sketch to generate a sinus signal with I2S and the  internal DAC output.  Same result:  Not Working!

HW:   Adafruit ESP32 feather and an oscilloscope

With the Board Manager 2.0.0 the sketch works but all the Board Manager  > 2.0.0  have same  bug

Is there any work around?

Note: I discovered the bug trying to evaluate the Adafruit ESP32 feather V2. You need to upgrade the Board Manager to 2.0.4 
        Same issue is shown in both boards Adafruit ESP32 feather  and Adafruit ESP32 feather V2

PilnyTomas, 2022-10-17:
Can you please try [this sketch](https://github.com/PilnyTomas/issues/blob/4b5315b5547165c9f080fd3e993dce24bf0fe7bf/issue_7252B/issue_7252B.ino) if it works for you? It does for me.

JoseEnriqueFA, 2022-10-17:
this sketch <https://github.com/PilnyTomas/issues/blob/4b5315b5547165c9f080fd3e993dce24bf0fe7bf/issue_7252B/issue_7252B.ino>   is also working for me

 

 

But:

        .use_apll = 1

       produces sporadic glitches with the Arduino  Board Manager  2.0.5   .   See:    https://github.com/espressif/esp-idf/issues/9321

 

      Alternatives:  

                      .use_apll = 0        (For my application  is fine)

      or

 

                     Arduino  Board Manager  1.0.6    .use_apll = 1    does not produce glitches

 

 

 

Best whishes

Enviado el: lunes, 17 de octubre de 2022 17:12
Para: espressif/arduino-esp32
CC: JoseEnriqueFA; Comment

PilnyTomas, 2022-11-09:
The topic seems to be dead, so I am closing it.

closed 2022-11-09 by PilnyTomas

JoseEnriqueFA, 2022-11-10:
> Enviado el: miércoles, 09 de noviembre de 2022 18:13

Hi,

 

After many hours used in this issue ....

You have to init I2S  depending the Board Manager you are using,  see my code bellow

In summary 

                        with Adafruit feather I follow using B.M. 1.0.6

                        with Adafruit feather v2  I use  B.M. 2.0.4

 

 

Nevertheless  take care of

 

1-  if you change    use_apll = false,     to true 

    with B.M. 2.0.4    you will get glitches 

     It does not happen with B.M. 1.0.6

 

2- In my application 

    I cannot use  B.M. 2.0.5 because there are other issues related   SPIFFS  ...

    I should investigate ....

 

3-   I have not  tested    B.M. 2.0.6

   

Good luck

 

////////////////////////////////////////////////////////////////////////////////////////////////////

///////////////////////////////////////////////////////////////////////////////////////////////////

#ifdef B_Manager_1_0_6 

 

void i2sInit()

{

   i2s_config_t i2s_config = {

    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN),

   .sample_rate =  I2S_SAMPLE_RATE_LOW,              // The format of the signal using ADC_BUILT_IN

    .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT, // is fixed at 12bit, stereo, MSB

    .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,   

    .communication_format = I2S_COMM_FORMAT_I2S_MSB,        

    .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,               

    .dma_buf_count = 8,

    .dma_buf_len = 64,

    .use_apll = false,

    .tx_desc_auto_clear = false,                            

    .fixed_mclk = 0                                         

   };

   

 

 

   i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL); //install and start i2s driver

   i2s_set_pin(I2S_NUM_0, NULL); //for internal DAC, this will enable both of the internal channels

 

 

 

}

#endif

 

////////////////////////////////////////////////////////////////////////////////////////////////////

// Board Manager superior a 2.0.4  needs for ADAFRUIT ESP32 feather V2

///////////////////////////////////////////////////////////////////////////////////////////////////

#ifdef B_Manager_>_2_0_4

 

void i2sInit()

{

   i2s_config_t i2s_config = {

    .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_TX | I2S_MODE_DAC_BUILT_IN),

    .sample_rate =  I2S_SAMPLE_RATE_LOW,              // The format of the signal using ADC_BUILT_IN

    .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT, // is fixed at 12bit, stereo, MSB

    .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,      

    .communication_format = I2S_COMM_FORMAT_STAND_MSB,                                                                                  //   B.M. 2.0.4

    .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,               

    .dma_buf_count = 8,

    .dma_buf_len = 64,

    .use_apll = false,

    .tx_desc_auto_clear = false,                            

    .fixed_mclk = 0                                         

   };

   

 

 

   i2s_driver_install(I2S_NUM_0, &i2s_config, 0, NULL); //install and start i2s driver

   i2s_set_pin(I2S_NUM_0, NULL); //for internal DAC, this will enable both of the internal channels

 

  

 

 

}

#endif

 

Hi, I ran into this issue yesterday. Took me hours of testing different boards and max chips, checking wiring and pin layout etc. to finally come to the conclusion that the only difference of my hitherto working code vs. non working code and device was 1.0.6 vs 2.0.6 esp32 framework.

Arduino 1.8.6, esp32 2.0.6 framework, esp32 dev board, max dac chip.

Downgraded to 1.0.6, dac works again immediately.

PilnyTomas, 2022-11-10:
Please understand the difference between IDF and arduino-esp32.
The function `i2s_driver_install` is IDF (please read the [documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/i2s.html)) - if something doesn't work, please report it to [IDF repository](https://github.com/espressif/esp-idf/issues) 

From version 2.0.3 of arduino-esp32 there is [I2S library](https://github.com/espressif/arduino-esp32/tree/2.0.3/libraries/I2S) using Arduino API ([Arduino Documentation](https://docs.arduino.cc/learn/built-in-libraries/i2s))
I am currently working on some fixes and updates for this library - you can follow them in PR #7117 

Arduino-esp32 is based on IDF so you can always use IDF functions and combine them with Arduino API. We pull updates done in IDF from time to time, which could cause breaking changes or introduce bugs, but most of the time they are fixing previous bugs.