espressif/arduino-esp32 #5939: Publish arduino-esp32 component in idf-compnenent-manager

igrr, 2021-11-28:
Currently the [instructions](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html) for installing arduino-esp32 as an IDF component use git to clone arduino-esp32 into the project components directory.

We could make arduino-esp32 available via component manager in ESP-IDF, simplifying installation and dependency management. For example, arduino-esp32 component could "depend" on the specific versions of esp-dsp, rainmaker, esp32-camera and other components arduino-esp32 depends on, and idf-component-manager will automatically install them.

The task is to:
* add component manifest (`idf_component.yml`) file to the repository ([docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-component-manager.html))
* add a CI job to publish arduino-esp32 component to https://components.espressif.com/, similar to https://github.com/espressif/esp-bsp/blob/master/.github/workflows/upload_component.yml
* update instructions for installing arduino-esp32 as an IDF component, i.e. `idf.py add-dependency espressif/arduino-esp32==2.1.0`
* (optional) add a scheduled CI job similar to https://github.com/espressif/arduino-esp32/pull/5842, to check that the instructions given in the docs keep working

Huckies, 2021-11-28:
You will add this library to the Espressif extension in VSCode, right? One of the charming things of ESP32 is the official support in VSCode, it makes coding much more painless than STM32 where Keil is commonly used.

igrr, 2021-12-01:
@Huckies Appreciate your kind words. That's correct, once arduino-esp32 is added to the component manager it should be easy to add it to an IDF project either in VS Code or Eclipse.

closed 2022-02-03 by me-no-dev

rosmo, 2022-12-17:
Looks like there was a CI job made for this but doesn't appear it ever succeeded? I can't find `arduino-esp32` in IDF Component registry.