espressif/arduino-esp32 #5940: Serial1 breaks when updateBaudRate() is used together with WIFI

savejeff, 2021-11-28:
I have found wierd behavior/a bug with the hardwareSerial.updateBaudRate function. 
It only happens if i also use WIFI

When i connect to wifi and then change the baud rate of Serial1, the serial port does only read 0x00.
it only works if i use the updateBaudRate  and WIFI is connected
if i use Serial.end(), Serial,begin() it works fine.

I use a ESP32 (Generic DevModule)
Windows 10
Tested with current version of Arduino Core/IDE as well as older ones with PlatformIO

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

#define Serial0 Serial

#include <WiFi.h>

//command to change the baud rate of the sensor to 115200
char cmd_baud_115200[] = { 0xB5, 0x62, 0x06, 0x00, 0x14, 0x00, 0x01, 0x00, 0x00, 0x00, 0xD0, 0x08, 0x00, 0x00, 0x00, 0xC2, 0x01, 0x00, 0x07, 0x00, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 0xC0, 0x7E };
#define send_cmd(cmd) Serial1.write((byte*)cmd, sizeof(cmd))

void setup() {
  Serial0.begin(115200);
  Serial1.begin(9600, SERIAL_8N1, 16, 17);
  Serial1.setRxBufferSize(1024);
  

  delay(3000);
  Serial0.println("Start");

  WiFi.mode(WIFI_STA);
  WiFi.begin("EXAMPLE_SSID", "EXAMPLE_PASSWORD");

  send_cmd(cmd_baud_115200); //changes baud rate of sensor
  Serial1.flush();
  delay(500);
  //Change baud rate to conform with ne baud rate of sensor

  //with this I only get 0x00 from read function
  Serial1.updateBaudRate(115200);

  //With this it works
  //Serial1.end();
  //Serial1.begin(115200, SERIAL_8N1, 16, 17);
  
  
}

void loop() {
  static unsigned long time0 = millis();

  if(millis() - time0 > 1000)
  {
    time0 = millis();
    Serial0.print(millis()); 
    Serial0.print("Wifi connected: ");
    Serial0.print((WiFi.status() == WL_CONNECTED));
    Serial0.println();
    
  }

  if(Serial1.available())
  {
      Serial0.print((char) Serial1.read());
  }

}

```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

SuGlider, 2021-11-29:
@savejeff 
Please let me know more information about your development environment, such as Arduino Core Version, board model, used IDE etc.

If possible, please run the code below using the same setup of the sketch used to demonstrate the issue and report back the output.

``` dart

void setup() {
  Serial.begin(115200);
  Serial.println("\nSystem Information\n");
  Serial.printf("Internal Total heap %d, internal Free Heap %d\n", ESP.getHeapSize(), ESP.getFreeHeap());
  Serial.printf("SPIRam Total heap %d, SPIRam Free Heap %d\n", ESP.getPsramSize(), ESP.getFreePsram());
  Serial.printf("ESP32 Chip model = %s, ChipRevision %d\n", ESP.getChipModel(), ESP.getChipRevision());
  Serial.printf("This chip has %d cores, Cpu Freq %d\n", ESP.getChipCores(), ESP.getCpuFreqMHz());
  Serial.printf("SDK Version %s\n", ESP.getSdkVersion());
  Serial.printf("Flash Size %d, Flash Speed %d\n", ESP.getFlashChipSize(), ESP.getFlashChipSpeed());

#if defined(BOARD_HAS_PSRAM)
  Serial.printf("Board has PSRAM\n");
#else
  Serial.printf("Board doesn't have PSRAM\n");
#endif
}

void loop() {
}
```

savejeff, 2021-11-29:
Hi!

thanks for the fast response. 

I'm using Arduino IDE 1.8.13 
ESP Arduino Core Version 2.0.1

I also tested this with PlatformIO and the latest version.

here the output of the program:
```

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1420
ho 0 tail 12 room 4
load:0x40078000,len:13540
load:0x40080400,len:3604
entry 0x400805f0

System Information

Internal Total heap 371256, internal Free Heap 346548
SPIRam Total heap 0, SPIRam Free Heap 0
ESP32 Chip model = ESP32-D0WDQ6, ChipRevision 1
This chip has 2 cores, Cpu Freq 240
SDK Version v4.4-dev-3569-g6a7d83af19-dirty
Flash Size 4194304, Flash Speed 80000000
Board doesn't have PSRAM

```

SuGlider, 2021-11-29:
@savejeff 
The first thing that I can see in the example code your posted is that 
```Serial1.setRxBufferSize(1024);```
Must be called before 
```Serial1.begin(9600, SERIAL_8N1, 16, 17);```
Otherwise it has no effect.

A error message shall be printed in UART0 regarding it. Maybe you have not enabled the debug option in the Arduino IDE.

But this doesn't seem to be the issue.
I'll try to reproduce the issue and I'll let you know along this week.

savejeff, 2021-12-01:
thanks for the info. i've corrected the serial buffer size call. 
I usually work in PlatformIO and have debug level at 3. sofar I haven't noticed an error msg. 

If I find time ill test if the example also produces the error if the external UBlox sensor is replaced with serial2. It would then be very easy to reproduce the error by just connecting serial2 and serial1 on a breadboard.

SuGlider, 2021-12-01:
@savejeff 
If you didn't notice any error message, maybe you are running on Arduino Core 1.0.6 with IDF 3.3.

Could you please check your Arduino configuration by running the code below:

``` dart
void setup () {
    Serial.begin(115200);

    Serial.printf("Internal Total heap %d, internal Free Heap %d\n", ESP.getHeapSize(), ESP.getFreeHeap());
    Serial.printf("SPIRam Total heap %d, SPIRam Free Heap %d\n", ESP.getPsramSize(), ESP.getFreePsram());
    Serial.printf("ChipRevision %d, Cpu Freq %d, SDK Version %s\n", ESP.getChipRevision(), ESP.getCpuFreqMHz(), ESP.getSdkVersion());
    Serial.printf("Flash Size %d, Flash Speed %d\n", ESP.getFlashChipSize(), ESP.getFlashChipSpeed());
}

void loop() { }
```

SuGlider, 2022-01-11:
@savejeff 

I was unable to reproduce the issue using Arduino Core V.2.0.2.
I don't have the Sensor you may be using, but I tried to reproduce the issue using UART2 to simulate it changing from 9600 to 115200.

This is the sketch used to test it:

```dart
#define Serial0 Serial

#include <WiFi.h>

//command to change the baud rate of the sensor to 115200
//char cmd_baud_115200[] = { 0xB5, 0x62, 0x06, 0x00, 0x14, 0x00, 0x01, 0x00, 0x00, 0x00, 0xD0, 0x08, 0x00, 0x00, 0x00, 0xC2, 0x01, 0x00, 0x07, 0x00, 0x03, 0x00, 0x00, 0x00, 0x00, 0x00, 0xC0, 0x7E };
//#define send_cmd(cmd) Serial1.write((byte*)cmd, sizeof(cmd))

void setup() {
  Serial0.begin(115200);
  Serial1.setRxBufferSize(1024);
  Serial1.begin(9600, SERIAL_8N1, 15, 14);   // 15=Rx1, 14=Tx1 || 16 = Rx2, 17 = Tx2 (default) 
  // Serial2 TX <--> Serial1 RX -- Must connect both pins
  Serial2.begin(9600);   // it will send bytes to Serial1 in a loopback connection to simulate the Sensor
  Serial2.print("Sensor Started.\n"); // this bytes will be kept in Serial1 buffer for reading it later

  delay(300);
  Serial0.println("Start");

  WiFi.mode(WIFI_STA);
  WiFi.begin("EXAMPLE_SSID", "EXAMPLE_PASSWORD");

  //send_cmd(cmd_baud_115200); //changes baud rate of sensor
  Serial2.updateBaudRate(115200);
  
  Serial1.flush();
  delay(500);
  //Change baud rate to conform with ne baud rate of sensor

  //with this I only get 0x00 from read function
  Serial1.updateBaudRate(115200);

  //With this it works
  //Serial1.end();
  //Serial1.begin(115200, SERIAL_8N1, 16, 17);
  
  
}

void loop() {
  static unsigned long time0 = millis();
  static uint8_t count = 0;

  if(millis() - time0 > 1000)
  {
    time0 = millis();
    Serial0.printf("\n[time: %d] Wifi connected: %d\n", time0, (WiFi.status() == WL_CONNECTED));
  }

  if(Serial1.available())
  {
      Serial0.print((char) Serial1.read());
  }
  Serial2.write('0' + count++);
  if (count == 10) {
    count = 0;
  }

  delay(10);
}
```

Output seems OK:
``` dart
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1324
ho 0 tail 12 room 4
load:0x40078000,len:13508
load:0x40080400,len:3604
entry 0x400805f0
Start
Sensor Started.
0123456789012345678901234567890123456789012345678901234567890123456789012345678901234
[time: 1942] Wifi connected: 0
56789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345
[time: 2952] Wifi connected: 1
67890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456
[time: 3962] Wifi connected: 1
78901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567
[time: 4972] Wifi connected: 1
89012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678
[time: 5982] Wifi connected: 1
90123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789
[time: 6992] Wifi connected: 1
01234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890
[time: 8002] Wifi connected: 1
12345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901
[time: 9012] Wifi connected: 1
23456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012
[time: 10022] Wifi connected: 1
34567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123
[time: 11032] Wifi connected: 1
45678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234
````

VojtechBartoska, 2022-04-08:
Hello, as we are not able to reproduce this issue, I'm closing it.

If needed please reopen this one or create a new issue. Thanks for your contribution.

closed 2022-04-08 by VojtechBartoska