espressif/arduino-esp32 #5942: Random crash on TCP/IP stack of ESP32

garudaonekh, 2021-11-28:
Hi,
I got this crash after some a few requests from client(My ESP32 act as server) and not able to find out the root cause as it happen sometimes after an hour of running. The sketch will tunnel http request thru MQTT response so user will send MQTT request to the sketch and the sketch will convert this MQTT as http request and push it to TCP/IP stack and then convert the HTTP response and send back as MQTT response.

My code is based on this two libraries
https://github.com/levkovigor/MQTT_VPN_ESP32
https://github.com/levkovigor/ppposclient

`PC: 0x4012c1e9: tcp_output at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_out.c line 1153
EXCVADDR: 0x00000010

Decoding stack results
0x4012c1e9: tcp_output at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp_out.c line 1153
0x401285b5: tcp_slowtmr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c line 1213
0x40128a46: tcp_tmr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c line 164
0x40125d97: tcpip_tcp_timer at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/timeouts.c line 140
0x40125c85: sys_check_timeouts at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/timeouts.c line 361
0x40125e8c: sys_timeouts_mbox_fetch at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/timeouts.c line 433
0x40120a84: tcpip_thread at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c line 109
0x4008cf7e: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143`

atanisoft, 2021-12-03:
Can you share the full backtrace? There should be more information as to the actual cause of the backtrace just above what you have provided.

Also since you have not specified which version of arduino-esp32 you are using it is a bit more difficult to pin down what version of components were used. However, based on the backtrace line numbers it would appear you are using arduino-esp32 1.0.6 (or earlier) which maps to ESP-IDF v3.3 (or earlier).

Based on the limited information you have provided I can only guess as to the cause of your crash but it looks like the packets being sent by the LwIP task are not being ack'd quickly enough leading to possible pbuf exhaustion. This is based on https://github.com/espressif/esp-lwip/blob/2acf959a2bb559313cd2bf9306c24612ba3d0e19/src/core/tcp_out.c#L1153 which your backtrace points to, but it is only a guess at this point.

garudaonekh, 2021-12-05:
Sorry, I can't get any more stacktrace beside the above. I am trying to use ESP-Prog which may help me with more info, but I can't get it work yet.

However, here's the configuration
```
PLATFORM: Espressif 32 (3.3.2) > Espressif ESP32 Dev Module

PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - toolchain-riscv-esp 1.80400.0 (8.4.0)
```

VojtechBartoska, 2022-04-07:
Hello, sorry for picking this up late. Do you still need some support @garudaonekh?

garudaonekh, 2022-04-08:
> Hello, sorry for picking this up late. Do you still need some support @garudaonekh?

Yes, please. it still happened and I have to use it this way.

VojtechBartoska, 2023-02-16:
Unfortunetely, I'm closing this as expired due to no answer. Thanks for understanding.

closed 2023-02-16 by VojtechBartoska