espressif/arduino-esp32 #5943: ../components/arduino/cores/esp32/esp32-hal-i2c-slave.c:363:12: error: 'i2c_slave_struct_t' {aka 'struct i2c_slave_struct_t'} has no member named 'lock'

rhr407, 2021-11-28:
I am facing the following error after adding arduino as an esp-idf component. The version of esp-idf is v4.4
```
../components/arduino/cores/esp32/esp32-hal-i2c-slave.c:363:12: error: 'i2c_slave_struct_t' {aka 'struct i2c_slave_struct_t'} has no member named 'lock'
```

me-no-dev, 2021-11-29:
yup :) will fix. you can enable hal locks and should compile fine.

rhr407, 2021-12-04:
Thanks, it's working now.

VojtechBartoska, 2022-01-10:
probably will be solved by #6108

VojtechBartoska, 2022-04-06:
@rhr407 Are you able to test your issue on development version [2.0.3-RC1](https://github.com/espressif/arduino-esp32/releases/tag/2.0.3-RC1) to check if this is still valid? You can take a look on [Docs](https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html) where is explained how to choose development release version in Arduino IDE.

closed 2022-04-07 by rhr407