espressif/arduino-esp32 #5947: ESP32-S2: pinMode() not working on certain pins

PaintYourDragon, 2021-11-30:
### Hardware:
Board:							Adafruit Feather ESP32-S2
Core Installation version:			2.0.1
IDE name:						Arduino IDE

### Description:
pinMode() has no effect on pins 9, 11 or A4. Other pins work as expected.
Replacing digitalWrite() with equivalent GPIO register accesses _does not_ work; issue is likely in pinMode().
Replacing pinMode() with equivalent gpio_set_direction() call _does_ work.
A different workaround, though I don’t know if this is universally correct fix or will break something else, is to edit esp32-hal-gpio.c and move line 41:
```#define USE_ESP_IDF_GPIO 1```
…up 2 lines so it’s in the CONFIG_IDF_TARGET_ESP32S2 condition rather than the else. Test code then works on all pins.

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp

// Just the standard Arduino blink, but on different pin. Also try 9 or A4:

void setup() {
  pinMode(11, OUTPUT);
}

void loop() {
  digitalWrite(11, HIGH);
  delay(100);
  digitalWrite(11, LOW);
  delay(100);
}
```

PaintYourDragon, 2021-11-30:
Tagging @ladyada on this to follow along on current status.

atanisoft, 2021-12-01:
Very likely `USE_ESP_IDF_GPIO` should be moved to the default for all chips.

ladyada, 2021-12-14:
@me-no-dev hiya can someone try to repro this, should we try to submit a PR?

me-no-dev, 2021-12-14:
@ladyada soon all GPIO will be rewritten, so I would guess no further action needs to be taken from your side.

VojtechBartoska, 2022-04-08:
Hello, GPIO was refactored. Can you help with testing this on v2.0.3-rc1? Thanks

ladyada, 2022-04-10:
tested gpio 11 with success!

VojtechBartoska, 2022-04-22:
Closing as solved.

closed 2022-04-22 by VojtechBartoska