espressif/arduino-esp32 #5948: ESP32-S2: Linker error undefined reference to `__sync_sub_and_fetch_4' (ESP32 is working)

klausj1, 2021-11-30:
### Hardware:
Board:							ESP32-S2 (WROVER / ESP32-S2-Saola-1)
Core Installation version:			2.0.1
IDE name:						Arduino IDE 1.8.16
Flash Frequency:					80Mhz
PSRAM enabled:                                     No
Upload Speed:					921600
Computer OS:                                         Windows 10

### Description:
I am trying to connect my ESP32-S2 to the Azure IoT Hub using [ESP32_AzureIoT - Azure IoT Hub library for esp32 devices in Arduino](https://github.com/VSChina/ESP32_AzureIoT_Arduino).

BTW: Is there a better way to connect the ESP32-S2 to Azure IoT than using this library?

When using the device ESP32SDev Module,  compile and linking works.

When using the device ESP32S2 Dev Module, I get a linker error:

```
c:/users/klaus/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r1/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: c:\Users\Klaus\Documents\IoTWorkbenchProjects\examples\esp32_get_started\Device\.build\libraries\ESP32_AzureIoT_Arduino\az_iot\c-utility\src\constbuffer.c.o:(.literal.CONSTBUFFER_Destroy+0x0): undefined reference to `__sync_sub_and_fetch_4'
c:/users/klaus/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r1/bin/../lib/gcc/xtensa-esp32s2-elf/8.4.0/../../../../xtensa-esp32s2-elf/bin/ld.exe: c:\Users\Klaus\Documents\IoTWorkbenchProjects\examples\esp32_get_started\Device\.build\libraries\ESP32_AzureIoT_Arduino\az_iot\c-utility\src\constbuffer.c.o: in function `CONSTBUFFER_Destroy':
C:\Users\Klaus\Documents\Arduino\libraries\ESP32_AzureIoT_Arduino\src\az_iot\c-utility\src/constbuffer.c:143: undefined reference to `__sync_sub_and_fetch_4'
collect2.exe: error: ld returned 1 exit status
```

I tracked this down to [this file](https://github.com/VSChina/ESP32_AzureIoT_Arduino/blob/master/src/az_iot/c-utility/inc/azure_c_shared_utility/refcount.h), which defines this:

```
#define DEC_RETURN_ZERO (0)
#define INC_REF(type, var) __sync_add_and_fetch((&((REFCOUNT_TYPE(type)*)var)->count), 1)
#define DEC_REF(type, var) __sync_sub_and_fetch((&((REFCOUNT_TYPE(type)*)var)->count), 1)
```

If I change the contents of this file in a way, that DEC_REF is defined differently:

```
#define DEC_RETURN_ZERO (0)
#define INC_REF(type, var) ++((((REFCOUNT_TYPE(type)*)var)->count))
#define DEC_REF(type, var) --((((REFCOUNT_TYPE(type)*)var)->count))
```

Then compiling and linking works. 

This is what I assume happens with ESP32 (not ESP32S2), but to be honest I do not really know what I am doing here.

Which definition is chosen depends on some defines like \_\_STDC_VERSION\_\_, \_\_STDC_VERSION\_\_, \_\_STDC_NO_ATOMICS\_\_, ARDUINO_ARCH_SAMD, FREERTOS_ARCH_ESP8266, but I could not figure out where these definitions are set and whats the difference between ESP32 and ESP32-S2.

(Still, the sketch is not working and delivers this runtime error:

```
Error: Time:Tue Nov 30 05:22:39 2021 File:C:\Users\Klaus\Documents\Arduino\libraries\ESP32_AzureIoT_Arduino\src\az_iot\c-utility\pal\socket_async.c Func:socket_async_create Line:134 Socket connect failed, not EINPROGRESS: 0
Info: Could not open the socket
Info: >>>Connection status: disconnected
Info: >>>Connection status: disconnected
Info: tlsio_openssl_close has been called when in neither TLSIO_STATE_OPEN nor TLSIO_STATE_ERROR.
```

But I would like to follow this up when I the linker error is clarifed.)

### Sketch: 
```cpp

/**
 * A simple Azure IoT example for sending telemetry to Iot Hub.
 */

#include <WiFi.h>
#include "Esp32MQTTClient.h"

#define INTERVAL 10000
#define MESSAGE_MAX_LEN 256
// Please input the SSID and password of WiFi
const char* ssid     = "***";
const char* password = "***";

/*String containing Hostname, Device Id & Device Key in the format:                         */
/*  "HostName=<host_name>;DeviceId=<device_id>;SharedAccessKey=<device_key>"                */
/*  "HostName=<host_name>;DeviceId=<device_id>;SharedAccessSignature=<device_sas_token>"    */
static const char* connectionString = "***";
static bool hasIoTHub = false;
static bool hasWifi = false;

void setup() {
  Serial.begin(115200);
  Serial.println("ESP32 Device");
  Serial.println("Initializing...");
  Serial.println(" > WiFi");
  Serial.println("Starting connecting WiFi.");

  delay(10);
  WiFi.mode(WIFI_AP);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    hasWifi = false;
  }
  hasWifi = true;
  
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.println(" > IoT Hub");
  if (!Esp32MQTTClient_Init((const uint8_t*)connectionString, true))
  {
    hasIoTHub = false;
    Serial.println("Initializing IoT hub failed.");
    return;
  }
}

void loop() {
  delay(10);
}
```

igrr, 2021-11-30:
Thank you for reporting this. We have added a few missing atomic builtins in IDF release/v4.4 branch, but looks like some are still missing. Will fix this in IDF and then arduino-esp32 will pull in the fix in the next release.

klausj1, 2021-11-30:
You are welcome ;)

Could you give me a hint for a quick workaround which I could use? Or is this just not possible?

igrr, 2021-12-01:
You could try to implement the missing functions in a way similar to https://github.com/espressif/esp-idf/blob/master/components/newlib/stdatomic.c, and add them somewhere in your sketch. I think linker might be able to find them. If you defined these functions in a .cpp or .ino file, make sure you declare them as `extern "C"`.

igrr, 2021-12-21:
The fix has been added to IDF `master` branch. We'll try to get it merged into release/v4.4 before the v4.4 release is tagged.

igrr, 2022-01-04:
Fix backported to IDF release/v4.4 in https://github.com/espressif/esp-idf/commit/c8c65a28548b425b81af7acc457dae7533f95572, it will be part of the next Arduino-esp32 release.

commit 78b2df74 (2022-01-18): IDF release/v4.4 f3e0c8bc41 (#6075)

esp-dsp: master 6b25cbb
esp-face: master 925c72e
esp-rainmaker: f1b82c7
esp32-camera: master 221d24d
esp_littlefs: master 5a13cd6

fixes: #5948
closed 2022-01-18 by me-no-dev