espressif/arduino-esp32 #5949: ESP32 ETH no ETH.end / disable

Typhoor, 2021-11-30:
I want do `end` or `stop` the ETH/ethernet component to conserve energy:

In ETH.H/CPP there is a `begin` method but no corressponding `end`method.
In the underlying Espressif esp-idf there is a `esp_eth_start` and a corresponding `esp_eth_stop`.

I'm using WT32_ETH01 and/or Olimex ESP32 POE mostly both are using the LAN 8720 PHY
Some have power pin some have not.

I observe that power raises from 60-80mA to 150mA when starting ETH (ethernet), but there is no way back.

Workaround so far is the set some static (flash/eeprom) markers and then reboot (not activating Ethernet). 
But this disturbes the other working tasks...

VojtechBartoska, 2022-04-07:
Hello @Typhoor, are you able to propose changes in PR?

playmiel, 2023-03-23:
Hello, I have the same problem, I would like to disable the Ethernet but I can't find any way to do it, do you have an idea?

aminzakizebarjad, 2023-05-06:
I think there is a way to do that   
`esp_eth_stop(ETH.eth_handle);`

Tarik2142, 2023-10-08:
https://github.com/espressif/esp-idf/blob/fcc43ef8ed8a4e03c3a795cb03ba560787254da6/components/esp_eth/test/test_emac.c#L193

VojtechBartoska, 2024-05-15:
Hello, for version 3.0.0 Networking was refactored. Can you please give it a try? 

I am for now closing this ticker as it is quite old, if needed, you can reopen it.

closed 2024-05-15 by VojtechBartoska

me-no-dev, 2025-03-07:
@hitecSmartHome here is the fix: https://github.com/espressif/arduino-esp32/pull/11048

hitecSmartHome, 2025-03-07:
Thank you! Will check it out!