espressif/arduino-esp32 #5951: esp32 deep sleep initiated by ext0 and ext1: why ext1 is not running

Paulsburg, 2021-11-30:
Make your question, not a Statement, inclusive. Include all pertinent information:

What you are trying to do  
Describe your system (Hardware, computer, O/S, core version, environment)  
Describe what is failing  
Show the shortest possible code that will duplicate the error  
Show the EXACT error message (it doesn't work is not enough)  
Then if someone is interested and knowledgeable you might get a answer.  All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
[ExceptionDecoder](https://github.com/me-no-dev/EspExceptionDecoder)

----------------------------- Remove above -----------------------------

### Hardware:
|||||||
|:---|---|---|---|---|---|
|<B>Board</B>|ESP32 Dev Module
|<B>IDE name</B>|Arduino IDE

### Description:
there is a switch from GPIO_NUM_25 to ground. The switch could be opened by a magnet.

case 1:
         the esp32 is not waking up, if <test = 1> and the switch is opened

case2:
         the esp32 is waking up, if <test = 2> and the switch is opened

this sketch just shows my current problem, its not part of the real program :-)

why, in case 1 the esp32 is not waking up?

### Sketch:  
```cpp

#include <Arduino.h>
#define wakeup_pin  GPIO_NUM_25

void setup() {
  Serial.begin(115200);
  delay(1000);
  
    pinMode(wakeup_pin, INPUT_PULLUP);
    gpio_pullup_en(wakeup_pin);
    gpio_pulldown_dis(wakeup_pin);

int test = 1;
uint64_t ext1_mask;

Serial.println("running case: " + String(test));

switch (test) {
   case (1):
      Serial.println("case 1 ext1"); 
      ext1_mask =  pow(2,wakeup_pin);
      esp_sleep_enable_ext1_wakeup(ext1_mask, ESP_EXT1_WAKEUP_ANY_HIGH);
      break;
  case (2):
     Serial.println("case 2 ext0"); 
     esp_sleep_enable_ext0_wakeup(wakeup_pin , HIGH);
     break;
  } // end of switch (case)

          Serial.println("############################################");
          Serial.println("     ESP32 now going into infinite deep sleep");
          Serial.println("                      wakeup only by GPIO");
          Serial.println("############################################");

  esp_deep_sleep_start();
  Serial.println("<<<<<< This must never be printed >>>>>>>>>");
} // end of setup()

void loop() {
}
```
### DEBUG:
```
21:12:33.840 -> running case: 2
21:12:33.840 -> case 2 ext0
21:12:33.840 -> ############################################
21:12:33.840 ->      ESP32 now going into infinite deep sleep
21:12:33.840 ->                       wakeup only by GPIO
21:12:33.840 -> ############################################

>>>> opening the switch <<<<

21:12:35.800 -> ets Jun  8 2016 00:22:57
21:12:35.800 -> 
21:12:35.800 -> rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
21:12:35.800 -> configsip: 0, SPIWP:0xee
21:12:35.800 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
21:12:35.800 -> mode:DIO, clock div:1
21:12:35.800 -> load:0x3fff0018,len:4
21:12:35.800 -> load:0x3fff001c,len:1216
21:12:35.800 -> ho 0 tail 12 room 4
21:12:35.800 -> load:0x40078000,len:10944
21:12:35.800 -> load:0x40080400,len:6388
21:12:35.800 -> entry 0x400806b4
21:12:36.880 -> running case: 2
21:12:36.880 -> case 2 ext0
...
...
```

Paulsburg, 2021-12-03:
If the Pin is connected to +3V by a 47k resistor, the reaction is as desired.     :-))

This changes the question to, why the program lines 

    gpio_pullup_en(wakeup_pin);
    gpio_pulldown_dis(wakeup_pin);

doesn´t activate the internal pullup resistors?

****

VojtechBartoska, 2022-04-07:
Hello @Paulsburg, is this still relevant? If yes, are you able to test this with v2.0.3-RC1?

closed 2022-04-21 by VojtechBartoska