espressif/arduino-esp32 #5952: Partition Table Problem!

hdpklm, 2021-11-30:
Hello.

I work in vscode with esp-android extension. windows 10
MCU: esp32 d2wd

I have a problem, when I set the ota_0 to 0xf1000 and ota_1 to 0xf2000. after build and upload, the MCU not work
but when I set ota_0 to 0xf10000 and ota_1 to 0xf3000. its work correctly.

this not work
```
# Name,		Type,	SubType,	Offset,			Size,		Flags
spiffs,		data,	spiffs,		0x09000		,	0x04000,
nvs,		data,	nvs,				,	0x02000,
app0,		app,	ota_0,				,	0xf1000,   <-----
app1,		app,	ota_1,				,	0xf2000,   <-----
nvs_key,	data,	nvs_keys,			,	0x01000,
otadata,	data,	ota,				,	0x02000,
```

this work
```
# Name,		Type,	SubType,	Offset,			Size,		Flags
spiffs,		data,	spiffs,		0x09000		,	0x04000,
nvs,		data,	nvs,				,	0x02000,
app0,		app,	ota_0,				,	0xf0000,   <-----
app1,		app,	ota_1,				,	0xf3000,   <-----
nvs_key,	data,	nvs_keys,			,	0x01000,
otadata,	data,	ota,				,	0x02000,
```

this is the error I get
```
rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
ets Jun  8 2016 00:22:57

```

chegewara, 2021-11-30:
This is arduino limitation. First app/ota partition has to be at offset `0x10000`.

VojtechBartoska, 2022-04-07:
@hdpklm sorry for asnwering you late, did you find a solution for your error?

hdpklm, 2022-04-08:
no, I dont know what is the problem. but finaly I use size "0xf0000" for each ota and the rest for spiffs

VojtechBartoska, 2022-05-03:
can be a solution for your use case. I'm closing this issue.

closed 2022-05-03 by VojtechBartoska