espressif/arduino-esp32 #5953: Unable to reconnect BT SPP after power cycling until deleted in phone and re-paired

MNGregO, 2021-12-01:
I am using the example titled _SerialToSerialBT_SSP_pairing_ with an Android phone (a stock Samsung Galaxy S7, running Android 8.0.0) with the app called _Serial Bluetooth Terminal_. When I first power up the ESP32 running the example, the _ESP32test_ Bluetooth device shows up when scanning for Bluetooth devices and I'm able to pair with it after accepting the code on my phone, and entering 'y' into the Arduino serial monitor.
I then open _Serial Bluetooth Terminal_ on the phone, select the _ESP32test_ device, click the connect button in the app and I'm able to send text back and forth from the ESP32 to the phone. All is good so far.
Here is the problem... If I power down the ESP32 and then go back into the _Serial Bluetooth Terminal_ app, I'm able the connect when I hit the connect button in the app, but I cannot send any text either direction anymore.
The only way I'm able to send text again is to tell the phone to delete _ESP32test_ from the list of paired devices and go through pairing again.

VojtechBartoska, 2022-04-07:
Hell, is this till relevant?

MNGregO, 2022-04-17:
It is to me - I've been stalled on the project I was going to use this for because it's too much to expect users to delete an re-pair every time they want to reconnect.

VojtechBartoska, 2022-10-10:
I'm closing this issue as expired due to no answer. If needed, you can reopen it. Thanks for understanding.

closed 2022-10-10 by VojtechBartoska