espressif/arduino-esp32 #5958: How to achieve lowest possible power consumption?

frogg1ven, 2021-12-01:
### Hardware:
Board:                                                      esp32 wroom 32e		
IDE name:						Platform.io
Computer OS:                                         Windows 10

### Description:
Hello,

I'm working on project with esp32 wroom 32e and lora e22, trying to transmit data to another device and then go to deep sleep for 15 minutes and repeat endlessly.
I'm trying to achieve lowest possible power consumption and I managed to get 0.15 mA, but it is not enough. In some forum I found very interesting thing, that some esp32 GPIOs have high state on deep sleep and power consupmpion may be higher, so I have tried to find out which GPIOs are high simply by resseting them,  and then I commented line by line.
```c
  gpio_reset_pin(GPIO_NUM_0);
  gpio_reset_pin(GPIO_NUM_2);
  gpio_reset_pin(GPIO_NUM_4);
  gpio_reset_pin(GPIO_NUM_5);
  gpio_reset_pin(GPIO_NUM_12);
  gpio_reset_pin(GPIO_NUM_13);
  gpio_reset_pin(GPIO_NUM_14);
  gpio_reset_pin(GPIO_NUM_15);
  gpio_reset_pin(GPIO_NUM_16);
  gpio_reset_pin(GPIO_NUM_17);
  gpio_reset_pin(GPIO_NUM_18);
  gpio_reset_pin(GPIO_NUM_19);
  gpio_reset_pin(GPIO_NUM_21); 
  gpio_reset_pin(GPIO_NUM_22); 
  gpio_reset_pin(GPIO_NUM_23);
  gpio_reset_pin(GPIO_NUM_25); 
  gpio_reset_pin(GPIO_NUM_26);
  gpio_reset_pin(GPIO_NUM_27);
  gpio_reset_pin(GPIO_NUM_32);
  gpio_reset_pin(GPIO_NUM_33);
  gpio_reset_pin(GPIO_NUM_34);
  gpio_reset_pin(GPIO_NUM_35);
  gpio_reset_pin(GPIO_NUM_36);
  gpio_reset_pin(GPIO_NUM_39);
```
I found out that, with this:
```c
  gpio_reset_pin(GPIO_NUM_21); 
  gpio_reset_pin(GPIO_NUM_22); 
  gpio_reset_pin(GPIO_NUM_25); 
```
Power consumption was very low (about 0.02 mA, but it was constant, even after wake up). So I tried `gpio_isolate_pin(); ` for those GPIOs, but it didn't work. At 0.02 mA serial monitor is showing messages but with this power consuption it is impossible for esp32 to work when its awake.
I was also trying to disable CLR like this:
```c
REG_CLR_BIT(RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN);
```
but this didn't seem to work.
My question is how can I get power consumption <=0.02 mA using those GPIOs or any other way?

### Sketch: 
```c
#include <Arduino.h>
#include <RadioLib.h>
#include <driver/rtc_io.h>
#include "soc/rtc_cntl_reg.h"
#include "soc/rtc.h"

#define uS_TO_S_FACTOR 1000000
SX1262 radio = new Module(5, 33, 27, 32);

void setup() {
  setCpuFrequencyMhz(40);
  Serial.begin(9600);

  Serial.print(F("[SX1262] Initializing ... "));
  int state = radio.begin();
  radio.setFrequency(869.5);
  radio.setBandwidth(125.0);
  radio.setSpreadingFactor(12);
  radio.setOutputPower(22);
  
  if (state == ERR_NONE) {
    Serial.println(F("success!"));
  } else {
    Serial.print(F("failed, code "));
    Serial.println(state);
    while (true);
  }

  Serial.println("lora sleep");
  radio.standby();
  radio.sleep();
  delay(3000);
  
  REG_CLR_BIT(RTC_CNTL_STATE0_REG, RTC_CNTL_ULP_CP_SLP_TIMER_EN);
  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);

  pinMode(5, INPUT);

  // gpio_reset_pin(GPIO_NUM_0);
  // gpio_reset_pin(GPIO_NUM_2);
  // gpio_reset_pin(GPIO_NUM_4);
  // gpio_reset_pin(GPIO_NUM_5);
  // gpio_reset_pin(GPIO_NUM_12);
  // gpio_reset_pin(GPIO_NUM_13);
  // gpio_reset_pin(GPIO_NUM_14);
  // gpio_reset_pin(GPIO_NUM_15);
  // gpio_reset_pin(GPIO_NUM_16);
  // gpio_reset_pin(GPIO_NUM_17);
  // gpio_reset_pin(GPIO_NUM_18);
  // gpio_reset_pin(GPIO_NUM_19);
  gpio_reset_pin(GPIO_NUM_21); 
  gpio_reset_pin(GPIO_NUM_22); 
  // gpio_reset_pin(GPIO_NUM_23);
  gpio_reset_pin(GPIO_NUM_25); 
  // gpio_reset_pin(GPIO_NUM_26);
  // gpio_reset_pin(GPIO_NUM_27);
  // gpio_reset_pin(GPIO_NUM_32);
  // gpio_reset_pin(GPIO_NUM_33);
  // gpio_reset_pin(GPIO_NUM_34);
  // gpio_reset_pin(GPIO_NUM_35);
  // gpio_reset_pin(GPIO_NUM_36);
  // gpio_reset_pin(GPIO_NUM_39);

  // rtc_gpio_isolate(GPIO_NUM_21);
  // rtc_gpio_isolate(GPIO_NUM_22);
  // rtc_gpio_isolate(GPIO_NUM_25);

  Serial.println("esp sleep");
  esp_sleep_enable_timer_wakeup(5 * uS_TO_S_FACTOR);
  esp_deep_sleep_start();
}
void loop() {}
```

MaxSvjatoha, 2021-12-02:
Interesting. How did you achieve 0.15mA to begin with? My deep sleep current is ~2-3mA on a WROOM board (like [this](https://www.amazon.com/Teyleten-Robot-ESP-WROOM-32-Development-Microcontroller/dp/B08246MCL5?th=1) one)

TD-er, 2021-12-02:
> Interesting. How did you achieve 0.15mA to begin with? My deep sleep current is ~2-3mA on a WROOM board (like [this](https://www.amazon.com/Teyleten-Robot-ESP-WROOM-32-Development-Microcontroller/dp/B08246MCL5?th=1) one)

One very important aspect here is the used voltage regulator and of course any connected chip (USB serial) or pull-up or pull-down resistor on the board.

The board you linked looks to be using an AMS1117, which is a very stable linear voltage regulator.
But it does have a rather high quiescent current, which is typically more than the 2 - 3 mA you're stating.
So I suspect your stated 2 - 3 mA current is probably not correct.
The problem with most voltage regulators with a low quiescent current is that they can't handle 500 mA or more, which is needed for an ESP. 
And another problem is that most regulators that do are currently hard to obtain in quantities when developing a board design.

frogg1ven, 2021-12-02:
>Interesting. How did you achieve 0.15mA to begin with? My deep sleep current is ~2-3mA on a WROOM board (like this one)

@MaxSvjatoha, I also had high power consumption because Lora e22 has to have NSS pin on high state to go to sleep, and during esp deep sleep GPIO (5 in my case) has low state so it wake up Lora and power consumption is rising. Also programmator should be disconnected from esp during measurement, because it raises power consumption. 
So maybe if you are using esp32 with other devices you have to make sure they are sleeping too.

VojtechBartoska, 2022-04-07:
@frogg1ven Is this still relevant?

VojtechBartoska, 2022-04-08:
Thanks for your answer, closing as expired. Hopefully see you in this repository in future as well.

closed 2022-04-14 by VojtechBartoska