espressif/arduino-esp32 #5965: REG_SPI_BASE used in ESP32 C3 soc/spi_reg.h not defined

Bodmer, 2021-12-02:
I think it should be defined as:
```
#define REG_SPI_BASE(i)     (DR_REG_SPI1_BASE + (((i)>1) ? (((i)* 0x1000) + 0x20000) : (((~(i)) & 1)* 0x1000 )))
```
Maybe this is an IDF issue?

Jason2866, 2021-12-03:
Yes, it is missing we us this for the C3 since we support it for Tasmota
```
#ifndef REG_SPI_BASE
#define REG_SPI_BASE(i)     (DR_REG_SPI1_BASE + (((i)>1) ? (((i)* 0x1000) + 0x20000) : (((~(i)) & 1)* 0x1000 )))
// SPI_MOSI_DLEN_REG is not defined anymore in esp32c3, instead use SPI_MS_DLEN_REG
#define SPI_MOSI_DLEN_REG(x) SPI_MS_DLEN_REG(x)
#endif // REG_SPI_BASE
```

Jason2866, 2021-12-03:
Btw you should add this to make the riscv toolchain fully compatible to the others.
```
/*
 * Xtensa toolchain declares `int32_t` as `int` but RISC-V toolchain
 * declares `int32_t` as `long int` which causes compilation errors.
 *
 * See:
 *   https://github.com/espressif/esp-idf/issues/6906
 *   https://github.com/espressif/arduino-esp32/issues/5086
 *
 */
#ifdef __riscv

#undef __INT32_TYPE__
#define __INT32_TYPE__      int

#undef __UINT32_TYPE__
#define __UINT32_TYPE__     unsigned int

#endif // __riscv
```

atanisoft, 2021-12-04:
@Bodmer This would need to be reported over on IDF side. There is likely a better approach to use than direct register manipulation (such as the HAL APIs) since register names can change from one version of IDF to another (especially in IDF v5 which is on esp-idf master now)

Bodmer, 2021-12-04:
I looked at the HAL API but [noted this ](https://github.com/espressif/esp-idf/blob/master/components/hal/esp32s3/include/hal/spi_ll.h#L17)in the header.  Or is this a WIP and eventually there will be consistent HAL for access for the ESP32 variants?

Bodmer, 2021-12-04:
This is the [referenced ReadMe](https://github.com/espressif/esp-idf/blob/master/components/hal/include/hal/readme.md) about not using HAL.

closed 2021-12-04 by Bodmer

atanisoft, 2021-12-04:
> Or is this a WIP and eventually there will be consistent HAL for access for the ESP32 variants?

It's not typically intended for external usage but in some cases it is slightly better than using raw register access which will vary based on target. Ideally you would be using the driver API directly but there are a few exceptions where the drivers do not expose the necessary functionality (TWAI being one in my use case).