espressif/arduino-esp32 #5966: old LORA_RST pin definition in variants/ttgo-lora32-v21new/

VeniceInventors, 2021-12-03:
In the [pins_arduino.h](https://github.com/espressif/arduino-esp32/blob/399f4ecbb3a4cef21e2bffa37adb6190356dfb76/variants/ttgo-lora32-v21new/pins_arduino.h#L30) pin definition file, at line 30 the reset pin is set to 12 but according to [Lilygo's site](http://www.lilygo.cn/prod_view.aspx?TypeId=50060&Id=1270) and this [pinout diagram](https://forum.arduino.cc/t/identify-ttgo-lora32-v2-1-6-digital-pin-and-analog-pin/874755), the reset pin should be set to 23.

```
< #define LORA_RST    12  // GPIO14 - SX1276 RST
> #define LORA_RST    23  // GPIO23 - SX1276 RST
```

atanisoft, 2021-12-03:
@tobozo can you include this update in your PR?

tobozo, 2021-12-03:
I hope this time the CI won't need 24h to finish its job :rofl: 

[image]

tobozo, 2021-12-10:
see PR [#5961](https://github.com/espressif/arduino-esp32/pull/5961)

atanisoft, 2021-12-14:
@me-no-dev this one can be closed since PR #5961 has been merged.

me-no-dev, 2021-12-14:
Thanks @atanisoft

closed 2021-12-14 by me-no-dev