espressif/arduino-esp32 #5967: SD card fail to mount

davidkleng, 2021-12-03:
### Hardware:
Board:							ESP32-WROOM-32U
Core Installation version:			platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
IDE name:						Platform.io
Flash Frequency:					40Mhz
Upload Speed:					115200
Computer OS:                                          Windows 11
ESP sdk version:                                       v4.4-dev-3703-gddc44956bf

### Description:
Im trying to initialize a SD card in SPI port. 

### Sketch: 
platformio.ini file:
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board = esp32dev
framework = arduino
monitor_speed = 115200
build_type = debug
build_flags = 
	${env.build_flags} 
	-D=${PIOENV} 
        -DCORE_DEBUG_LEVEL=5
upload_port = COM3
monitor_port = COM3
monitor_filters = esp32_exception_decoder
board_build.partitions = default_16MB.csv
board_upload.flash_size = 16MB
board_upload.maximum_size = 16777216
; board_build.flash_mode = qio
extra_scripts = replace_fs.py
board_build.filesystem = littlefs
platform_packages =
    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
    espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r1
    toolchain-xtensa32@~2.80400.0  
lib_deps = 
    https://github.com/adafruit/Adafruit_ADS1X15.git
    https://github.com/adafruit/RTClib.git	
    https://github.com/taranais/NTPClient.git
    https://github.com/xreef/PCF8574_library
    https://github.com/yaacov/ArduinoModbusSlave
    https://github.com/yubox-node-org/AsyncTCPSock
    https://github.com/yubox-node-org/ESPAsyncWebServer
lib_ignore =
    AsyncTCP
```

The code I wrote: 
```
#include "FS.h"
#include "SD.h"
#include <SPI.h>

#define SD_CS 5
#define SD_SCK 18
#define SD_MISO 19
#define SD_MOSI 23

    if (!SD.begin(SD_CS))
    {
        if (serialDebug)
            Serial.println("SD CARD: INITIALIZATION FAILED");
        delay(1000);
    }
    else
    {
        SDMounted = true;
        if (serialDebug)
            Serial.println("SD CARD: Card Mount OK");
    }
```

also, i tried with no luck to start the spi first. 

The serial output of the code is:
```
[  3798][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  3798][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[  3798][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[  3803][E][sd_diskio.cpp:790] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  4112][W][sd_diskio.cpp:104] sdWait(): Wait Failed
[  4112][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
```

VojtechBartoska, 2021-12-03:
@P-R-O-C-H-Y PTAL, thanks!

davidkleng, 2021-12-03:
I get it to work, changing the Platformio.ini file:

```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board = esp32dev
framework = arduino
monitor_speed = 115200
build_type = debug
; build_flags = 
; 	${env.build_flags} 
; 	-D=${PIOENV} 
;     -DCORE_DEBUG_LEVEL=5
upload_port = COM3
monitor_port = COM3
monitor_filters = esp32_exception_decoder
board_build.partitions = default_16MB.csv
board_upload.flash_size = 16MB
board_upload.maximum_size = 16777216
; board_build.flash_mode = qio
extra_scripts = replace_fs.py
board_build.filesystem = littlefs
platform_packages =
    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
    espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r1
    toolchain-xtensa32@~2.80400.0  
lib_deps = 
    https://github.com/adafruit/Adafruit_ADS1X15.git
    https://github.com/adafruit/RTClib.git	
    https://github.com/taranais/NTPClient.git
    https://github.com/PaulStoffregen/OneWire.git
    https://github.com/xreef/PCF8574_library
    https://github.com/yaacov/ArduinoModbusSlave
    https://github.com/yubox-node-org/AsyncTCPSock
    https://github.com/yubox-node-org/ESPAsyncWebServer
lib_ignore =
    AsyncTCP
```

just commented the 
```
build_flags = 
	${env.build_flags} 
	-D=${PIOENV} 
        -DCORE_DEBUG_LEVEL=5
```
and it worked just fine, no idea why. Maybe someone can explain it.

P-R-O-C-H-Y, 2021-12-07:
Hi @davidkleng,
when using custom pins for Sd card, you need to init the SPI first with custom pins before calling `SD.begin(SD_CS);`

Add this line before calling SD.begin:
`SPI.begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS);`

davidkleng, 2021-12-07:
@P-R-O-C-H-Y , thanks for your help. 

I tried that, problem persists. 
Also, the SD.begin() function does SPI.begin()

VojtechBartoska, 2021-12-15:
Hello @davidkleng, seems it could be problem with PlatformIO set up of your environment. Please take a look on this issue #6021, it can be relevant.

bhupiister, 2021-12-15:
Try
```

   #define SCLK_PIN 18
   #define MISO_PIN 38
   #define MOSI_PIN 23
   #define SS_PIN 4

    SPIClass * hspi = NULL;
    hspi = new SPIClass(SPI);
    hspi->begin(SCLK_PIN, MISO_PIN, MOSI_PIN, SS_PIN);
    SD.begin(SS_PIN, *hspi *, 10000000);
```
or simply try

`SD.begin(SS_PIN, SPI, 10000000);`

davidkleng, 2021-12-20:
I tried 

```
// Define CS pin for the SD card module
#define SD_CS 5
#define SCLK_PIN 18
#define MISO_PIN 19
#define MOSI_PIN 23
```

```// Initialize SD card
  SPIClass *hspi = NULL;
  hspi = new SPIClass(SPI);
  hspi->begin(SCLK_PIN, MISO_PIN, MOSI_PIN, SD_CS);
  if (!SD.begin(SD_CS, *hspi, 10000000))
  {
    Serial.println("Card Mount Failed");
    return;
  }```

It didn't work. 

Serial output:

`[   331][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[   331][E][sd_diskio.cpp:790] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[   634][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed`

davidkleng, 2021-12-20:
It works with @bhupiister code

> Try
>
> ```
>
>    #define SCLK_PIN 18
>    #define MISO_PIN 38
>    #define MOSI_PIN 23
>    #define SS_PIN 4
>
>     SPIClass * hspi = NULL;
>     hspi = new SPIClass(SPI);
>     hspi->begin(SCLK_PIN, MISO_PIN, [...]

closed 2021-12-20 by davidkleng

ADolbyB, 2023-06-21:
> Hi @davidkleng, when using custom pins for Sd card, you need to init the SPI first with custom pins before calling `SD.begin(SD_CS);`
> 
> Add this line before calling SD.begin: `SPI.begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS);`

This worked for me. Many Thanks!!