espressif/arduino-esp32 #5969: Default value for sdkconfig parm CONFIG_HTTPD_MAX_REQ_HDR_LEN=512 too small in Arduino

craigdetter, 2021-12-03:
Using a form method="post" from a web page to the ESP32 device (using SoftAP, and web_server built from file_server example). 
     _&lt;form id="setup-form" action="/domain/setup_form_content" method="post"&gt; ... &lt;/form&gt;_

Causes error condition with message:
**_"Header fields are too long for server to interpret"_**

The header length of a Wireshark trace (see below) shows 554 byte long header being sent by Google Chrome.

### Hardware:
Board:							ESP32 Adafruit Huzzah32 
Core Installation version:			v5.0-dev-553-g42cce06704-dirty
IDE name:						Arduino IDE or VSCode
Flash Frequency:				       N/A
PSRAM enabled:                                     N/A
Upload Speed:				       N/A
Computer OS:                                         Mac OSX Catalina latest with Google Chrome latest version

### Description:
Describe your problem here

The default in build for the sdkconfig configuration parameter is: 
      CONFIG_HTTPD_MAX_REQ_HDR_LEN=512  (bytes).
It needs to be set at least 768 bytes, but 1024 would be better.

This is a problem for all development platforms and ESP32 targets, but a real burden for customers using Arduino because the binary isn't easy to recompile.

On Google Chrome latest version running on MacOS on the latest Catalina OS:
using Wireshark trace - shows **the "post" header is 554 bytes long (not incl. CR/LF)** 

_Host: 192.168.4.1\r\n
Connection: keep-alive\r\n
Content-Length: 95\r\n
Cache-Control: max-age=0\r\n
Upgrade-Insecure-Requests: 1\r\n
Origin: http://192.168.4.1\r\n
Content-Type: application/x-www-form-urlencoded\r\n
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/96.0.4664.55 Safari/537.36\r\n
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9\r\n
Referer: http://192.168.4.1/\r\n
Accept-Encoding: gzip, deflate\r\n
Accept-Language: en-US,en;q=0.9,pl;q=0.8\r\n
\r\n_

On Chrome "get" header is at least 478 bytes ....

_Host: 192.168.4.1\r\n
Connection: keep-alive\r\n
Upgrade-Insecure-Requests: 1\r\n
User-Agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/96.0.4664.55 Safari/537.36\r\n
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,image/avif,image/webp,image/apng,*/*;q=0.8,application/signed-exchange;v=b3;q=0.9\r\n
Referer: http://192.168.4.1/\r\n
Accept-Encoding: gzip, deflate\r\n
Accept-Language: en-US,en;q=0.9,pl;q=0.8\r\n
\r\n

**Wireshark Trace - Packet 218 ("post" from Chrome to ESP32):**
[image]

**Wireshark Trace - Packet 219 (response from ESP32 to Chrome):**
[image]

_

atanisoft, 2021-12-04:
@craigdetter Submit a PR over on https://github.com/espressif/esp32-arduino-lib-builder to modify the value(s) for this and if accepted it will make it's way over here.

VojtechBartoska, 2021-12-06:
relates to https://github.com/espressif/esp32-arduino-lib-builder/issues/52

LenShustek, 2022-01-14:
This problem has been talked about for *years* in various forums. Will it ever get fixed??

me-no-dev, 2022-01-19:
nice... this issue is here since December. And is already fixed in the lib-builder. will land here shortly.

closed 2022-01-19 by me-no-dev

LenShustek, 2022-01-19:
Much appreciated, thanks!
But just for the record, it had been reported as early as July 2019; see https://github.com/espressif/arduino-esp32/issues/2983.

LenShustek, 2022-01-25:
It appears to be fixed in idf-release/v4.4, but can you give us newbies some guidance about using that on Windows? 
The installation instructions say to use the Arduino IDE Boards Manager, but it needs a link to a file like package_esp32_index.json that I can't find in the V4.4 repo. 
When I instead try to clone the repo in GitHub desktop I get Authentication failed for github.com/espressif/arduino-esp32/tree, and my username and password doesn't cut it.

atanisoft, 2022-01-25:
> can you give us newbies some guidance about using that on Windows?

https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

All you need to do is use the correct boards manager URL in Arduino IDE and install v2.0.x and it should work without issues. You don't need GitHub Desktop or other tools unless you need bleeding edge (not recommended in most cases).

LenShustek, 2022-01-25:
Doesn't that give me V2.0.2 from Dec 23, 2021, and not the idf-release/v4.4 from 6 days ago that is 2 commits ahead of master?

atanisoft, 2022-01-25:
Ahh, yes that is true. The changes on idf-release/v4.4 are temporary in nature and will be removed once PR https://github.com/espressif/arduino-esp32/pull/6174 has been merged to master. It would be best to wait for that to be merged before trying to pick up the updated version.

LenShustek, 2022-01-26:
Ok, thanks; I'll keep waiting. I tried all sort of bizarre ways to get the Arduino IDE to be happy with the idf-release/v4.4 that I could download, but without success.

fillybrese, 2022-02-14:
@LenShustek you can go to your Arduino sketch directory and:

```
mkdir ./hardware/espressif &&
cd ./hardware/espressif &&
git clone -b idf-release/v4.4 https://github.com/espressif/arduino-esp32.git esp32 && 
git checkout 533c51c593355616686488318a99cc892c14e881
cd ./esp32/tools &&
python get.py
```

atanisoft, 2022-02-18:
@LenShustek it is available now on master, if you are using Arduino IDE you would need to follow the instructions for git clone as documented here: https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#windows (scroll down for Linux or MacOS). You will not be able to use the boards manager approach until the next release.

atanisoft, 2022-02-18:
Sorry you are right. `master` is still using 512, you can pull the updated values from branch `idf-release/v4.4`.

A minor update to the previously linked instructions:
Replace `git clone https://github.com/espressif/arduino-esp32.git esp32` with
`git clone --branch idf-release/v4.4 https://github.com/espressif/arduino-esp32.git esp32`

If you are using a graphical Git client you should be able to specify the branch to clone as `idf-release/v4.4`