espressif/arduino-esp32 #5971: Cannot re-init BLEDevice(). Keep getting Error panic'ed (StoreProhibited).

kzka90, 2021-12-04:
### Hardware:
Board:							**ESP32 Dev Module**
Core Installation version:			**1.0.6**
IDE name:						**Arduino IDE**
Flash Frequency:					**80Mhz**
PSRAM enabled:                                     **no**
Upload Speed:					**921600**
Computer OS:                                        **Windows 10**

### Description:

I'm trying to switch between Wi-Fi and BLE without needing to restart everytime I want to do so. I want to be able to fully deinit every library before using the other one. But it keeps crashing when transitioning from WiFi to BLE.

### Sketch: 
Here is how I am stopping and starting my variables:
```
void startWifi(){
  WiFi.begin("SSID","PASS");
  while (WiFi.status() != WL_CONNECTED) {
    delay (500);
  }
}

void stopWifi(){
    WiFi.disconnect(true);
    WiFi.mode(WIFI_MODE_NULL);
}

void startBLE(){
  BLEDevice::init("Device");
  if(pServer==NULL){
    pServer = BLEDevice::createServer();
    pServer->setCallbacks(new MyServerCallbacks());
    BLEService* pService = pServer->createService(BLEUUID((uint16_t)0x181A));
    binDescriptor.setValue("Write");
    binCharacteristic.addDescriptor(&binDescriptor);
    binCharacteristic.setCallbacks(new MyCallbacks());
    pService->addCharacteristic(&binCharacteristic);

    versionDescriptor.setValue("Version");
    versionCharacteristic.addDescriptor(&versionDescriptor);
    versionCharacteristic.setCallbacks(new MyCallbacks());
    versionCharacteristic.setValue("1");
    pService->addCharacteristic(&versionCharacteristic);

    pService->start();
    pServer->getAdvertising()->start();
    delay(500);
  } 
}

void stopBLE(){
     BLEDevice::deinit(false);
     delete pServer;
     pServer = NULL;
}
```

I've been able to go from BLE to WiFi with success. But once I get to WiFi I cannot re-init the BLE without restarting.

I've also tried to denit the BLE with the memory release flag as true:

`BLEDevice::deinit(false);`

But no luck...

Digging deeper in the **BLEDevice.cpp** file I've found that the crash happens exactly at this line:

`errRc = esp_bluedroid_init(); //HERE IS WHEN IT CRASHES` which is inside the `BLEDevice::init` method.

```
esp_bluedroid_status_t bt_state = esp_bluedroid_get_status();
if (bt_state == ESP_BLUEDROID_STATUS_UNINITIALIZED) {
	errRc = esp_bluedroid_init(); //HERE IS WHEN IT CRASHES
	if (errRc != ESP_OK) {
		log_e("esp_bluedroid_init: rc=%d %s", errRc, GeneralUtils::errorToString(errRc));
		return;
	}
}
```

### Debug Messages:

```
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4000c46c  PS      : 0x00060430  A0      : 0x8013f332  A1      : 0x3fff7720  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00002558  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000255  A8      : 0x80082020  A9      : 0x3fff76c0  
A10     : 0x00000000  A11     : 0x00001800  A12     : 0x00000000  A13     : 0x00000003  
A14     : 0x00001800  A15     : 0x00000004  SAR     : 0x00000000  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000254  

ELF file SHA256: 0000000000000000

Backtrace: 0x4000c46c:0x3fff7720 0x4013f32f:0x3fff7730 0x40134b0e:0x3fff7750 0x401347ab:0x3fff7770 0x401347eb:0x3fff7790 0x40090a4e:0x3fff77c0
```

Here is my decoded exception:

```
PC: 0x4000c46c
EXCVADDR: 0x00000000

Decoding stack results
0x4013f32f: l2c_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/l2cap/l2c_main.c line 831
0x40134b0e: btu_init_core at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_init.c line 89
0x401347ab: btu_task_start_up at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c line 278
0x401347eb: btu_task_thread_handler at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c line 226
0x40090a4e: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

VojtechBartoska, 2022-04-08:
Hello, sorry for answering so late. You run your code on v1.0.6 which is now old, can you please test this again on v2.0.3-rc1?

VojtechBartoska, 2022-05-05:
Hello,

due to no answer I'm closing this issue as expired.

If needed, please reopen it.

Thanks for understanding.

closed 2022-05-05 by VojtechBartoska