espressif/arduino-esp32 #5972: I2C Wire.requestFrom() takes way to long on error

pgrawehr, 2021-12-04:
This is probably related to #5875

I'm using `Wire.requestFrom(address, 1);` also to scan the I2C bus. If it returns 1, a device is present, otherwise not. For some reason, the current version (both the official 2.0.1 release as well as the latest master including the fix from #5910) take very long in case of an error. Consider this example program (slightly modified from the default WireScan example):

```cpp
#include "Wire.h"

void setup() {
  Serial.begin(115200);
  Wire.begin();
  Serial.printf("Wire timeout: %dms", Wire.getTimeOut());
}

void loop() {
  byte error = 0;
  byte address = 0;
  int nDevices = 0;

  delay(5000);

  Serial.println("Scanning for I2C devices ...");
  for(address = 0x01; address < 0x7f; address++){
    // Wire.beginTransmission(address);
    // error = Wire.endTransmission();
    error = Wire.requestFrom(address, (byte)1) == 1 ? 0 : 2;
    if (error == 0){
      Serial.printf("I2C device found at address 0x%02X\n", address);
      nDevices++;
    } else if(error != 2){
      Serial.printf("Error %d at address 0x%02X\n", error, address);
    }
    else
    {
      Serial.printf("No device found at address 0x%02X\n", address);
    }
  }
  if (nDevices == 0){
    Serial.println("No I2C devices found");
  }
}
```

With the default implementation (using the two commented lines above) the scan is fast and one loop takes around 2 seconds. The above code, however, takes about 1 second for each address it scans (except for the addresses where a device is present). The timeout for the I2C bus is set to the default of 50ms. 

### Expected Behavior

`Wire.readFrom(address, 1);` takes 50ms in case there's no device at the given address.

### Actual Behavior

It takes about 1 second.

### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			Tested 2.0.1 and 399f4ecbb3a4cef21e2bffa37adb6190356dfb76. The problem did not occur with 1.0.6
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Windows 10

pgrawehr, 2021-12-07:
Just found that this is a possible duplicate of #5934 .

VojtechBartoska, 2022-04-06:
@pgrawehr yes, thanks for finding this. Let's close this one and follow mentioned issue above as it's already in our Roadmap.

closed 2022-04-06 by VojtechBartoska