espressif/arduino-esp32 #5974: GURU Meditation Error for strcmp() while parsing JSON from AWS IOT CORE!

reddy9698, 2021-12-04:
### Description:
Hi,
I'm trying to do a simple IOT automation with AWS IOT CORE.
How it basically works is I update the thing shadow to the desired state.
The ESP32-WROOM is subscribed to the delta of the shadow.
Once the ESP receives the data and the output of the GPIO is updated it reports back the state of the GPIO to the reported state of the same Shadow on the IOT core.

The shadow document looks like this : `{
  "desired": {
    "status": "on",
    "light1": "on",
    "light2": "off",
    "light3": "on"
  },
  "reported": {
    "status": "on",
    "light1": "off",
    "light2": "off",
    "light3": "on"
  }
}`
#########################################################################
and this is the cpp code that I use : 

### Sketch: 
`if(msgReceived == 1)
    {
        delay(100);
        msgReceived = 0;
        Serial.print("Received Message:");
        Serial.println(rcvdPayload);
        StaticJsonDocument<512> doc;
        DeserializationError error = deserializeJson(doc, rcvdPayload);
        

        if (error) {
           Serial.print("deserializeJson() failed: ");
           Serial.println(error.c_str());
           
           return;
        }

        const char* light1_state = doc["state"]["light1"];
        const char* light2_state = doc["state"]["light3"];        
               
        Serial.print("AWS Says:");
        Serial.println(light2_state);
        

        if(strcmp(light1_state, "on") == 0)
        {
         
         Serial.println("Turning Lamp1 On");
         digitalWrite(0, HIGH);
         client.publish(AWS_IOT_PUBLISH_TOPIC, sndPayloadOn);
         
        }
        else if(strcmp(light1_state,"off") == 0)
        {
         
         Serial.println("Turning Lamp1 Off");
         digitalWrite(0, LOW);
         client.publish(AWS_IOT_PUBLISH_TOPIC, sndPayloadOff);
        }

        else if(strcmp(light2_state,"on") == 0)
        {
         
         Serial.println("Turning Lamp2 On");
         digitalWrite(0, LOW);
         client.publish(AWS_IOT_PUBLISH_TOPIC, sndPayloadOn2);
        }
        
        else if(strcmp(light2_state,"off") == 0)
        {
         
         Serial.println("Turning Lamp2 Off");
         digitalWrite(0, LOW);
         client.publish(AWS_IOT_PUBLISH_TOPIC, sndPayloadOff2);
        }
   }`

**###Problem:
The code runs perfectly fine when I work with light1 and the ESP32 reacts accordingly without any problems.
But when I try to update light2 or light3 from the IOT core the ESP32 runs into a GURU Meditation error as follows :**

`Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x40001277  PS      : 0x00060a30  A0      : 0x800d1444  A1      : 0x3ffb1b20
A2      : 0x00000000  A3      : 0x3f41d256  A4      : 0x3ffb1a10  A5      : 0x3ffc6b1c
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x0000006e  A9      : 0x00000066
A10     : 0x00000003  A11     : 0x00bb1835  A12     : 0x00000000  A13     : 0x0000000a
A14     : 0x00000001  A15     : 0x3ffb0060  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400012c5  LEND    : 0x400012d5  LCOUNT  : 0xfffffffe

ELF file SHA256: 0000000000000000

Backtrace: 0x40001277:0x3ffb1b20 0x400d1441:0x3ffb1b30 0x400d5364:0x3ffb1fb0 0x40089852:0x3ffb1fd0`

And after debugging this is what it shows : 

### Debug Messages:
`C: 0x40001277
EXCVADDR: 0x00000000

Decoding stack results
0x400d11df: loop() at C:\Users\Administrator\Documents\Arduino\basic_aws_test/basic_aws_test.ino line 261
0x400d4c68: loopTask(void*) at C:\Users\Administrator\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\main.cpp line 23
0x4008981a: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
`

The line no 261 is  :  if(strcmp(light1_state, "on") == 0)
and it only malfunctions when light 2 or light 3 values are updated.

###################################################################################

### Hardware:
Board:							ESP32 -WROOM
IDE name:						Platform.io
Flash Frequency:					80Mhz
PSRAM enabled:                                     yes
Upload Speed:					115200
Computer OS:                                        Windows 10

atanisoft, 2021-12-04:
Your json payload is not matching what you expect and you are trying to compare a null pointer to a string constant. You should verify the json payload prior to using it. ArduinoJson even offers a few methods for doing just this and warns that accessing a non-existing element (or invalid conversion) will result in a null value.

reddy9698, 2021-12-06:
> Your json payload is not matching what you expect and you are trying to compare a null pointer to a string constant. You should verify the json payload prior to using it. ArduinoJson even offers a [...]

if I try the code for updating just the light1 or light2 it's working fine. The problem is when I try to update light1 and light2 at the same time. 

   ` const char* light1_state = doc["state"]["light1"];`

if I only use this one line and change light1 to light2 it works without any error.

 ` const char* light1_state = doc["state"]["light1"];
    const char* light2_state = doc["state"]["light3"];  `
But when I try to the above method for two different things only then I'm getting the error according to what you've mentioned.
Any suggestions as to where I might be going wrong?

atanisoft, 2021-12-06:
As I noted previously, you are comparing a string constant against a null pointer which is why it crashes. Why you are getting a null pointer is something you will need to determine based on the json payload you are receiving and parsing via ArduinoJson. You should add some error checking to your code to verify that parameters are provided where you expect them to be.

reddy9698, 2021-12-06:
> As I noted previously, you are comparing a string constant against a null pointer which is why it crashes. Why you are getting a null pointer is something you will need to determine based on the [...]

Okay, I will try doing the same. Thank you so much for taking your time to reply.

closed 2021-12-10 by reddy9698