espressif/arduino-esp32 #5977: ESP-32: linker error WiFiClientSecure functions building with esp-idf

georgeman93, 2021-12-05:
### Hardware:
Board:							ESP32-WROOM-32
Core Installation version:			2.0.1
IDE name:						VSCode 1.62.3 ESP-IDF v4.4-dev
Computer OS:                                        MacOSX Big Sur 11.5.2

I am trying to build arduino-esp32 as an esp-idf component as per [these](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html) expressif docs. No device plugged in yet.

The WiFiClientSecure library isn't linking (or maybe compiling properly).

main.cpp
```
#include <WiFiClientSecure.h>
//file: main.cpp
#include "Arduino.h"

void setup(){
    Serial.begin(115200);
    WiFiClientSecure net = WiFiClientSecure();
}

void loop(){
    Serial.println("loop");
    delay(1000);
}
```

running idf.py build produces the following output 

```
> Executing task: cmake --build . <

[0/1] Re-running CMake...
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- globbed source files /Users/georgeoconnor/test/build/CMakeFiles/3.22.0/CompilerIdC/CMakeCCompilerId.c;/Users/georgeoconnor/test/build/CMakeFiles/3.22.0/CompilerIdCXX/CMakeCXXCompilerId.cpp;/Users/georgeoconnor/test/build/bootloader/CMakeFiles/3.22.0/CompilerIdC/CMakeCCompilerId.c;/Users/georgeoconnor/test/build/bootloader/CMakeFiles/3.22.0/CompilerIdCXX/CMakeCXXCompilerId.cpp;/Users/georgeoconnor/test/build/bootloader/project_elf_src_esp32.c;/Users/georgeoconnor/test/build/project_elf_src_esp32.c
-- Project sdkconfig file /Users/georgeoconnor/test/sdkconfig
-- App "test" version: 1
-- Adding linker script /Users/georgeoconnor/test/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /Users/georgeoconnor/esp/esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /Users/georgeoconnor/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /Users/georgeoconnor/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /Users/georgeoconnor/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /Users/georgeoconnor/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /Users/georgeoconnor/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /Users/georgeoconnor/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /Users/georgeoconnor/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-time.ld
-- Adding linker script /Users/georgeoconnor/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- globbed source files /Users/georgeoconnor/test/components/arduino-mqtt/src/MQTTClient.cpp;/Users/georgeoconnor/test/components/arduino-mqtt/src/lwmqtt/client.c;/Users/georgeoconnor/test/components/arduino-mqtt/src/lwmqtt/helpers.c;/Users/georgeoconnor/test/components/arduino-mqtt/src/lwmqtt/packet.c;/Users/georgeoconnor/test/components/arduino-mqtt/src/lwmqtt/string.c
-- Components: app_trace app_update arduino arduino-mqtt arduinojson asio bootloader bootloader_support bt cbor cmock coap console cxx driver efuse esp-tls esp32 esp_adc_cal esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_ipc esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_ringbuf esp_rom esp_serial_slave_link esp_system esp_timer esp_websocket_client esp_wifi espcoredump esptool_py expat fatfs freemodbus freertos hal heap idf_test ieee802154 jsmn json libsodium log lwip main mbedtls mdns mqtt newlib nghttp nvs_flash openssl openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash spiffs tcp_transport tcpip_adapter tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant xtensa
-- Component paths: /Users/georgeoconnor/esp/esp-idf/components/app_trace /Users/georgeoconnor/esp/esp-idf/components/app_update /Users/georgeoconnor/test/components/arduino /Users/georgeoconnor/test/components/arduino-mqtt /Users/georgeoconnor/test/components/arduinojson /Users/georgeoconnor/esp/esp-idf/components/asio /Users/georgeoconnor/esp/esp-idf/components/bootloader /Users/georgeoconnor/esp/esp-idf/components/bootloader_support /Users/georgeoconnor/esp/esp-idf/components/bt /Users/georgeoconnor/esp/esp-idf/components/cbor /Users/georgeoconnor/esp/esp-idf/components/cmock /Users/georgeoconnor/esp/esp-idf/components/coap /Users/georgeoconnor/esp/esp-idf/components/console /Users/georgeoconnor/esp/esp-idf/components/cxx /Users/georgeoconnor/esp/esp-idf/components/driver /Users/georgeoconnor/esp/esp-idf/components/efuse /Users/georgeoconnor/esp/esp-idf/components/esp-tls /Users/georgeoconnor/esp/esp-idf/components/esp32 /Users/georgeoconnor/esp/esp-idf/components/esp_adc_cal /Users/georgeoconnor/esp/esp-idf/components/esp_common /Users/georgeoconnor/esp/esp-idf/components/esp_eth /Users/georgeoconnor/esp/esp-idf/components/esp_event /Users/georgeoconnor/esp/esp-idf/components/esp_gdbstub /Users/georgeoconnor/esp/esp-idf/components/esp_hid /Users/georgeoconnor/esp/esp-idf/components/esp_http_client /Users/georgeoconnor/esp/esp-idf/components/esp_http_server /Users/georgeoconnor/esp/esp-idf/components/esp_https_ota /Users/georgeoconnor/esp/esp-idf/components/esp_https_server /Users/georgeoconnor/esp/esp-idf/components/esp_hw_support /Users/georgeoconnor/esp/esp-idf/components/esp_ipc /Users/georgeoconnor/esp/esp-idf/components/esp_lcd /Users/georgeoconnor/esp/esp-idf/components/esp_local_ctrl /Users/georgeoconnor/esp/esp-idf/components/esp_netif /Users/georgeoconnor/esp/esp-idf/components/esp_phy /Users/georgeoconnor/esp/esp-idf/components/esp_pm /Users/georgeoconnor/esp/esp-idf/components/esp_ringbuf /Users/georgeoconnor/esp/esp-idf/components/esp_rom /Users/georgeoconnor/esp/esp-idf/components/esp_serial_slave_link /Users/georgeoconnor/esp/esp-idf/components/esp_system /Users/georgeoconnor/esp/esp-idf/components/esp_timer /Users/georgeoconnor/esp/esp-idf/components/esp_websocket_client /Users/georgeoconnor/esp/esp-idf/components/esp_wifi /Users/georgeoconnor/esp/esp-idf/components/espcoredump /Users/georgeoconnor/esp/esp-idf/components/esptool_py /Users/georgeoconnor/esp/esp-idf/components/expat /Users/georgeoconnor/esp/esp-idf/components/fatfs /Users/georgeoconnor/esp/esp-idf/components/freemodbus /Users/georgeoconnor/esp/esp-idf/components/freertos /Users/georgeoconnor/esp/esp-idf/components/hal /Users/georgeoconnor/esp/esp-idf/components/heap /Users/georgeoconnor/esp/esp-idf/components/idf_test /Users/georgeoconnor/esp/esp-idf/components/ieee802154 /Users/georgeoconnor/esp/esp-idf/components/jsmn /Users/georgeoconnor/esp/esp-idf/components/json /Users/georgeoconnor/esp/esp-idf/components/libsodium /Users/georgeoconnor/esp/esp-idf/components/log /Users/georgeoconnor/esp/esp-idf/components/lwip /Users/georgeoconnor/test/main /Users/georgeoconnor/esp/esp-idf/components/mbedtls /Users/georgeoconnor/esp/esp-idf/components/mdns /Users/georgeoconnor/esp/esp-idf/components/mqtt /Users/georgeoconnor/esp/esp-idf/components/newlib /Users/georgeoconnor/esp/esp-idf/components/nghttp /Users/georgeoconnor/esp/esp-idf/components/nvs_flash /Users/georgeoconnor/esp/esp-idf/components/openssl /Users/georgeoconnor/esp/esp-idf/components/openthread /Users/georgeoconnor/esp/esp-idf/components/partition_table /Users/georgeoconnor/esp/esp-idf/components/perfmon /Users/georgeoconnor/esp/esp-idf/components/protobuf-c /Users/georgeoconnor/esp/esp-idf/components/protocomm /Users/georgeoconnor/esp/esp-idf/components/pthread /Users/georgeoconnor/esp/esp-idf/components/sdmmc /Users/georgeoconnor/esp/esp-idf/components/soc /Users/georgeoconnor/esp/esp-idf/components/spi_flash /Users/georgeoconnor/esp/esp-idf/components/spiffs /Users/georgeoconnor/esp/esp-idf/components/tcp_transport /Users/georgeoconnor/esp/esp-idf/components/tcpip_adapter /Users/georgeoconnor/esp/esp-idf/components/tinyusb /Users/georgeoconnor/esp/esp-idf/components/ulp /Users/georgeoconnor/esp/esp-idf/components/unity /Users/georgeoconnor/esp/esp-idf/components/usb /Users/georgeoconnor/esp/esp-idf/components/vfs /Users/georgeoconnor/esp/esp-idf/components/wear_levelling /Users/georgeoconnor/esp/esp-idf/components/wifi_provisioning /Users/georgeoconnor/esp/esp-idf/components/wpa_supplicant /Users/georgeoconnor/esp/esp-idf/components/xtensa
-- Configuring done
-- Generating done
[... cut ...]
/Users/georgeoconnor/test/components/arduino/libraries/WiFiClientSecure/src/WiFiClientSecure.cpp:152: undefined reference to `start_ssl_client(sslclient_context*, char const*, unsigned int, int, char const*, char const*, char const*, char const*, char const*, bool, char const**)'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
The terminal process "bash '-c', 'cmake --build .'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

```

The arduino CMakeLists.txt seems to include the right source files. I am not a pro at diagnosing linker issues and am not sure what to try next. 

Any ideas?

marcovannoord, 2022-01-05:
I have the same issue here, with:
Board: ESP32-WROOM-32
Core Installation version: 2.0.2 (via https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2idf/platform-espressif32-2.0.2.zip)
Computer OS: Windows 10
Python version: 3.9

georgeman93, 2022-01-09:
Not yet, will revisit soon

maximov-ru, 2022-01-24:
I have same error

bbsan2k, 2022-01-25:
Please check whether you enabled PSK ciphers for mbedtls - this fixed it for me!

johnboiles, 2022-03-17:
Thanks @bbsan2k you set me on the right track. Specifically I had to set `Enable PSK Verification`

[image]

VojtechBartoska, 2022-04-08:
Hello, this issue seems like a wrong combination of versions btw. ESP-IDF and Arduino core. Now version 2.0.0 and newest are compatible with ESP-IDF v4.4 for using Arduino as IDF component.

Can I consider this solved and close it?

VojtechBartoska, 2022-04-22:
Hello, due to no answer I'm closing this issue as expired.

If you still need help, please reopen it.

Thanks for understanding!

sanketa-54, 2022-05-25:
Hi @VojtechBartoska 
I'm facing the same issue .
please can you confirm which version of arduino-esp32 works for esp-idf 4.4.

me-no-dev, 2022-05-25:
@sanketa-54 2.0.3 and master both use/build with 4.4.1

closed 2022-06-14 by VojtechBartoska

johnboiles, 2023-06-10:
Just ran into this again in a new project. Had to go into menuconfig and set `Enable PSK Verification`

```
 - framework-arduinoespressif32 @ 3.20009.0 (2.0.9)
 - framework-espidf @ 3.40404.0 (4.4.4)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
 ```

Googling led me back to my solution above :)

wailay, 2025-03-30:
Also getting this in my new project when using the `WiFiClientSecure` library with

```
arduino-esp32 v3.2.0
ESP-IDF v5.4.0
```

wailay, 2025-04-25:
> > Also getting this in my new project when using the `WiFiClientSecure` library with
> > ```
> > arduino-esp32 v3.2.0
> > ESP-IDF v5.4.0
> > ```
> 
> [@wailay](https://github.com/wailay) Did you ever find a solution for this?

I had to enable `Enable PSK Verification`