espressif/arduino-esp32 #5981: ESP crashes when clients disconnects from softAP. "free() target pointer is outside heap areas"

davidkleng, 2021-12-06:
### Hardware:
Board:							ESP32-WROOM-32U
Core Installation version:			platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
IDE name:						Platform.io
Flash Frequency:					40Mhz
Upload Speed:					115200
Computer OS:                                          Windows 11
ESP sdk version:                                       v4.4-dev-3703-gddc44956bf

### Description:
The ESP32 is working as softAP. I can connect a client and use the seb server without any problem. 
When i disconnect the client, the ESP crashes and restarts.

### Sketch: 
platformio.ini file:
```
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board = esp32dev
framework = arduino
monitor_speed = 115200
build_type = debug
build_flags = 
	${env.build_flags} 
	-D=${PIOENV} 
        -DCORE_DEBUG_LEVEL=5
upload_port = COM3
monitor_port = COM3
monitor_filters = esp32_exception_decoder
board_build.partitions = default_16MB.csv
board_upload.flash_size = 16MB
board_upload.maximum_size = 16777216
; board_build.flash_mode = qio
extra_scripts = replace_fs.py
board_build.filesystem = littlefs
platform_packages =
    platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git
    espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r1
    toolchain-xtensa32@~2.80400.0  
lib_deps = 
    https://github.com/adafruit/Adafruit_ADS1X15.git
    https://github.com/adafruit/RTClib.git	
    https://github.com/taranais/NTPClient.git
    https://github.com/xreef/PCF8574_library
    https://github.com/yaacov/ArduinoModbusSlave
    https://github.com/yubox-node-org/AsyncTCPSock
    https://github.com/yubox-node-org/ESPAsyncWebServer
lib_ignore =
    AsyncTCP
```

WiFi int:

```
void initWiFi_AP(bool useStaticIP = false)
{
    if (serialDebug)
        Serial.println("SETUP: WIFI AP SETUP");

    WiFi.mode(WIFI_AP);

    delay(100);

    WiFi.softAP(APname.c_str(), APpass.c_str());

    delay(200);

    if (serialDebug)
        Serial.println("SETUP: WIFI DISABLE POWER SAVE");

    Serial.println(esp_wifi_set_ps(WIFI_PS_NONE));
    delay(100);

    Serial.print("AP NAME: ");
    Serial.println(APname.c_str());

    WiFi.setHostname(APname.c_str());
    WiFi.softAPsetHostname(APname.c_str());
    delay(100);

    if (useStaticIP)
        WiFi.softAPConfig(APip, APgateway, APsubnet);

    delay(100);

    if (serialDebug)
        Serial.println("");
    if (serialDebug)
        Serial.print("WIFI: Iniciado AP:\t");
    if (serialDebug)
        Serial.println(WiFi.softAPSSID());
    if (serialDebug)
        Serial.print("WIFI: IP address:\t");
    if (serialDebug)
        Serial.println(WiFi.softAPIP());
    if (serialDebug)
        Serial.print("WIFI: pass:\t\t");
    if (serialDebug)
        Serial.println(APpass.c_str());
}
```

### Debug Messages:
```
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 17 - AP_STADISCONNECTED
WEB SOCKET: client #2 disconnected
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c", line 267, function: heap_caps_free
abort() was called at PC 0x40123f1b on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x400895d8:0x3ffb1db0 0x40089855:0x3ffb1dd0 0x40123f1b:0x3ffb1df0 0x400829ef:0x3ffb1e20 0x4008769d:0x3ffb1e40 0x4000bec7:0x3ffb1e60 0x40179f91:0x3ffb1e80 0x400f6c01:0x3ffb1ea0 0x401859a3:0x3ffb1ec0 0x40100aff:0x3ffb1ee0 0x40100b5e:0x3ffb1f00 0x400e4b9c:0x3ffb1f50 0x400e4be9:0x3ffb1f70 0x40108595:0x3ffb1fb0 0x4008a866:0x3ffb1fd0
  #0  0x400895d8:0x3ffb1db0 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x40089855:0x3ffb1dd0 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #2  0x40123f1b:0x3ffb1df0 in __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c:63 (discriminator 8)
  #3  0x400829ef:0x3ffb1e20 in heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c:232
  #4  0x4008769d:0x3ffb1e40 in _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c:42
  #5  0x4000bec7:0x3ffb1e60 in ?? ??:0
  #6  0x40179f91:0x3ffb1e80 in operator delete(void*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/del_op.cc:46
  #7  0x400f6c01:0x3ffb1ea0 in AsyncWebSocket::~AsyncWebSocket() at .pio\libdeps\esp32dev\ESP Async WebServer\src/AsyncWebSocket.cpp:996
  #8  0x401859a3:0x3ffb1ec0 in std::_Function_handler<void (AsyncWebHandler* const&), AsyncWebServer::AsyncWebServer(unsigned short)::{lambda(AsyncWebHandler*)#2}>::_M_invoke(std::_Any_data const&, AsyncWebHandler* const&) at .pio\libdeps\esp32dev\ESP Async WebServer\src/WebServer.cpp:41
      (inlined by) _M_invoke at c:\users\david\.platformio\packages\toolchain-xtensa32@2.50200.97\xtensa-esp32-elf\include\c++\5.2.0/functional:1871
  #9  0x40100aff:0x3ffb1ee0 in std::function<void (AsyncWebHandler* const&)>::operator()(AsyncWebHandler* const&) const at .pio\libdeps\esp32dev\ESP Async WebServer\src/WebServer.cpp:85
  #10 0x40100b5e:0x3ffb1f00 in LinkedList<AsyncWebHandler*, LinkedListNode>::free() at .pio\libdeps\esp32dev\ESP Async WebServer\src/StringArray.h:166
      (inlined by) AsyncWebServer::reset() at .pio\libdeps\esp32dev\ESP Async WebServer\src/WebServer.cpp:190
  #11 0x400e4b9c:0x3ffb1f50 in restartServer() at src/main.cpp:92
  #12 0x400e4be9:0x3ffb1f70 in loop() at src/main.cpp:230
  #13 0x40108595:0x3ffb1fb0 in loopTask(void*) at C:\Users\david\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:23
  #14 0x4008a866:0x3ffb1fd0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

VojtechBartoska, 2022-04-07:
Hello, can you please validate this also with v2.0.3-RC1? Thanks

VojtechBartoska, 2022-04-22:
@davidkleng any news?

VojtechBartoska, 2022-05-04:
I'm closing this issue as expired due to no answer and mentioned Peripherals were refactored lately.

If it's still needed please reopen the issue.

Thanks for understanding.

closed 2022-05-04 by VojtechBartoska