espressif/arduino-esp32 #5982: SdFat Dedicated SPI causes xQueueGenericSend error

nseidle, 2021-12-06:
SdFat has a method for dedicated SPI transfers to increase read/write speeds:

    sd.begin(SdSpiConfig(pin_microSD_CS, DEDICATED_SPI, SD_SCK_MHZ(16))); //Fails
    //sd.begin(SdSpiConfig(pin_microSD_CS, SHARED_SPI, SD_SCK_MHZ(16))); //Works

But dedicated SPI causes v2 of the core to error out with the following:

    assertion "pxQueue->pcHead != ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == ((void *)0) || pxQueue->u.xSemaphore.xMutexHolder == xTaskGetCurrentTaskHandle()" failed: file "IDF/components/freertos/queue.c", line 782, function: xQueueGenericSend

Decoding the stack trace shows the FatFile is actively trying to write right before things blow up.

[image]

When using SHARED_SPI, everything works great but the transfer rates are much slower, hence why I'm here.

The code for an ESP32 WROOM to reproduce is below. I've tried to reduce it as much as possible.

The original issue was reported to Bill [here](https://github.com/greiman/SdFat/issues/349). Ideally the dedicated SPI feature of SdFat would be supported by the ESP32 SPI driver. This would help out the large community that uses SdFat instead of the built in SD library, but I totally understand that the ESP32 group can't support every feature of every library.

    int pin_microSD_CS = 25;
    
    #include <SPI.h>
    #include "SdFat.h"
    
    SdFat sd;
    SdFile mainFile;
    char mainFileName[40] = "mainFile.txt";
    
    TaskHandle_t writeTaskHandle = NULL;
    const uint8_t writeTaskPriority = 0; //3 being the highest, and 0 being the lowest
    const int writeTaskStackSize = 10000;
    
    void setup()
    {
      Serial.begin(115200);
      delay(500);
      Serial.println("Go");
    
      sd.begin(SdSpiConfig(pin_microSD_CS, DEDICATED_SPI, SD_SCK_MHZ(16))); //Fails
      //sd.begin(SdSpiConfig(pin_microSD_CS, SHARED_SPI, SD_SCK_MHZ(16))); //Works
    
      Serial.print("Creating main file - ");
    
      if (sd.exists(mainFileName))
        sd.remove(mainFileName);
    
      if (mainFile.open(mainFileName, O_RDWR | O_CREAT | O_TRUNC) == true)
        Serial.println("Success!");
      else
        Serial.println("Fail");
    
      if (writeTaskHandle == NULL)
        xTaskCreatePinnedToCore(
          writeToSDTask,
          "WriteSD", //Just for humans
          writeTaskStackSize, //Stack Size
          NULL, //Task input parameter
          writeTaskPriority, //Priority
          &writeTaskHandle, //Task handle
          //0); //Core where task should run, 0=FreeRTOS, 1=Arduino
          1); //Core where task should run, 0=FreeRTOS, 1=Arduino
    }
    
    void loop()
    {
      delay(10);
    }
    
    void writeToSDTask(void *e)
    {
      while (true)
      {
        mainFile.write("File write\r\n");
    
        delay(1);
        taskYIELD();
      }
    }

me-no-dev, 2021-12-07:
This is not an error on our side. What the exception says is that different threads begun and ended the transaction. The reason why core v2 gives the error is that a check was added in newer IDF versions (pre v2 was running on fairly old IDF 3.3). SdFat needs to be fixed and to ensure that the same task that starts a transaction, also ends it.

closed 2021-12-07 by me-no-dev

nseidle, 2021-12-07:
Got it. Thanks!