espressif/arduino-esp32 #5990: Memory leak while periodically turning on and off Wi-Fi

garageeks, 2021-12-07:
What you are trying to do?

Describe your system( Hardware, computer, O/S, core version, environment).

Describe what is failing.
Show the shortest possible code that will duplicate the error.
Show the EXACT error message(it doesn't work is not enough).
All of this work on your part shows us that you have worked to solve YOUR problem. The more complete your issue posting is, the more likely someone will volunteer their time to help you.

If you have a Guru Meditation Error or Backtrace, ***please decode it***:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							esp32-wroom-32e
Core Installation version:			2.0.1
IDE name:						PlatformIO
ChipRevision 3, Cpu Freq 240, SDK Version v4.4-dev-3569-g6a7d83af19-dirty
Flash Size 16777216, Flash Speed 40000000

### Description:
Describe your problem here
My application send periodically send data via HTTP. I have an optional "power save" function that switches off Wi-Fi after sending data,  turns it on before sending data and enables ESP light sleep with a duration of 1 second.
When this function is activated, I observe a memory leak because free heap memory starts at 247KB and goes down to around 70KB, at this point it either fails to connect to server or crashes.
IF "power save" is left off, and the Wi-Fi connection is always on, there is no memory leak.

The issue appears with SDK 1.0.6 as well as latest release 2.0.1

Below an exemplified sketch with only the relevant WiFi code.

Here are some uptime values with the correspondent free heap value.
For the first approx. 90 minutes you can see the value remains basically the same, when power save is turned on, it momentarily increases and then start dropping with each WiFI transmission.

millis | free_heap
11017 | 247308 | PowerSave OFF
61022 | 247308 |  
123954 | 246868 |  
184145 | 246868 |  
244516 | 246856 |  
304522 | 246856 |  
364565 | 246868 |  
424823 | 246868 |  
506154 | 245988 |  
566160 | 245792 |  
626415 | 246672 |  
686491 | 246672 |  
746497 | 247112 |  
806503 | 246672 |  
866509 | 246672 |  
926515 | 246672 |  
986521 | 246664 |  
1046527 | 246672 |  
5346888 | 247332 |  
5357483 | 266728 | PowerSave ON
5418738 | 266160 |  
5471446 | 265340 |  
5536579 | 264524 |  
5591155 | 263704 |  
5657149 | 263340 |  
5713493 | 262736 |  
5777978 | 261928 |  
5833527 | 261100 |  
5899808 | 260944 |  
20136538 | 157064 |  
20192677 | 156728 |  
20254681 | 156088 |  

### Sketch:  (leave the backquotes for [code formatting](https://help.github.com/articles/creating-and-highlighting-code-blocks/))
```cpp
[... 227 lines cut ...]

Platformio.ini
```
[env:esp32dev]
;platform = espressif32
;board = esp32dev
;framework = arduino
platform = https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
board = esp32dev
framework = arduino
platform_packages = platformio/framework-arduinoespressif32 @ https://github.com/espressif/arduino-esp32.git#2.0.1
lib_deps = 
	bblanchon/ArduinoJson@^6.17.2
	robtillaart/RunningMedian@^0.2.1
	ayushsharma82/ElegantOTA@^2.2.4
	jandrassy/TelnetStream@^1.2.1
upload_port = COM[4]
monitor_port = COM[4]
monitor_speed = 115200
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

garageeks, 2022-01-06:
Anyone did experience this memory leak while periodically turn off and on Wi-Fi to save power?
I'd like to know if the bug is in the framework or in the sequence of commands I'm using.
For now I have a crutch in my code that restarts ESP32 when heap memory drops below 100K but it's ugly.
We should really understand why this happens considering that Wi-Fi is a primary feature of ESP32.

TD-er, 2022-01-06:
As a matter of fact, I noticed a few days ago that my node indeed crashed due to memory issues when constantly reconnecting (and also turning off and on the WiFi) but I had forgotten about this issue.

Have not looked into it yet

VojtechBartoska, 2022-05-03:
Hi, I'm closing the issue due to no resposne. If needed you can reopen it. Thanks for your contribution!

closed 2022-05-03 by VojtechBartoska

belazi, 2023-10-05:
I have the same problem here.
There is a memory leak when disconnecting and reconnecting Wifi.

**Hardware:**
Lilygo ttgo t-energy
ESPRESSIF-ESP32-WROVER-B
QSPI Flash 4MB/8MB PSRAM

Laxilef, 2024-04-06:
Same problem

VojtechBartoska, 2024-04-15:
@Laxilef even on 3.0.0-rc1?

Laxilef, 2024-04-15:
@VojtechBartoska haven't tested it yet. 3.0.0-rc1 doesn't work with platformio.