espressif/arduino-esp32 #5991: esptool.exe 3.1.0 not work in x86 Windows, only x64

sh-user, 2021-12-07:
### Hardware:
Board:							ESP32C3 Dev Module
Core Installation version:			2.0.1
IDE name:						Arduino IDE 1.8.13
Computer OS:                                        Windows 8.1 x86 (32-bit)

### Description:
esptool error while compiling any example
```
fork/exec C:\Users\Alex1\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.1.0/esptool.exe: This version of %1 is not compatible with the version of Windows you're running. Check your computer's system information and then contact the software publisher.
Error compiling for board ESP32C3 Dev Module.
```

atanisoft, 2021-12-07:
@sh-user It's time to upgrade to a newer version of Windows. Most (all?) 32 bit platforms are no longer supported by Microsoft let alone the dependencies of various tools.

closed 2021-12-08 by VojtechBartoska

softhack007, 2021-12-14:
see discussion in https://github.com/espressif/arduino-esp32/issues/6012