espressif/arduino-esp32 #5993: ESP32 LAN8720 Repeating restart on ETH.begin()

MedvedBrtnik, 2021-12-08:
Hi,
I have a problem with the Ethernet library. I am using the LAN8720 Ethernet module. In the previous version 1.x.x, everything works correctly. In version 2.0.x, ESP32 restarts repeatedly on the line with ETH.begin (). 
I enclose the source code:
`
#include <ETH.h> // quote to use ETH
#include <WiFi.h>
#include <Wire.h>
#include <WebServer.h> // Introduce corresponding libraries

#define ETH_ADDR 1
#define ETH_POWER_PIN 17 // Do not use it, it can cause conflict during the software reset.
#define ETH_POWER_PIN_ALTERNATIVE 0
#define ETH_MDC_PIN 23
#define ETH_MDIO_PIN 18
#define ETH_TYPE ETH_PHY_LAN8720
#define ETH_CLK_MODE ETH_CLOCK_GPIO0_IN
//#define ETH_CLK_MODE ETH_CLOCK_GPIO17_OUT

//Static IP configuration
IPAddress local_ip(192, 168, 0, 13);
IPAddress gateway(192, 168, 0, 1);
IPAddress subnet(255, 255, 255, 0);
IPAddress dns1(8, 8, 8, 8);
IPAddress dns2 = (uint32_t)0x00000000;

// Web page
String myhtmlPage =
String("") + "\r\n" +
"<html>" + "\r\n" +
"<head>" + "\r\n" +
"    <title>ESP32 WebServer Test</title>" + "\r\n" +
"    <script>" + "\r\n" +
"        function getData() {" + "\r\n" +
"            var xmlhttp;" + "\r\n" +
"            if (window.XMLHttpRequest) {" + "\r\n" +
"                xmlhttp = new XMLHttpRequest();" + "\r\n" +
"            }" + "\r\n" +
"            else {" + "\r\n" +
"                xmlhttp = new ActiveXObject(\"Microsoft.XMLHTTP\");" + "\r\n" +
"            }" + "\r\n" +
"            xmlhttp.onreadystatechange = function() {" + "\r\n" +
"                if (this.readyState == 4 && this.status == 200) {" + "\r\n" +
"                    document.getElementById(\"txtRandomData\").innerHTML = this.responseText;" + "\r\n" +
"                }" + "\r\n" +
"            };" + "\r\n" +
"            xmlhttp.open(\"GET\", \"getRandomData\", true); " + "\r\n" +
"            xmlhttp.send();" + "\r\n" +
"        }" + "\r\n" +
"    </script>" + "\r\n" +
"</head>" + "\r\n" +
"<body>" + "\r\n" +
"    <div id=\"txtRandomData\">Random data:</div>" + "\r\n" +
"    <input type=\"button\" value=\"random\" onclick=\"getData()\">" + "\r\n" +
"</body>" + "\r\n" +
"</html>";

bool eth_connected = false;

WebServer server(80); // Declare the WebServer object

void handleRoot() // Callback
  {
  server.send(200, "text/html", myhtmlPage); //!!! Note that returning to the web page requires "text / html" !!!
  }

void handleAjax() // Callback
  {
  String message = "Random data: ";
  message += String(random(10000));        // Get random number
  server.send(200, "text/plain", message); // Send message back to page
  }

void WiFiEvent(WiFiEvent_t event) {
  switch (event) {
    case SYSTEM_EVENT_ETH_START:
      Serial.println("ETH Started");
      //set eth hostname here
      ETH.setHostname("VArio-eth");
      break;
    case SYSTEM_EVENT_ETH_CONNECTED:
      Serial.println("ETH Connected");
      break;
    case SYSTEM_EVENT_ETH_GOT_IP:
      Serial.print("ETH MAC: ");
      Serial.print(ETH.macAddress());
      Serial.print(", IPv4: ");
      Serial.print(ETH.localIP());
      if (ETH.fullDuplex()) {
        Serial.print(", FULL_DUPLEX");
        }
      Serial.print(", ");
      Serial.print(ETH.linkSpeed());
      Serial.println("Mbps");
      eth_connected = true;
      break;
    case SYSTEM_EVENT_ETH_DISCONNECTED:
      Serial.println("ETH Disconnected");
      eth_connected = false;
      connectWiFi();
      break;
    case SYSTEM_EVENT_ETH_STOP:
      Serial.println("ETH Stopped");
      eth_connected = false;
      break;
    default:
      break;
    }
  }

void connectWiFi() {
  WiFi.begin("kiwi", "2XkolemKola");
  Serial.print("Connecting to WiFi");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    }
  Serial.println("");
  Serial.print("Connected to the WiFi network with IP address:");
  Serial.println(WiFi.localIP());
  }

void setup() {
  pinMode(ETH_POWER_PIN_ALTERNATIVE, OUTPUT);
  digitalWrite(ETH_POWER_PIN_ALTERNATIVE, LOW);
  delay(250);
  digitalWrite(ETH_POWER_PIN_ALTERNATIVE, HIGH);

  Serial.begin(115200);
  Serial.println("Start...");

  Serial.println("Setting for event Eth.");
  WiFi.onEvent(WiFiEvent);

  Serial.println("Configure Eth.");
   //The problem is manifested a line below. 
  ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE); // Enable ETH

  //Serial.println("Configure IPv4 address.");
  //ETH.config(local_ip, gateway, subnet, dns1, dns2); // Static IP, leave without this line to get IP via DHCP

  int counter = 0;
  while (!((uint32_t)ETH.localIP())) {
    delay(500);
    Serial.print(".");
    counter++;
    if (counter > 20) {
      break;
      }
    } // Waiting for IP (leave this line group to get IP via DHCP)

  if (!eth_connected) connectWiFi();

  server.on("/", handleRoot);                        // Register link and callback function
  server.on("/getRandomData", HTTP_GET, handleAjax); // Request and callback function of the get method sent by ajax in the registration web page

  server.begin(); // Start server
  Serial.println("Web server started");

  }

void loop() {
  server.handleClient(); // Handling requests from clients
  }
`

khoih-prog, 2022-02-03:
You can try my wrapper library [**WebServer_WT32_ETH01**](https://github.com/khoih-prog/WebServer_WT32_ETH01), which takes care of the breaking changes in core v2.0.0+. 

Check [**Important Notes**](https://github.com/khoih-prog/WebServer_WT32_ETH01#Important-notes)

Try this example [**AdvancedWebServer**](https://github.com/khoih-prog/WebServer_WT32_ETH01/tree/main/examples/AdvancedWebServer)

VojtechBartoska, 2022-04-22:
Closing as solved. If needed please reopen the issue.

closed 2022-04-22 by VojtechBartoska