espressif/arduino-esp32 #5994: ESP32-S2 Com port in Windows 8.1

waltr1, 2021-12-08:
Trying to upload code (or UF2 bootloader) into Adafruit Feather ESP32-S2 and bare ESP32-S2 modules through the Native USB hardware.

----------------------------- Remove above -----------------------------

### Hardware:
Board:							Adafruit Feather ESP32-S2
Core Installation version:			2.0.1
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                    yes
Upload Speed:					921600
Computer OS:                                        Windows 8.1

### Description:
When plugging USB into PC no Com Port shows in Device manager. Had dialog in Adafruit's Forum and found that only Windows 10's Microsoft drivers work. Testing ESP32-S2 on a Win10 PC at work and Com port shows.

There are many developers that still use Win 7 or win8. We need driver to support the ESP32-S2 for Win7 & Win8.

VojtechBartoska, 2021-12-09:
Closing the issue, as @atanisoft mentioned. It's not feasible to support older OS versions forever. Please consider updating your OS @waltr1 :) Hope it's fair enough solution for you and you understand our approach.

xalier1, 2021-12-17:
I am disappointed that  support for the native ESP32-S2 usb is not planned for Windows 7/8.  In  the link

xalier1, 2021-12-17:
Sorry my previous incomplete comment.

I am disappointed that support for the native ESP32-S2 usb is not planned for Windows 7/8. I agree with waltr1 that there are lots of developers using windows 7/8 still.   Also, in the  link:
    https://docs.espressif.com/projects/esp-idf/en/v4.3.1/esp32s2/api-guides/usb-console.html
Says the following when talking about what happens on connection of the ESP32-S2 to a computer.
"A serial port will appear in the system. On most operating systems (Windows 8 and later, Linux, macOS) driver installation is not required. Find the port name using Device Manager (Windows) or by listing /dev/ttyACM* devices on Linux or /dev/cu* devices on macOS. "
This  v4.3.1 documentation is misleading if  there are no plans for Windows 8 to  support the USB CDC driver for the ESP32s2.

me-no-dev, 2021-12-17:
Have you tried Zadig driver for Windows? Three is nothing to support on our side per say, it's Windows that needs drivers. The CDC post is standard.

aryehassbian, 2022-04-02:
Yes, I agree with both above - I have been working with windows 8.1 64 on all my developments and I am not sure why i am NOT being able to use ESP32-S2 Com Port on Windows 8.1.
I have tried Zadig with NO LUCK at all - There should be a driver for it supporting Windows 8.1 - still a supported Windows by Microsoft till end of 2023 - Please let us know when we will have a proper driver to work with.

I also tried to see if it is recognizable in Windows 10 and IT WEAS NOT as well.
We will need to have a proper driver to work with Windows 8.1 and also Windows 10
[https://paste.pics/2d3092bd81421778e9fdbab938a17b92?spm=a2g0s.imconversation.0.0.731a3e5fo5mzvl](url)
Please support ASAP.
Thanks

atanisoft, 2022-04-03:
@aryehassbian Open a new issue and provide a reproducible test case. Windows itself needs to support USB-CDC devices and not specifically the ESP32-S2. If you are using custom descriptors it is possible you have an error in the descriptor which is why Windows is not recognizing the device when plugged in. I haven't seen this issue on any of my available devices or machines.

kutukvpavel, 2022-04-25:
@waltr1 @xalier1 @aryehassbian The support actually is there, the required (class-)driver is called usbser.sys ans is included with Win7 and Win8 too. Win10 just bothers to install it automatically for all CDC devices, while older OSs don't, you have to do it manually.

I was surprised how little information is there about getting modern VCP devices running under Win7, so I created a repository with necessary documentation links and INF files for ESP32: https://github.com/kutukvpavel/Esp32-Win7-VCP-drivers