espressif/arduino-esp32 #5995: Enable menuconfig options for Arduino

aguileraGit, 2021-12-09:
Is it possible to change a config file (boards.txt for example) to add/change the menuconfig options? I'm specifically wanting to add the low power [BLE options](https://github.com/espressif/esp-idf/issues/947#issuecomment-500312453) while working in the Arduino IDE.

[ESP-IDF Project Configuration Option Reference](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#configuration-options-reference)

Yes, I understand a 32KHz will need needed.

### Hardware:
|||
|:---|---|
|<B>Board</B>|ESP32-C Mini 1 Dev Module|
|<B>IDE name</B>|Arduino IDE|
|<B>Computer OS</B>|Windows 10 or Mac OSX|

atanisoft, 2021-12-09:
Menu config entries require a rebuild of the ESP-IDF components to take effect (generally). The only option is unfortunately to build a custom version of the ESP-IDF components or use arduino as a component in an IDF build.

closed 2021-12-13 by aguileraGit

murarduino, 2023-05-10:
> Menu config entries require a rebuild of the ESP-IDF components to take effect (generally). The only option is unfortunately to build a custom version of the ESP-IDF components or use arduino as a component in an IDF build.

I have an idea, but I don't know if it will work.
Copy the cores/esp32 directory from the Arduio-ESP32 library into the project's components directory and rename the esp32 directory to Arduino. Avoid using setup and loop frames (functions) in the code, just borrow the Arduino-ESP32 code.
There is no problem with the compilation test at present, but it is not known whether this method is feasible.

murarduino, 2023-05-10:
The question now is how ESP-IDF uploads files to SPIFFS