espressif/arduino-esp32 #5997: xTaskCreate doesn't work on version 2

DonatelloX, 2021-12-09:
### Hardware:
Board:							ESP32 Dev Module?
Core Installation version:			2.01
IDE name:						Arduino IDE 1.18.15
Flash Frequency:					240Mhz
PSRAM enabled:                                    no
Upload Speed:					921600
Computer OS:                                        Windows 10

### Description:
xTaskCreate doesn't work on version 2, but it works on version 1.6.

For example in this code, tasks are never called, so messages with Serial.println are never displayed:
```
  xTaskCreate(      TaskMotor,          /* Task function. */
                    "TaskMotor",        /* String with name of task. */
                    10000,            /* Stack size in bytes. */
                    NULL,             /* Parameter passed as input of the task */
                    1,                /* Priority of the task. */
                    NULL);            /* Task handle. */

   xTaskCreate(     TaskTestMotor,          /* Task function. */
                    "TaskTestMotor",        /* String with name of task. */
                    10000,            /* Stack size in bytes. */
                    NULL,             /* Parameter passed as input of the task */
                    1,                /* Priority of the task. */
                    NULL);            /* Task handle. */

[...]

void TaskMotor( void * pvParameters ) {
  while(1) {
    Serial.println("TaskMotor");
    vTaskDelay(1000 / portTICK_PERIOD_MS);
  }    
}

void TaskTestMotor( void * pvParameters ) {
  while(1) {
    Serial.println("TestMotor");
    vTaskDelay(1000 / portTICK_PERIOD_MS);
  }
}
```
While it works fine with version 1.6.

pgrawehr, 2021-12-09:
Works fine for me. I think not much would work at all if this were generally broken. Have you properly initialized the serial port?

DonatelloX, 2021-12-09:
I have analyzed the problem better: in version 2.0.1 "xTaskCreate" does not work if "esp_now_add_peer" is used. For example in this code, "xTaskCreate" doesn't work in 2.0.1:
```
#include <WiFi.h>
#include <esp_now.h>

//ESP-NOW
// Set the SLAVE MAC Address
uint8_t slaveAddress[] = {0x98, 0xCD, 0xAC, 0x60, 0xE1, 0xE4};
// PMK and LMK keys
static const char* PMK_KEY_STR = "1234567890123456";
static const char* LMK_KEY_STR = "1234567890123456";

typedef struct struct_esp_now {
  float temperature;
  float humidity;
  char serial_char;
};
// Create a struct_message called myData
struct_esp_now ESPnow_Data;

void setup() {
  Serial.begin(115200);

  Serial.println("");
  delay(500);
 
  Serial.println("*** START ***");

  // Set device as a Wi-Fi Station
  WiFi.mode(WIFI_STA);
  // Init ESP-NOW
  if (esp_now_init() != ESP_OK) {
    Serial.println("ESP-NOW: There was an error initializing");
    return;
  }
  // Setting the PMK key
  esp_now_set_pmk((uint8_t *)PMK_KEY_STR);

  // Register the slave
  esp_now_peer_info_t slaveInfo;
  memcpy(slaveInfo.peer_addr, slaveAddress, 6);
  slaveInfo.channel = 0;
  // Setting the master device LMK key
  for (uint8_t i = 0; i < 16; i++) {
    slaveInfo.lmk[i] = LMK_KEY_STR[i];
  }
  slaveInfo.encrypt = true;
  
  // Add slave
  **_//Without the bold lines it works fine in 2.0.1._**        
  **if (esp_now_add_peer(&slaveInfo) != ESP_OK){
    Serial.println("ESP_NOW: There was an error registering the slave");
    return;
  }**

   xTaskCreate(     TaskTestMotor,          /* Task function. */
                    "TaskTestMotor",        /* String with name of task. */
                    10000,            /* Stack size in bytes. */
                    NULL,             /* Parameter passed as input of the task */
                    1,                /* Priority of the task. */
                    NULL);            /* Task handle. */
                    
}

void TaskTestMotor( void * pvParameters ) {
  while(1) {
    Serial.println("Motor_test");
    vTaskDelay(1000 / portTICK_PERIOD_MS);
  }
}

void loop() {

}
```

Everything works fine with version 1.0.6!

atanisoft, 2021-12-10:
@DonatelloX are you 100% certain that `esp_now_add_peer` is not blocking execution of remaining steps?

DonatelloX, 2021-12-10:
@atanisoft thank you!  `return` terminates the `setup` routine.
The `esp_now_add_peer` behaves differently between 1.0.6 and 2.0.1.

closed 2021-12-15 by DonatelloX