espressif/arduino-esp32 #6000: Define the macro CONFIG_DISABLE_HAL_LOCKS in sdkconfig.h,  cause compilation error

JasonWong08, 2021-12-10:
### Hardware:
Board:							ESP32 Dev Module
Core Installation version:			2.0.1
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
PSRAM enabled:                                      no
Upload Speed:					921600
Computer OS:                                         Windows 10

### Description:
Define the macro CONFIG_DISABLE_HAL_LOCKS in sdkconfig.h as following:
#define CONFIG_DISABLE_HAL_LOCKS 1

It will cause compilation error:
C:\Users\wjf-1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\cores\esp32\esp32-hal-i2c-slave.c: In function 'i2cSlaveDeinit':
C:\Users\wjf-1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\cores\esp32\esp32-hal-i2c-slave.c:363:12: error: 'i2c_slave_struct_t' {aka 'struct i2c_slave_struct_t'} has no member named 'lock'
     if(!i2c->lock){
            ^~
C:\Users\wjf-1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\cores\esp32\esp32-hal-i2c-slave.c: In function 'i2cSlaveWrite':
C:\Users\wjf-1\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1\cores\esp32\esp32-hal-i2c-slave.c:380:12: error: 'i2c_slave_struct_t' {aka 'struct i2c_slave_struct_t'} has no member named 'lock'
     if(!i2c->lock){
            ^~

So I suggest add conditional compilation as following:
```cpp
#if !CONFIG_DISABLE_HAL_LOCKS    //Line 363
    if(!i2c->lock){
        log_e("Lock is not initialized! Did you call i2c_slave_init()?");
        return ESP_ERR_NO_MEM;
    }
#endif

#if !CONFIG_DISABLE_HAL_LOCKS    //Line 382
    if(!i2c->lock){
        log_e("Lock is not initialized! Did you call i2c_slave_init()?");
        return ESP_ERR_NO_MEM;
    }
#endif
```

atanisoft, 2021-12-10:
duplicate of https://github.com/espressif/arduino-esp32/issues/5943. Fixed on master but not in a release yet.

JasonWong08, 2021-12-14:
Thanks a lot.

JasonWong08, 2022-01-14:
I checked the latest release (Core Installation version: 2.0.2), the following conditional compilation is not added yet, as follows:
```c++
#if !CONFIG_DISABLE_HAL_LOCKS    // in the function esp_err_t i2cSlaveDeinit(uint8_t num)
    if(!i2c->lock){
        log_e("Lock is not initialized! Did you call i2c_slave_init()?");
        return ESP_ERR_NO_MEM;
    }
#endif

#if !CONFIG_DISABLE_HAL_LOCKS    // in the function size_t i2cSlaveWrite(uint8_t num, const uint8_t *buf, uint32_t len, uint32_t timeout_ms)
    if(!i2c->lock){
        log_e("Lock is not initialized! Did you call i2c_slave_init()?");
        return ESP_ERR_NO_MEM;
    }
#endif
```

mrengineer7777, 2022-01-25:
See https://github.com/espressif/arduino-esp32/pull/6108.  It should be in the next release.

VojtechBartoska, 2022-04-11:
Closing as duplicate which is already solved. Thanks for your contribution.

closed 2022-04-11 by VojtechBartoska