espressif/arduino-esp32 #6003: all methods in the class BLEAdvertising are void and do not return any error codes

pabloandresm, 2021-12-10:
There is not a single methond in the class BLEAdvertising located at BLEAdvertising.h/.cpp that returns any error code.

For example...BLEAdvertising::start().....that calls the syscall esp_ble_gap_start_advertising(), which returns an error, it is just ignored, and the BLEAdvertising::start() returns void.

Is there any logical explanation for this? I don't see any extra complication in the code by just returning the result of esp_ble_gap_start_advertising().

And the same thing goes for all the methods of this class.

VojtechBartoska, 2022-05-03:
Hi,

I'm closing the issue due to no response. 

If needed you can reopen it. 

Thanks for your contribution & understanding!

pabloandresm, 2022-05-03:
> Hello, can you please retest this on v2.0.3-rc1?

Hi there,

Sorry I do not understand your request. The code did not change. The bug is still there.
Not one method of BLEAdvertising returns any error code.
I mentioned the example of ::start() that is void type. It could return the error code of esp_ble_gap_start_advertising().

Same thing for all other methods. All of void type.

VojtechBartoska, 2022-05-04:
I see, reopening.

FidelSch, 2024-10-15:
Hello
I would like to contribute, as this seems like a fairly simple fix. Could this be assigned to me?
As this is my first contribution, any guidelines are welcome.
Thanks in advance!

Parsaabasi, 2025-01-08:
Hello @FidelSch 
Are you still interested in the fix? 
[HERE](https://docs.espressif.com/projects/arduino-esp32/en/latest/contributing.html) is the guideline for the contributors. Please go study it and suggest your changes

Thanks.

pabloandresm, 2025-01-08:
Hello,
I would like to implement the fix. I was the one that reported it.

Parsaabasi, 2025-03-11:
Hi @pabloandresm 
Do you have any updates on this issue?

Jason2866, 2025-03-12:
closing since no feedback

closed 2025-03-12 by Jason2866