espressif/arduino-esp32 #6004: HelloServer example doesn't work with ESP32C3

sh-user, 2021-12-10:
### Hardware:
Board: ESP32C3 Dev Module
Core Installation version: 2.0.1
IDE name: Arduino IDE 1.8.13

### Description:
HelloServer example doesn't work with ESP32C3
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4038195a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x484
load:0x403ce000,len:0x90c
load:0x403d0000,len:0x2398
SHA-256 comparison failed:
Calculated: e0c357829f601add6a50d6ea19f89687ec5acf531610506a38b33d09b4a71c7e
Expected: 77332826ec7bc491df854452f675071e3a094e80858dd7d91dd7e27940204216
Attempting to boot anyway...
entry 0x403ce000 
```
### Sketch:  
```
#include <WiFi.h>
#include <WiFiClient.h>
#include <WebServer.h>
#include <ESPmDNS.h>

const char* ssid = "........";
const char* password = "........";

WebServer server(80);

const int led = 13;

void handleRoot() {
  digitalWrite(led, 1);
  server.send(200, "text/plain", "hello from esp32!");
  digitalWrite(led, 0);
}

void handleNotFound() {
  digitalWrite(led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i = 0; i < server.args(); i++) {
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  digitalWrite(led, 0);
}

void setup(void) {
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin("esp32")) {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot);

  server.on("/inline", []() {
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
}

void loop(void) {
  server.handleClient();
  delay(2);//allow the cpu to switch to other tasks
}
```

felmue, 2021-12-10:
Hello @sh-user 

hmm, the code you posted runs fine on my ESP32C3 - two thoughts:
- from your log `mode:DIO, clock div:1` and assuming I am understanding the documentation correctly this means the flash frequency is 80 MHz. The default however seems to be 40 MHz (e.g. `mode:DIO, clock div:2`). Maybe try to change the flash frequency to 40 MHz for a test?
- I've seen ESP32 boards reset while using WiFi due to the power source being too weak. Have you tried a different power source / different USB cable?

Thanks
Felix

sh-user, 2021-12-10:
`GetChipID `example works fine

sh-user, 2021-12-10:
@felmue the reason was in the RESET button, I caught interference with a wire 10 cm long and the ESP32C3-12F module was cyclically reset

felmue, 2021-12-11:
Hello @sh-user 

glad to hear you figured it out. And thank you for letting me know.

Thanks
Felix

adumont, 2021-12-12:
Could you explain how to fix the issue? I am having the same issue here... (**with an empty sketch**)

sh-user, 2021-12-12:
@adumont check pullup resistor CHIP_EN and supply voltage

adumont, 2021-12-12:
The board is simply plugged via USB to the computer. Could you elaborate on
what I should do, or point me to some page explaining it. Do I need to put
a pull up resistor on some pin? Do I need to provide external power to the
board?

sh-user, 2021-12-12:
No need to supply external power, usually USB is enough, check that the voltage on the RST pin 3.3V.
https://docs.espressif.com/projects/esp-idf/en/v4.3-beta1/esp32c3/hw-reference/esp32c3/user-guide-devkitm-1.html
I assembled my board using the ESP32C3-12F module and soldered the button to the RST pin, when connected to the router, the module reset

adumont, 2021-12-13:
But why would it work fine (it flashes programs without error, and then it
works fine) using the alternative cpq/mdk tools, and not the official
Arduino IDE + arduino_esp32 tools?
--
Alex

sh-user, 2021-12-13:
Write your sketch here, I will check it on my module

adumont, 2021-12-13:
As this issue was closed I opened #6013 yesterday. In #6013 they tell me
that revision 2 isn't supported.
--
Alex

sh-user, 2021-12-13:
@adumont try flash rate: 40 MHz

adumont, 2021-12-13:
Yes I tries yesterday already, with the same result :(
--
Alex

atanisoft, 2021-12-14:
@adumont Your issue does not appear to be related to this but instead the chip revision is not supported by the current arduino-esp32 build.

closed 2021-12-14 by sh-user

git67frederick, 2023-10-28:
comment out all lines with digitalWrite and try again...works now for me  M5StampC3
//digitalWrite(led,

alexceltare2, 2023-11-07:
Try changing `const int led = 13;` to 8 or whatever LED you have in your devboard. Worked for me.

short-cheese-the-3rd, 2025-02-28:
> Try changing `const int led = 13;` to 8 or whatever LED you have in your devboard. Worked for me.

This worked for me too. Thank you so much @alexceltare2

libreleee, 2025-05-16:
const int led = 8;   //const int led = 13;  <- 8

The led is connected to pin 8 and pin 13 seems to be related to Reset
When I change it to pin 8, the LED works and it is visible in the web browser.

[image]