espressif/arduino-esp32 #6007: ESP-NOW assert failed: xQueueSemaphoreTake queue.c:1545 (( pxQueue ))

pszcoding, 2021-12-11:
Hello there, 

I tried to send data from one ESP32 to another

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module
IDE name:						Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:					       115200
Computer OS:                                         Windows 10

### Description:
Then I start to send the data the ESP gives me an error over Serial and reboot:
```
...
14:25:14.358 -> Bytes received: 8
14:25:14.358 -> 
14:25:14.358 -> assert failed: xQueueSemaphoreTake queue.c:1545 (( pxQueue ))
14:25:14.358 -> 
14:25:14.358 -> 
14:25:14.358 -> Backtrace:0x4008305d:0x3ffaf9500x4008b271:0x3ffaf970 0x40090335:0x3ffaf990 0x4008c1d1:0x3ffafac0 0x400d3766:0x3ffafb00 0x400d2201:0x3ffafb20 0x401017ed:0x3ffafb40 0x40102047:0x3ffafb60 0x40119ecc:0x3ffafb90 0x40115529:0x3ffafbb0 0x4013e126:0x3ffafcc0 0x4013e189:0x3ffafd10 0x4013fd65:0x3ffafd30 0x4013cc50:0x3ffafd50 
14:25:14.358 -> 
14:25:14.358 -> 
14:25:14.358 -> 
14:25:14.358 -> 
14:25:14.358 -> ELF file SHA256: 0000000000000000
14:25:14.358 -> 
14:25:14.358 -> Rebooting...
...
```

### Reciver
```
#include <SPI.h>
#include <LiquidCrystal_I2C.h>
#include "WiFi.h"
#include <esp_now.h>

int motor1Pin1 = 35; 
int motor1Pin2 = 32; 
int enable1Pin = 34; 

const int freq = 30000;
const int pwmChannel = 0;
const int resolution = 8;
int dutyCycle = 200;
int c = 0;
int timeout = 0;
int statusToReturn = 6;

char msg[5];
uint8_t broadcastAddress[] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00}; //changed for the Issue

String success;
int returnedStatus;

typedef struct struct_message {
    bool vor;
    int speed;
} struct_messageIn;
struct_message msgIn;

typedef struct struct_messageOut {
    int status;
} struct_messageOut;
struct_messageOut msgOut;

void OnDataSent(const uint8_t *mac_addr, esp_now_send_status_t status) {
  Serial.print("\r\nLast Packet Send Status:\t");
  Serial.println(status == ESP_NOW_SEND_SUCCESS ? "Delivery Success" : "Delivery Fail");
  if (status == 0){
    success = "Delivery Success :)";
  }
  else{
    success = "Delivery Fail :(";
  }
}

// Callback when data is received
void OnDataRecv(const uint8_t * mac, const uint8_t *incomingData, int len) {
  memcpy(&msgIn, incomingData, sizeof(msgIn));
  Serial.print("Bytes received: ");
  Serial.println(len);
  Serial.println(msgIn);
  bool vor = msgIn.vor;
  int speed = msgIn.speed;
  timeout = 0;
  if(speed == 0){
    digitalWrite(motor1Pin1, LOW);
    digitalWrite(motor1Pin2, LOW);
    statusToReturn = 1;
  }
  else{
    if(vor){
      digitalWrite(motor1Pin1, HIGH);
      digitalWrite(motor1Pin2, LOW);
      statusToReturn = 2;
    }
    else{
      digitalWrite(motor1Pin1, LOW);
      digitalWrite(motor1Pin2, HIGH);
      statusToReturn = 3;
    }
    ledcWrite(pwmChannel, speed);
  }
}

void setup() {
  Serial.begin(115200);

  if (esp_now_init() != ESP_OK) {
    Serial.println("Error initializing ESP-NOW");
    return;
  }
  esp_now_register_send_cb(OnDataSent);  
  // Register peer
  esp_now_peer_info_t peerInfo = {};
  memcpy(peerInfo.peer_addr, broadcastAddress, 6);
  peerInfo.channel = 0;  
  peerInfo.encrypt = false;
  
  // Add peer        
  if (esp_now_add_peer(&peerInfo) != ESP_OK){
    Serial.println("Failed to add peer");
    return;
  }
  // Register for a callback function that will be called when data is received
  esp_now_register_recv_cb(OnDataRecv);
}

void loop() {

}

```
### Transmitter
```
#include  <SPI.h>
#include "WiFi.h"
#include <esp_now.h>

int potentiometerPin = 36;
int potentiometerPinX = 35;
int potentiometerPinY = 34;
int maxSpeed = 255;
bool vor = true;
char msg[5];
uint8_t broadcastAddress[] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00}; //changed for the Issue

String success;
int returnedStatus;

typedef struct struct_message {
    bool vor;
    int speed;
} struct_message;
struct_message BME280Readings;

typedef struct struct_messageIn {
    int status;
} struct_messageIn;
struct_messageIn incomingReadings;

void OnDataSent(const uint8_t *mac_addr, esp_now_send_status_t status) {
  Serial.print("\r\nLast Packet Send Status:\t");
  Serial.println(status == ESP_NOW_SEND_SUCCESS ? "Delivery Success" : "Delivery Fail");
  if (status ==0){
    success = "Delivery Success :)";
  }
  else{
    success = "Delivery Fail :(";
  }
}

// Callback when data is received
void OnDataRecv(const uint8_t * mac, const uint8_t *incomingData, int len) {
  memcpy(&incomingReadings, incomingData, sizeof(incomingReadings));
  Serial.print("Bytes received: ");
  Serial.println(len);
  returnedStatus = incomingReadings.status;
}
 

void setup() {
  Serial.begin(115200);

  if (esp_now_init() != ESP_OK) {
    Serial.println("Error initializing ESP-NOW");
    return;
  }
  esp_now_register_send_cb(OnDataSent);  
  // Register peer
  esp_now_peer_info_t peerInfo = {};
  memcpy(peerInfo.peer_addr, broadcastAddress, 6);
  peerInfo.channel = 0;  
  peerInfo.encrypt = false;
  
  // Add peer        
  if (esp_now_add_peer(&peerInfo) != ESP_OK){
    Serial.println("Failed to add peer");
    return;
  }
  // Register for a callback function that will be called when data is received
  esp_now_register_recv_cb(OnDataRecv);

}

void loop() {
  int analogValue = analogRead(potentiometerPin);
  int analogValueX = analogRead(potentiometerPinX);
  int analogValueY = analogRead(potentiometerPinY);

  float percent = floatMap(analogValue, 0, 4095, 0, 1);
  float percentX = floatMap(analogValueX, 0, 4095, 0, 1);
  float percentY = 1 - floatMap(analogValueY, 0, 4095, 0, 1);

  int activeSpeed = 0;
  
  if(percentY > 0.28 || percentY < 0.22){
    vor = percentY > 0.25;
    if(vor){
      activeSpeed = round(((percentY-0.25)*1.333333)*(percent*maxSpeed));
    }
    else{
      activeSpeed = round((0.25-percentY)*1.333333*(percent*maxSpeed));
    }
    
  }

  BME280Readings.vor = vor;
  BME280Readings.speed = activeSpeed;

  esp_err_t result = esp_now_send(broadcastAddress, (uint8_t *) &BME280Readings, sizeof(BME280Readings));
   
  if (result == ESP_OK) {
    Serial.println("Sent with success");
  }
  else {
    Serial.println("Error sending the data");
  }

  delay(500);
  
}

float floatMap(float x, float in_min, float in_max, float out_min, float out_max) {
  return (x - in_min) * (out_max - out_min) / (in_max - in_min) + out_min;
}

```

P.S. 
[image]

pszcoding, 2021-12-11:
Exception Decoder says:

Decoding stack results
0x4008305d: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c line 402
0x4008b271: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c line 129
0x40090335: __assert_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/assert.c line 85
0x4008c1d1: xQueueSemaphoreTake at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/queue.c line 1545
0x400d3766: ledcWrite at C:\Users\user\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.1-RC1\cores\esp32\esp32-hal-ledc.c line 235
0x400d2201: OnDataRecv(unsigned char const*, unsigned char const*, int) at C:\Users\user\Documents\Arduino\project\v1.1\reciver/reciver.ino line 71

closed 2021-12-11 by pszcoding

tzurbri, 2023-05-03:
What was the problem ? And what have you done?

and3rson, 2024-03-06:
Same here. @psz-dev mind sharing the solution?

Horashi0, 2024-03-24:
Mind sharing how you fixed it? @psz-dev