espressif/arduino-esp32 #6008: BLEServer.h callback onDisconnect() missing esp_ble_gatts_cb_param_t parameter

pabloandresm, 2021-12-11:
The BLEServer.h defines callbacks for connection and disconnection.

For the connect, there are 2 virtual methods, 1 without and 1 with esp_ble_gatts_cb_param_t.

For the disconnect, there is only 1 virtual method, without that parameter.

At least for coherency, there should be an onDisconnect() with that parameter.

This fix wouldn't even impact the code already written, as it is an extra virtual method.

bear24rw, 2022-04-05:
Has there been any movement on this? I have the same issue.

bear24rw, 2022-04-05:
Here is a patch that resolve this issue I believe:

```patch
diff --git a/libraries/BLE/src/BLEServer.cpp b/libraries/BLE/src/BLEServer.cpp
index 7f156eff..f9f7d451 100644
--- a/libraries/BLE/src/BLEServer.cpp
+++ b/libraries/BLE/src/BLEServer.cpp
@@ -207,6 +207,7 @@ void BLEServer::handleGATTServerEvent(esp_gatts_cb_event_t event, esp_gatt_if_t
 		case ESP_GATTS_DISCONNECT_EVT: {
 			if (m_pServerCallbacks != nullptr) {         // If we have callbacks, call now.
 				m_pServerCallbacks->onDisconnect(this);
+				m_pServerCallbacks->onDisconnect(this, param);
 			}
             if(m_connId == ESP_GATT_IF_NONE) {
                 return;
@@ -374,6 +375,12 @@ void BLEServerCallbacks::onDisconnect(BLEServer* pServer) {
 	log_d("BLEServerCallbacks", "<< onDisconnect()");
 } // onDisconnect
 
+void BLEServerCallbacks::onDisconnect(BLEServer* pServer, esp_ble_gatts_cb_param_t* param) {
+	log_d("BLEServerCallbacks", ">> onDisconnect(): Default");
+	log_d("BLEServerCallbacks", "Device: %s", BLEDevice::toString().c_str());
+	log_d("BLEServerCallbacks", "<< onDisconnect()");
+} // onDisconnect
+
 void BLEServerCallbacks::onMtuChanged(BLEServer* pServer, esp_ble_gatts_cb_param_t* param) {
 	log_d("BLEServerCallbacks", ">> onMtuChanged(): Default");
 	log_d("BLEServerCallbacks", "Device: %s MTU: %d", BLEDevice::toString().c_str(), param->mtu.mtu);
diff --git a/libraries/BLE/src/BLEServer.h b/libraries/BLE/src/BLEServer.h
index f3ca5be6..804e1457 100644
--- a/libraries/BLE/src/BLEServer.h
+++ b/libraries/BLE/src/BLEServer.h
@@ -134,6 +134,7 @@ public:
 	 * @param [in] pServer A reference to the %BLE server that received the existing client disconnection.
 	 */
 	virtual void onDisconnect(BLEServer* pServer);
+	virtual void onDisconnect(BLEServer* pServer, esp_ble_gatts_cb_param_t *param);
 
 	/**
 	 * @brief Handle a new client connection.
```

VojtechBartoska, 2022-04-11:
Hello @bear24rw, did you test this on v2.0.3-rc1?

bear24rw, 2022-04-11:
@VojtechBartoska I have only tested with `v2.0.0`

VojtechBartoska, 2022-04-12:
ok, are you able to retest it on v2.0.3-rc1? Thanks

bear24rw, 2022-04-13:
@VojtechBartoska I have just tested the patch on `2.0.3-RC1`

```cpp
    class ServerCallbacks : public BLEServerCallbacks {
        void onConnect(BLEServer* server)
        {
            Serial.println("ble: onConnect");
        };

        void onDisconnect(BLEServer* server, esp_ble_gatts_cb_param_t* param)
        {
            Serial.printf("ble: onDisconnect (%s)", BLEUtils::gattCloseReasonToString(param->disconnect.reason).c_str());
        }
    };
```

bear24rw, 2022-07-15:
Has there been any movement on this? This patch would be helpful to debug some disconnect issues on my project.

PR #7559 (2022-12-14): Added onDisconnect with gatt parameters to BLEServerCallbacks, equivalent to the existing onConnect
-----------
## Description of Change
This change adds a onDisconnect with parameters to BLEServerCallbacks, equivalent to onDisconnect, which allows the code to know which (of the multiple connections) is disconnecting. Similar to how onConnect can be used with esp_ble_gatts_cb_param_t now.

## Tests scenarios
Tested on ESP32-S3-DevKitC-1, Platform-IO 5.1.1 (framework=Arduino), while connecting with 2 BLE devices (android and ios) simultaneous, reconnecting them one at a time.

## Related links
 [... cut]
closed 2022-12-14 by me-no-dev