espressif/arduino-esp32 #6011: Unable to change the FreeRTOS tick rate

dizcza, 2021-12-11:
### Hardware:
Board: M5Core2
Core Installation version: arduino-esp32 v2.0.1
IDE name: Arduino IDE
Flash Frequency: 240Mhz
PSRAM enabled: Enabled
Upload Speed: 115200
Computer OS: Ubuntu 20.04

### Description
I've searched for the options how to modify the FreeRTOS frequency in Arduino and found these:
* https://www.esp32.com/viewtopic.php?t=3510
* hacking the sdkconfig, though I didn't get how exactly. Copying the pre-built sdkconfig file is really not a way to go.

Defining `configTICK_RATE_HZ` before other includes doesn't help. In the sketch below, I expect to get the delay of ~10ms with the `vTaskDelay` function since `delay` is just a wrapper for `vTaskDelay` according to https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-misc.c#L173. But the actual delay is ~1ms, a wrong value. `portTICK_PERIOD_MS` seems to be set correctly though (10 ms per tick).

### Sketch
```cpp
#define configTICK_RATE_HZ 100

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

void setup() {
  ESP_LOGI("main", "Freertos tick rate %d, portTICK_PERIOD_MS %u, configUSE_16_BIT_TICKS %u",
          configTICK_RATE_HZ, portTICK_PERIOD_MS, configUSE_16_BIT_TICKS);
  int64_t start = esp_timer_get_time();
  TickType_t tick10 = pdMS_TO_TICKS(10);
  vTaskDelay(tick10);
//  delay(10);
  int64_t waited_us = esp_timer_get_time() - start;
  ESP_LOGI("main", "Waited %lld us, ticks=%u", waited_us, tick10);
}

void loop() {

}
```

### Serial output
```
[    29][I][tick_rate.ino:9] setup(): Freertos tick rate 100, portTICK_PERIOD_MS 10, configUSE_16_BIT_TICKS 0
[    30][I][tick_rate.ino:15] setup(): Waited 443 us, ticks=1
```

atanisoft, 2021-12-12:
This can only be changed via sdkconfig *AND* only if you use arduino as a component of ESP-IDF. Without doing this any changes will not take effect as the ESP-IDF code is pre-compiled with a common config file (refer to tools/sdk/{target}/sdkconfig for details)

dizcza, 2021-12-13:
Got it. Thanks.

closed 2021-12-13 by dizcza