espressif/arduino-esp32 #6012: esptool 64bit installed on Windows x86 (32bit)

softhack007, 2021-12-11:
While the Arduino IDE (v1.8.16) work perfectly on Windows7 32bit, the ESP32 tools fail to compile my sketch:
```
Arduino: 1.8.16 (Windows 7), Board: "WEMOS D1 MINI ESP32, 80MHz, Default, 240MHz (WiFi/BT), 921600"

fork/exec C:\Users\Eee PC\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\3.1.0/esptool.exe: Die Version von %1 ist nicht mit der ausgeführten Windows-Version kompatibel. Öffnen Sie die Systeminformationen des Computers, um zu überprüfen, ob eine x86-(32 Bit)- oder eine x64-(64 Bit)-Version des Programms erforderlich ist, und wenden Sie sich anschließend an den Herausgeber der Software.

Error compiling for board WEMOS D1 MINI ESP32.

```

I have checked the esptool.exe installed by the ESP32 board package using "depends", and in fact it seems to be the 64bit version.

BTW, I saw that you wiped away a similar question from another user, saying "time to get an up-to-date PC and OS". Actually I am using an older Laptop (Win7 32bit) in my workshop, as a kind of "secondary arduino coding facility". And it works perfect for all my Arduinos including an UNO, a MEGA, and NANA EVERY. Just my new D1 R32 board should not work in this setup? 

C'mon, you can do better ... If the esptool.exe cannot be installed for win32, then at least some information could be provided on how to find a 32bit version for download.

softhack007, 2021-12-11:
PS: on the espressif website, it say that 32bit and 64bit versions are available for windows - at least I understand that a 32bit Windows if supported?

https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

> Starting with the Arduino IDE version 1.6.4, Arduino allows installation of third-party platform packages using Boards Manager. We have packages available for Windows, macOS, and Linux (x86, amd64, armhf and arm64).

atanisoft, 2021-12-12:
Windows 7 (and other 32bit OSes) have been de-supported by many of the underlying tools used for building and by Microsoft. It's time to upgrade to a 64bit OS.

softhack007, 2021-12-13:
Hello @atanisoft, are you saying that espressif are lying to their customers? Because their website clearly states that board packages for x86 are available.

Technical question: will the following  produce an "esptool" that works in arduino-esp32?
* install python 2.7.18 from python.org, make sure to select "add to PATH"
* open windows command, enter the following:
 ```
pip install setuptools wheel
 pip install esptool
```

atanisoft, 2021-12-13:
I'm not saying they are lying at all. The documentation may be clearly about the Arduino IDE and not the esp32 code.

atanisoft, 2021-12-13:
Also missed the bit about pip etc, that may give a usable esptool but it won't be used in the IDE which expects and uses esptool.exe which has only been made available using python x64 for a very long time.

Also note that python 2.x is already on the desupport path and may already be unsupported.

As noted originally, it is time to upgrade to a supported OS.

softhack007, 2021-12-13:
Hi @VojtechBartoska, yes i can understand that officially supporting another platform with just a few remaining users is a lot of effort. I was more looking for some do-it-yourself guide on how to get the toolchain up and running on x86, unfortunately got no help here.

@atanisoft I know you might be worried about the safety of my "tinker laptop" when major vendors are not providing active  support for Win32 any more. I can assure you that I know what i'm doing, and I am clear about the risks. My tinker laptop is not connected to the internet, and just needed for some quick arduino hacking. However it does not help much when you suggest to spend >350€ (for a new laptop that can run win64) just to work with a device that costs around 10€. Maybe you also understand me on this.

A last question - i would also be willing to "downgrade" arduni-esp32 to an older version - Can you indicate which release still does include working support for win32?

igrr, 2021-12-13:
For the DIY approach: to get an esptool.exe which would work with arduino-esp32, you need to do the following using a 32-bit Python installation:

https://github.com/espressif/esptool/blob/5c742218233098934300abe5b9c3d357a7d139cb/.github/workflows/build_esptool.yml#L31-L36

(i.e. install pyinstaller, then use pyinstaller to create an .exe file from [esptool](https://github.com/espressif/esptool))

It's not impossible to add this step to our automated builds of esptool, but I need to check if it's still possible to install 32-bit Python in the CI environment.

closed 2021-12-14 by me-no-dev

softhack007, 2021-12-14:
Maybe you can also add a note for windows users, to clarify the minimum required OS. Something like we find on python.org "ESP32 core V2.0.x cannot be used on Windows 7 or earlier. Please also note that 32bit OS are no longer supported by the toolchain."

And to answer my own question:
**ESP32 version 1.0.6** still works on win32. You might want to install python 3.8.x because it is the latest release supporting windows7.

:1st_place_medal: Thanks for the DIY guide, i will try it over the weekend :+1:

atanisoft, 2021-12-14:
@softhack007 It's actually slightly simpler, it's the 32bit windows versions that are no longer supported out-of-box. The included esptool.exe is compiled as x64 only (due to Python limitations noted above).

Edwardphantom, 2022-12-13:
Hi ppl , just doing my part:

Esp32 version 1.0.6 work like a charm on Windows 8.1 32-bits, V 2.x and up don't work on 32 bits anymore.

have a nice day!

[image]

VojtechBartoska, 2022-12-14:
Hi @Edwardphantom, as mentioned above. 32bit Windows version are no longer directly supported.

5chufti, 2023-06-13:
if the project does not care about 32bit Versions of Windows, then why are all other packages available for Win32bit but the final one to complete the compile???

me-no-dev, 2023-06-13:
> if the project does not care about 32bit Versions of Windows, then why are all other packages available for Win32bit but the final one to complete the compile???

this have nothing to do with `this` project. Tools come from upstream. We are just using them

5chufti, 2023-06-13:
yeah, but you decide the content of the package - and at the moment it lets you believe it installed correctly on Win32 - I didn't see any warning albeit some previous posts suggest there should be some.
And as esptool is perfectly running on Win32 with python 3.7 installed, it should be at least possible to build the tool once to include - no need for constant rebuild in CI, just consider it being "static" (as Win32).

me-no-dev, 2023-06-13:
Most tools, even though marked as win64, do work fine on win32 (or at least that is how they were reported). Another issue is that those tools get updated, bugs fixed and necessary new features added, so they can not be static as you propose. Windows binaries apart from being turned into EXEs, need to also be signed so that they do not trip defender and such. You can rebuild the binary yourself on your computer and replace the one installed by the IDE. If compiled on your computer, you do not need to sign it.

5chufti, 2023-06-13:
[esptool_win32.zip](https://github.com/espressif/arduino-esp32/files/11734098/esptool_win32.zip)
although not considered "useful" I "compiled" the recent version and provide it here for the benefit of whoever may need them.
I did copy it to a different machine and did not get any warning from defender, but YMMV.

laurieboshell, 2024-01-04:
A big thankyou to the help offered for 32 bit support. Some posters might not appreciate that some of us need to use a 32 bit OS for various reasons, and on the latest hardware too, myself included. I need to use Win10 32 bit, I support a medical billing software product. It's easier than firing up a VM each time.

5chufti, 2024-01-11:
hmmm, can't say much about that - just that I dl'd the sources and all the python stuff and compiled it.
Windows defender didn't and does not complain about the exectutables on any of my pcs...
Haven't experienced any suspicious symptoms on any of these pcs either - maybe because I don't operate as admin on my windos boxes?

dman271828, 2024-02-11:
I found a solution. I replaced the esptool.exe 64bit with the 32 bit version. I already have Thonny installed with its esptool ( pip install esptool). That version is python works with 32bit. I just copied the esptool.exe from its Thonny library folder to the folder with the esptool that is 64 bit for Arduino. After that I did the Arduino compile Blink test with an ESP-CAM with GPIO 4 led. It Blinked with success!
[image]

henigo, 2024-04-23:
It is important to continue supporting 32 bit architecture.